betaflight/betaflight #8881: Which Current Sensor is installed on SPRacing H7 Extreme

TheIsotopes, 2019-09-13:
@hydra can you tell me which current sensor is installed on the SPRacing H7 Extreme.

- in the web and in the manual stands 110A current sensor is installed, default should be `set ibata_scale = 300` then
- but a default scale of 225 `set ibata_scale = 225` says it has a 146A current sensor installed

what is right now?

maGfpv, 2019-09-13:
good question. default readings are way off

hydra, 2019-09-14:
It's the INA139, same current sensor I've been using on all the SP Racing products for years. it works well  except for the fact that it's readings are inaccurate at low currents when configured for 110A.

One resistor (RL) is used to determine the range based on the other large resistor (RS) which handles the power.  The 110A limit is based on these two resistors.  For more information refer to the INA139 datasheet.

http://www.ti.com/lit/ds/symlink/ina169.pdf

There's a few videos on the net which show how to calibrate it.

TheIsotopes, 2019-09-14:
@hydra ok, because on the other boards the same current sensor is installed and there is a default of 300, it should be the same with the h7 extreme

hydra, 2019-09-14:
because the RL resistor value is different, i.e. different Mv/A

TheIsotopes, 2019-09-14:
ok, i understand ... thx

closed 2019-09-14 by TheIsotopes