betaflight/betaflight #8882: Why is there 3times msp activated as defaul on SPRacing H7 Extreme

maGfpv, 2019-09-13:
@hydra - why is there always additionally msp for uart 1 and 2 activated as default? i thought it should only be activated once for usb vcp?

hydra, 2019-09-14:
it's so that you can connect up bluetooth hardware in the field, feel free to disable MSP on any ports you don't need it on.  it doesn't have any noticable impact if the ports are unused.  I'm open to suggestions for reasons why it might be useful to only have it on VCP by default.

mikeller, 2019-09-14:
One of the things to consider with respect to activating MSP on UARTs by default is that, in the case that the user connects a device that starts sending data on power up (like some RX) to the UART in question, this will most likely lead to the firmware being overloaded with failing MSP requests, and wedging whenever MSP is enabled on this UART (i.e. after a reset to defaults).

closed 2019-10-21 by stale[bot]