betaflight/betaflight #8883: No uarts in port tab - kakute f4v2 - bf 4.1 rc1 and rc2

JurKramer, 2019-09-13:
**Describe the bug**
I can't see any uarts in the port tab

**To Reproduce**
Tried rc1 and rc2, when i flash 4.0 uart are visible again

'''# version
# Betaflight / STM32F405 (S405) 4.1.0 Sep 12 2019 / 22:11:25 (4433dbb51) MSP API: 1.42

# start the command batch
batch start

# resources
resource BEEPER 1 NONE
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00
resource PPM 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_RX 1 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource INVERTER 3 NONE
resource I2C_SCL 1 NONE
resource I2C_SDA 1 NONE
resource LED 1 NONE
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 NONE
resource SPI_SCK 3 NONE
resource SPI_MISO 1 NONE
resource SPI_MISO 3 NONE
resource SPI_MOSI 1 NONE
resource SPI_MOSI 3 NONE
resource ESCSERIAL 1 NONE
resource ADC_BATT 1 NONE
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 NONE
resource FLASH_CS 1 NONE
resource OSD_CS 1 NONE
resource GYRO_EXTI 1 NONE
resource GYRO_CS 1 NONE
resource USB_DETECT 1 NONE

# timer
timer C07 NONE
timer B00 NONE
timer B01 NONE
timer A03 NONE
timer A02 NONE
timer C08 NONE

# dma
dma ADC 1 NONE
dma pin C07 NONE
dma pin B00 NONE
dma pin B01 NONE
dma pin A03 NONE
dma pin A02 NONE
dma pin C08 NONE

# master
set mag_bustype = SPI
set mag_i2c_device = 0
set mag_hardware = NONE
set baro_bustype = SPI
set baro_i2c_device = 0
set serialrx_provider = FPORT
set motor_pwm_protocol = DSHOT600
set battery_meter = NONE
set beeper_inversion = OFF
set beeper_od = ON
set tlm_inverted = OFF
set tlm_halfduplex = ON
set system_hse_mhz = 0
set max7456_spi_bus = 0
set max7456_preinit_opu = OFF
set dashboard_i2c_bus = 0
set flash_spi_bus = 0
set gyro_1_bustype = GYROAUTO
set gyro_1_spibus = 0
set gyro_1_sensor_align = CW0
set gyro_2_spibus = 0

profile 0

rateprofile 0

# end the command batch
batch end'''

**Setup / Versions**
 - Flight controller Kakute F4 v2

**Additional context**
It also wont see the gyro, accel and baro meter

Docteh, 2019-09-13:
Hi, go into the cli and punch in "defaults". It'll apply the configuration and come up with the missing items.

One bit still in progress:
https://github.com/betaflight/betaflight-configurator/pull/1651

JurKramer, 2019-09-13:
cool, it works, thanks

closed 2019-09-13 by JurKramer