betaflight/betaflight #8884: BF4.1rc2 Leds blink when using VTX channel as modifier

aleksirastas, 2019-09-13:
When using VTX modifier for led color assigning leds blink. The color is changing as it should be but leds just blink in appropriate color. When using normal led color or AUX channel the led behavior is normal solid color. 

Steps to reproduce the behavior:
Assign led color based on VTX channel. 

**Expected behavior**
Leds start blinking

**Flight controller configuration**
``` # led
led 0 0,0::CV:0
led 1 1,0::CV:0
led 2 2,0::CV:0
led 3 3,0::CV:0
led 4 4,0::CV:0
led 5 5,0::CV:0
led 6 6,0::CV:0
led 7 7,0::CV:0

# color
color 0 0,0,255
color 1 117,0,206
color 2 238,9,255
color 3 0,0,255
color 4 240,0,255
color 10 30,0,255
color 11 120,0,255

# mode_color
mode_color 7 0 9```

**Setup / Versions**
 - FF_racepit

**Additional context**
4.0.x and earlier bf versions are working great.

etracer65, 2019-09-13:
I think I see the problem. Looks like if it isn't actually talking to a vtx it erroneously defaults to showing that pit mode is active (blinking).

Try this:

[betaflight_4.1.0_FF_RACEPIT.hex.zip](https://github.com/betaflight/betaflight/files/3611711/betaflight_4.1.0_FF_RACEPIT.hex.zip)

etracer65, 2019-09-13:
Actually this is not a bug. Betaflight 4.0 behaves the same way. In this case the flashing at power on is an indication that the flight controller is not communicating with the VTX and can't update the status. If you had the VTX connected and the UART set properly for smart audio or tramp (depending on the VTX) the LEDs will only flash when the VTX is in pit mode.

aleksirastas, 2019-09-14:
How come that Smart Audio is wokring, I can change my channels and the led color changes as it should?

aleksirastas, 2019-09-14:
[image]

etracer65, 2019-09-14:
The blinking is supposed to happen if either there's no communication with the VTX or the VTX is reporting that it's in pit mode. Can you test with 4.0.6 and see if the behavior is the same?

aleksirastas, 2019-09-14:
I updated from 4.0.6 and there it was working correctly. I just copied the led config from 4.0.6 setup.

aleksirastas, 2019-09-14:
And I have 3 identical quads and same problem with all of them using 4.1rc2

etracer65, 2019-09-19:
I tested more and still can't replicate any problems. Here's a video of BF4.1 (RC3+) with a smartaudio connected VTX. Used your LED settings (except I only have 4 connected). I cycle through 4 channel changes.

https://youtu.be/KpvLxgfpdRM

The only ways I can replicate the blinking VTX overlay is if the VTX wasn't communicating (smartaudio not working) or if the VTX is in pit mode. Both of these cases are by design.

Are you sure you have your `vtxtable` set up properly for your VTX? You haven't told us which VTX you're using. It looks like you just copied the first example and that may not be appropriate for your VTX - particularly the power settings. From your `diff` it would appear to be set up for the TBS Unify Pro32 Nano 5G8. If that's not what you have, take a look here and find an example for your VTX (scroll down):

https://github.com/betaflight/betaflight/blob/master/docs/VTX.md#vtx-table

aleksirastas, 2019-09-19:
I need to try it with rc3. Im still using rc2. This will need to wait till next week. 
I have unify nano pro32 thus the settings.

closed 2019-10-26 by stale[bot]