betaflight/betaflight #8888: Tramp protocol VTX quits changing band and channel

0crap, 2019-09-14:
**Describe the bug**
If I change a band or channel on my Matek VTX it does it sometimes once and after that it will not change Band or Channel anymore. Funny thing is that changing the power setting keeps functional.
On the OSD you see dashes behind BAND and CHAN in the stuck situation, see picture.
I have the VTX on raceband but it seems to read back * U F  5843  200 
The U F part in this line is not normal.

[image]

**To Reproduce**
Change channels using OSD or LUA.

**Expected behavior**
Be able to change channels as often as I like without getting stuck.

**Flight controller configuration**
``` # diff all

# version
# Betaflight / MATEKF722 (MKF7) 4.1.0 Sep 14 2019 / 13:11:48 (7e160e138) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722
mcu_id 002700193237510d31313935
signature 

# name: REVERB5

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# feature
feature -AIRMODE
feature -DYNAMIC_FILTER
feature LED_STRIP

# beeper
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB
beeper -BLACKBOX_ERASE

# serial
serial 4 8192 115200 57600 0 115200

# led
led 0 11,15::AB:0
led 1 11,14::AB:0
led 2 4,14::AB:0
led 3 4,15::AB:0

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 900 1100 0 0
aux 1 1 1 900 1100 0 0
aux 2 13 2 1900 2100 0 0
aux 3 28 1 1400 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 400

# rxfail
rxfail 3 h
rxfail 5 s 1000

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 250
set gyro_lowpass2_hz = 0
set gyro_notch1_hz = 240
set gyro_notch1_cutoff = 230
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = 4,-3,35
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1015
set max_check = 1990
set rc_smoothing_input_hz = 40
set rc_smoothing_derivative_hz = 20
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set max_aux_channels = 4
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 420
set vbat_warning_cell_voltage = 340
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_min_sats = 6
set pid_process_denom = 1
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_altitude = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_distance = ON
set osd_rssi_alarm = 45
set osd_cap_alarm = 1200
set osd_rssi_pos = 2489
set osd_tim_2_pos = 2518
set osd_flymode_pos = 2465
set osd_current_pos = 2496
set osd_mah_drawn_pos = 2511
set osd_craft_name_pos = 363
set osd_display_name_pos = 33
set osd_gps_speed_pos = 463
set osd_gps_sats_pos = 46
set osd_home_dir_pos = 54
set osd_home_dist_pos = 55
set osd_debug_pos = 353
set osd_avg_cell_voltage_pos = 2505
set osd_core_temp_pos = 438
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_gps_sats_show_hdop = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 2
set vtx_freq = 5843
set vcd_video_system = PAL
set gyro_rpm_notch_q = 700
set name = REVERB5
set display_name = BAS

profile 0

# profile 0
set anti_gravity_gain = 8000
set feedforward_transition = 10
set iterm_relax_cutoff = 10
set p_pitch = 50
set f_pitch = 180
set p_roll = 46
set f_roll = 180
set f_yaw = 100
set d_min_roll = 0
set d_min_pitch = 0
set d_min_advance = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 15
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 78

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save ```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: LED 2
A15: LED_STRIP
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: USB_DETECT
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2:
    CH1: LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: LED_STRIP
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - MATEKF722
 - MATEK FCHUB-VTX, using the vtx tables recommended on MATEK's website.
 - VTX connected to UART

Juzzle1, 2019-09-14:
I had the same issue with my FCHUB-VTX but fixed it with settings published by Matek (http://www.mateksys.com/?portfolio=fchub-vtx#tab-id-6)

vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 400
save

Juzzle1, 2019-09-14:
FYI I've just double checked and mine is definitely working OK through OSD and LUA (I also have MATEKF722 and FCHUB-VTX) running 4.1 RC2.

0crap, 2019-09-15:
@Juzzle1 Yes, I used the example on the Matek website.
I have two of these stacks and they give the same erratic behaviour.
So I'm not yet convinced it's defective hardware.

Can you please test it again by selecting Raceband and walk through the channels with LUA or OSD?
Sometimes it takes a while and I have to select CH 1,2,3,4,5,6,7,8 and back to e.g. 1 and then hangs.
Always like the picture above with the dashes on the OSD.
Thanks for your time.

0crap, 2019-09-15:
OK, I reverted to 4.0.6 and configured the bare minimum.
I can't reproduce the issue. So no defective hardware if you ask me.

```
# diff all

# version
# Betaflight / MATEKF722 (MKF7) 4.0.6 Sep  1 2019 / 00:50:56 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722
manufacturer_id 
mcu_id 002700193237510d31313935
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set acc_calibration = 4,0,24
set mag_hardware = NONE
set baro_hardware = NONE
set max_aux_channels = 4
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 2
set vtx_freq = 5843

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

Juzzle1, 2019-09-15:
OK you're definitely on to something here. 

I've just tried cycling through channels R1 to R8 and back to R1, and the change from R8 to R1 didn't work. There's also a huge delay changing channels, which I don't remember being there on 4.0 - can you check if you've just downgraded?

I recorded it with DVR  (https://youtu.be/oB1sXdYdkIg)
Watch from 2:25, where it's on channel R8 and I switch to R1.

The channel switches to another frequency and then immediately  goes back to R8 again.
I waited for a little while, then at around ~3:00 I tried setting R1 again two more times and only the last attempt worked. (I switched to channel 1 on the DVR at ~3:16)

I'm not seeing the dashes you describe though...

0crap, 2019-09-15:
Thx for the vid, on 3:04 you gave it a second set, if you let it sit and let it struggle will it go or not?
Maybe it times out and will give you the dashes.
For me it's random when it hangs itself, not only from CH8 -> CH1. Might happen at any random channel switch.

I hope someone else with a Tramp VTX can test this, so we know it's not only a Matek problem.

Juzzle1, 2019-09-16:
I've managed to reproduce the dashes, and yes it happens on any channel, not just from R8 to R1.

[image]

I've annotated a video which shows:

Successful change from R6 to R7
Successful change from R7 to R8
Failure changing from R8 to R1
Repeated attempts to move from channel R8, along with 'dashes' in OSD.
https://youtu.be/tK-LmQJHT00

FYI I've just upgraded to 4.1 RC3 and the problem is still there.

Juzzle1, 2019-09-16:
For comparison here is my config:

```
# Version
Betaflight / MATEKF722 (MKF7) 4.1.0 Sep 15 2019 / 08:58:50 (644ab5d90) MSP API: 1.42

board_name MATEKF722
mcu_id 003f00483237510431313935
signature 

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# feature
feature LED_STRIP
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon
beacon RX_SET

# serial
serial 0 1024 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led
led 0 3,4:W:AT:0
led 1 3,5:W:AT:0
led 2 3,6:W:AT:0
led 3 3,7:W:AT:0
led 4 7,4:E:AT:0
led 5 7,5:E:AT:0
led 6 7,6:E:AT:0
led 7 7,7:E:AT:0

# mode_color
mode_color 6 1 6
mode_color 7 0 7

# aux
aux 0 0 0 1650 2100 0 0
aux 1 2 0 1950 2100 0 0
aux 2 2 0 1575 1675 0 0
aux 3 2 0 1175 1300 0 0
aux 4 13 1 1825 1950 0 0
aux 5 13 1 1325 1425 0 0
aux 6 15 1 1075 1150 0 0
aux 7 15 1 1575 1675 0 0
aux 8 19 1 1075 1150 0 0
aux 9 19 1 1200 1300 0 0
aux 10 19 1 1325 1400 0 0
aux 11 35 2 1600 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = 15,6,29
set mag_hardware = NONE
set dshot_idle_value = 1
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set current_meter = ESC
set battery_meter = ESC
set ibata_scale = 114
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_cap_alarm = 1100
set osd_alt_alarm = 400
set osd_vbat_pos = 14743
set osd_rssi_pos = 14392
set osd_link_quality_pos = 57
set osd_remaining_time_estimate_pos = 14765
set osd_flymode_pos = 14753
set osd_g_force_pos = 14785
set osd_vtx_channel_pos = 14356
set osd_current_pos = 14806
set osd_mah_drawn_pos = 14776
set osd_craft_name_pos = 14338
set osd_altitude_pos = 448
set osd_warnings_pos = 14730
set osd_battery_usage_pos = 14793
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set name = JUZZLE

profile 0

# profile 0
set idle_min_rpm = 18

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 25
set thr_expo = 50
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: LED 2
A15: LED_STRIP
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: USB_DETECT
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: FREE
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2:
    CH1: LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: LED_STRIP
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 1
DMA2 Stream 3: MOTOR 2
DMA2 Stream 4: MOTOR 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 4

```

0crap, 2019-09-16:
@Juzzle1 Awesome work! Seems to me bug is verified.
Now hope @mikeller will add the BUG label to this ticket.

I'll ask on BF SLACK as well if anyone with a, non MATEK, Tramp VTX can do some testing.
To make sure it's not only the MATEK VTX giving issues.

Juzzle1, 2019-09-16:
@0crap it would also be interesting to see if it's affecting people with the same MATEK FC but a different Tramp VTX.

etracer65, 2019-09-16:
Yes, there is something wrong. I was able to replicate with MATEKF722 and FCHUB-VTX. Digging a little I see that the VTX seems to be occasionally ignoring a command to change frequency and then things get in a messed up state as the frequency from the VTX doesn't match the expected value.

Has anybody been able to confirm this problem with a non-Matec VTX? Also does anybody know of a build where it works ok?

ijustwannafly, 2019-09-17:
Speedybee TX500.  IRC Tramp.

The issue happened on this setup.  I was also setting power levels to 3.  When I went back and set power levels to 5 with 25, 100, 200, 400, 600 labels and 25, 25, 200, 500, 500 values I was able to switch channels and bands via osd.  Lua script failed after the first channel change.

I also noticed the vtx is not coming out of the low power setting despite the option "Low power on disarm" set to ON.  It stays at 25mW.

Older working betaflight low power disarm would result in 500mW on the same vtx when armed and 25mW on disarm.

etracer65, 2019-09-17:
I tested on a SpeedyBee TX500 and can't replicate.

In low-level debugging I see that the underlying cause is that the Matek FCHUB-VTX is periodically ignoring (not responding to) commands. Seems random in nature. This is triggering an undesirable behavior in other changes made to the Tramp driver (that wasn't expecting the ignored commands). Before these changes the code used a different method of sending changes/polling that would have been more tolerant (just by chance - not design) to these unexpected ignored commands so that's why it's a problem now.

The SpeedyBee TX500 is responding properly to all commands so it didn't encounter the problem.

The Matek FCHUB-VTX appears to have an implementation problem leading to the ignored commands. We can't fix that but can try to work around it.

I'm working on the problem.

etracer65, 2019-09-17:
@0crap @Juzzle1 Please try this version. Should resolve the problems with the Matek FCHUB-VTX and generally make Tramp control more robust.

[betaflight_4.1.0_MATEKF722.hex.zip](https://github.com/betaflight/betaflight/files/3623516/betaflight_4.1.0_MATEKF722.hex.zip)

Juzzle1, 2019-09-17:
@etracer65 I can confirm that channel / band / power changes every time without fail - thanks very much for looking into this.

I will mention that the first time i tested this firmware, I cycled from Ch R1 to R4, at which point my OSD locked up, and aux switches were not functioning, but board was responisve via USB in configurator.

I had to reboot the board, and then have spent the last 45 mins cycling through channels trying to reproduce but couldn't.

Much more likely it was something to do with my FrSky TX / RX but though I should mention it as I hadn't seen that happen before.

Thanks!

ijustwannafly, 2019-09-18:
I did more testing on my speedybee tx500.  This is on 4.1 RC1 mind you.  The test hex is for a matek so I can't try that and the speedybee tx500 is being used on a lumenier lux v2 osd micro.

Here are my observations.

With the vtxtable configured as here https://www.speedybee.com/product_images/download/manual/SB-TX500-manual.pdf I can change bands and channels up to a certain point.

When I try to go to raceband it does not switch over.  I did this by watching the vtx leds while the band is changed.  For all bands except raceband there is a corresponding led to indicate the band.  The leds change when going from boscam through to fatshark.  When going from fatshark to raceband the led does not change.  It stays on the led position that indicates fatshark.

When changing power, 25mW to 200mW works.  200mW to 500mW does not work.  Again, this is based on the led status.  Solid red led is unlock mode and 25mW.  Solid red with blinking orange is 200mW and that succeeds.  Trying to go to 500mW does not result in a solid red and solid orange led.  

Despite setting up the vtx table according to the vtx manual AND the betaflight vtx wiki, it seems the vtx is unable to get into raceband or into 500mW on this particular vtx.

Tramp related?  vtxtable related?  betaflight induced?  It's hard to tell.

etracer65, 2019-09-18:
@ijustwannafly I just tested a SpeedyBee TX500 and stepped through all 40 channels (verifying with a receiver) - no problem changing to any channel in any band.

Regarding the 500mw power: The SpeedyBee TX500 has a bug in its Tramp implementation and is reporting its max power output as 200mw. So Betaflight is limiting the settings to this. This was not implemented in 4.0 so that's why it worked there. I would suggest contacting SpeedyBee and reporting the problem - possibly they have a firmware update to fix the bug.

If you'd like to try the test firmware:

[betaflight_4.1.0_LUXF4OSD.hex.zip](https://github.com/betaflight/betaflight/files/3624311/betaflight_4.1.0_LUXF4OSD.hex.zip)

If won't do anything for the 500mw problem with this vtx, but maybe it will resolve your channel problem (although I could never replicate the problem with this vtx). Also, here is the `vtxtable` I used:

```
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500
```

ijustwannafly, 2019-09-18:
@etracer65 

I tested your test hex and used your vtxtable.

It still does not properly get into raceband on the speedybee.

Maybe I should start a new issue thread?

Again, I am watching the leds on the vtx when changing channels and power.  On 4.0.6, the leds properly show the bands (especially raceband) and the power levels.

On 4.1 (including the test) the vtx is unable to be switched into raceband.  Could you please test your side again by watching the leds for the status as you change your band/channel/power.  

I do not believe my vtx is the issue as it works as expected on 4.0.6.

I understand the bf team would prefer I contact speedybee but I doubt they would provide a firmware for this vtx (think back to the akk debacle) and it makes me wonder what happen to the "open protocol" regarding peripherals (vtx for example) that was hinted during the akk debacle.

etracer65, 2019-09-18:
@ijustwannafly For the SpeedyBee TX500: setting aside the 500mw concern which is a known issue based on the erroneous data reported by the VTX, I have no problems whatsoever switching to any band/channel I've defined in the `vtxtable`. This includes RACE band. I've confirmed that the VTX actually transmits on the selected channels (using a receiver to make sure the video is actually transmitted - the most important part!) and also the LEDs on the VTX correctly show band 5 selected. Here's a short video of the VTX changing from FATSHARK (band 4) to RACE (band 5):

https://youtu.be/bIdE8wq17bc

Perhaps you're confused by the bands on this VTX? It actually supports 6 bands with band 6 being "LOW RACE". Perhaps you're confused by the band LEDs and expecting the last band indicator to light up? Band 6 (LOW RACE) is not going to be accessible unless you configure it in the `vtxtable`. For example, using the `vtxtable` settings:
```
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE L CUSTOM 5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500
```
Allows me to also select the "LOWRACE" band and the 6th band LED properly lights up on the VTX.

For reference, here's the band/channel information for this VTX:

[image]

I can't find any problems with band/channel switching with this VTX. Perhaps you should validate whether it's transmitting on RACE band by using a receiver as that's the real test. What the LEDs display (although correct as far as I can see) doesn't really matter when using VTX control.

ijustwannafly, 2019-09-18:
@etracer65 your video shows your vtx works according to the led status.

My speedybee doesn't shift its led from the 4th position to the 5th position (as in your video).

I am not confused about the bands as I linked the manual for the speedybee with the low race band table.

What you are able to do (according to the video) I cannot do on 4.1 rc1.  I can however accomplish getting into raceband on 4.0.6.  Just to clarify,  I have my goggles and 7 inch 5.8 monitor on during these tests and it does not register raceband when on 4.1 rc1.  4.0.6 works proper.

I hope I am missing something simple but even just going back to 4.0.6 and configuring the bare minimum needed for vtx control the speedybee works.  Flashing 4.1 rc1 and configuring the bare minimum for vtx control (your vtxtable) my speedybee does not work properly.  I am at a loss.  I guess I have to stay on 4.0.6 since my speedybee doesn't like 4.1.

etracer65, 2019-09-18:
@ijustwannafly I think you simply haven't "unlocked" the VTX. When I put mine back in "locked" mode it behaves as you describe - won't go into RACE band (band 5). Well not exactly true... It will go into R4, R5 and R6 but not change to any of the the other RACE band channels. Also BOSCAM-B (band 2) can't be accessed. This matches the specifications from the manual:

[image]

An indication that you're in "locked" mode is that the band indicator will be blinking. Push and hold the button for 10 seconds to toggle between locked and unlocked. Power cycle the VTX after changing.

ijustwannafly, 2019-09-18:
@etracer65 My tx500 is definitely unlocked.  My red led is constantly on and on 4.0.6 switching to 500mW gives me a constant orange led.  500/200mW is not possible in locked mode

etracer65, 2019-09-18:
Then I have no solution for you. On the 2 samples I have the SpeedyBee TX500 is working perfectly. Possibly you have a version with bugs in the firmware. Possibly try re-locking and unlocking it as a test.

ijustwannafly, 2019-09-19:
@etracer65 It just seems weird that on 4.0.6 my vtx works correctly.  On 4.1 rc1 it refuses to work properly.  The only change is the fc firmware version.  I could try rc3 but I don't think it would make a difference at this point.  The big difference between 4.0.6 and rc1 is the vtx tables and even when setting vtx tables according to the wiki it just refuses to work properly.  

But I concede.  If you believe my hardware is at fault there's not much else I can do.  Just another item on wasted money that works on one version and not another.  Time to get out of this side of the hobby...it's either too much fragmentation or proprietary.  So much for middle ground and "open".

closed 2019-09-19 by mikeller

0crap, 2019-09-22:
Sorry, I've been off the grid for a week because of unexpected private problems.
So I could not reply/test.
All OK now and I just tested the latest nightly, including this fix.

The VTX works great on two setups, one VTX connected to a MATEKF722, the other to a MAKTEKF405STD.

Thx for the effort!

ndalisay, 2019-10-19:
got same problem with the final release 4.1 on transtec f7 and matek hv vtx.. I cant set raceband 1-2 but the rest of the channels and band are working fine... but on 4.0.6 i dont have any problems

Hernie22, 2019-11-03:
I have a similar problem with the IFlight TwinG F7 FC and the iFlight 5.8G mini VTX 500mW (m3). I can change settings in Betaflight while connected to the PC, so i know the FC is talking to the VTX.  When going into the TA menu on the OSD its all blank lines and at the top it says the the power lever is 25mW and there is nothing i can do to change it.

PR #8903 (2019-09-19): Increase Tramp retry count and fix lockup if VTX out-of-sync with band/channel settings
Fixes #8888 

Some VTX don't respond to every request so it's necessary to retransmit. A previous PR fixed a bug that caused more retries than expected and made the actual retries only be 2 as configured. Unfortunately this exposed the problem with the non-responding VTX with cases where it failed to respond to both retries. The problem was accidentally masked by a previous bug with the retry counter causing many extra retries to actually occur.

This then exposed a secondary problem in `vtx_common` that le [... cut]