betaflight/betaflight #8894: Bitbang gives 100% esc telemetry error

HrrrPippi, 2019-09-15:
**Describe the bug**
I dont know if this is a Blheli_S (Jesc) or firmware problem, but when i updated to betaflight 4.1 RC1, the configurator shows 98-100% error in the motor tab at all motor speeds.
I disabled dshot_bitbang, and the error went away.

I have tried with dshot 300 and dshot600, tried running 4k/4k, 8k/8k, but still 100% error.

Tried with or without the timer settings from the matekf722se snippit from the rpm/telemetry page, but same result.

**To Reproduce**
Update to BF 4.1, enable dshot telemetry and go to the motor tab and power on the ESC's to see 100% esc telemetry errors.

**Expected behavior**
Please make sure Jesc firmware works well with BF 4.1 RC1

**Flight controller configuration**
```
# version
# Betaflight / MATEKF722SE (MF7S) 4.1.0 Sep  5 2019 / 20:14:40 (62f0eb7ed) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
mcu_id 003000343237510a34363730
signature 

# name: Flos2tyle

# timer
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)

# dma
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 3 2048 115200 57600 0 115200

# aux
aux 0 0 0 1575 1825 0 0
aux 1 1 1 1950 2050 0 0
aux 2 27 4 1300 1400 0 0
aux 3 13 0 1375 1425 0 0
aux 4 28 1 950 1050 0 0
aux 5 35 0 1750 1850 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 999

# master
set gyro_lowpass2_hz = 400
set gyro_notch1_hz = 160
set gyro_notch1_cutoff = 142
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_min_hz = 80
set dyn_lpf_gyro_min_hz = 320
set dyn_lpf_gyro_max_hz = 800
set acc_calibration = 106,-50,471
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 11
set serialrx_provider = IBUS
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_idle_value = 50
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT600
set vbat_scale = 109
set ibata_scale = 300
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 10000
set osd_tim1 = 0
set osd_tim2 = 17
set osd_vbat_pos = 8609
set osd_rssi_pos = 10697
set osd_tim_2_pos = 406
set osd_g_force_pos = 88
set osd_throttle_pos = 8225
set osd_current_pos = 8656
set osd_mah_drawn_pos = 14807
set osd_motor_diag_pos = 8248
set osd_craft_name_pos = 10282
set osd_altitude_pos = 366
set osd_debug_pos = 8545
set osd_warnings_pos = 14666
set osd_avg_cell_voltage_pos = 14785
set osd_core_temp_pos = 34
set osd_stick_overlay_left_pos = 4353
set osd_stick_overlay_right_pos = 4374
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_fft = ON
set osd_profile = 3
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5695
set name = Flos2tyle

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 112
set dyn_lpf_dterm_max_hz = 272
set dterm_lowpass2_hz = 240
set pid_at_min_throttle = OFF
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set d_min_advance = 0
set idle_min_rpm = 14

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

rateprofile 1

# rateprofile 1
set roll_rc_rate = 103
set pitch_rc_rate = 103
set yaw_rc_rate = 103
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 66
set tpa_rate = 50
set tpa_breakpoint = 1800
set throttle_limit_type = SCALE

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1
```
```
A00: SERIAL_TX 4
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: ADC_EXT
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: USB_DETECT
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 4
DMA1 Stream 3: FREE
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 3
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
``` 
**Setup / Versions**
 -Matek F722mini;
 - BlHeli_S 4in1 ESC running Jesc firmware version  JESC-J15 (1.4) with Telemetry (2.0)

HrrrPippi, 2019-09-15:
Update
I went and updated the JESC firmware to JESC-J15 (2.1) with Telemetry (2.1)
The error went away with the update, and bitbang is working fine now.

Dont know if there should be a note in the update info in the configurator, when updating to bf 4.1.x, about making sure to be at the most recent version of JESC firmware in order to make bidir dshot work.

Go ahead and close the issue if you think

jflyper, 2019-09-15:
@HrrrPippi 
Just FYI, information given in your original issue description indicates bit-banging DShot is not in effect.

- Bit-banging is not enabled (`dshot_bitbang = OFF') and
- DMA allocation list does not list any 'DSHOT_BITBANG', but `MOTOR`s).

HrrrPippi, 2019-09-15:
@jflyper oh shoot.. Forgot to change back to bitbanger on, before making the diff. I did get the error when dshot_bitbanger = on but the esc Rpm telemetry errors went away with dshot_bitbanger = off..

etracer65, 2019-09-15:
Try RC3.