betaflight/betaflight #8902: ESC´s do not receive throttle signal with Bidirectional DShot enabled

hObBiT-iM-bLuTrAuScH, 2019-09-17:
When Bidirectional DShot is enabled in the configurator, the esc´s only do three beeps when connecting the battery. I get telemetry but can not send any information from the TX using FPORT anymore. I can see coming signal from TX in betaflight. 

Also, i can not connect/read/flash ESC´s in Blheli suite, when Bidirectional DShot is enabled.

Everything fine when Bidirectional DShot is disabled.

# diff all

# version
# Betaflight / STM32F405 (S405) 4.1.0 Sep 15 2019 / 09:13:22 (644ab5d90) MSP API: 1.42

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 0034002c3337470b37393837
signature 

# resources
resource MOTOR 6 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource SERIAL_TX 11 A08
resource SERIAL_RX 6 NONE
resource SERIAL_RX 12 C07

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 2 16384 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 8192 115200 57600 0 115200
serial 31 1024 115200 57600 0 115200

# aux
aux 0 0 0 900 1050 0 0
aux 1 13 3 1600 2100 0 0
aux 2 35 2 1800 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 1 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set acc_calibration = -192,-63,310
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set dshot_idle_value = 300
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_switch_mode = STAGE2
set align_board_roll = 180
set bat_capacity = 1550
set small_angle = 180
set pid_process_denom = 1
set transient_throttle_limit = 10
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 1300
set osd_vbat_pos = 2049
set osd_rssi_pos = 2074
set osd_link_quality_pos = 89
set osd_vtx_channel_pos = 2432
set osd_current_pos = 407
set osd_mah_drawn_pos = 2061
set osd_motor_diag_pos = 396
set osd_craft_name_pos = 2456
set osd_warnings_pos = 2217
set osd_battery_usage_pos = 5
set osd_disarmed_pos = 2250
set osd_esc_tmp_pos = 402
set osd_esc_rpm_pos = 163
set osd_esc_rpm_freq_pos = 192
set osd_core_temp_pos = 394
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 3
set vtx_freq = 5865
set vtx_pit_mode_freq = 5865
set vcd_video_system = NTSC
set gyro_1_align_yaw = 1800

mikeller, 2019-09-17:
Can you please provide the output of `resource show all` as instructed when opening this issue?

hObBiT-iM-bLuTrAuScH, 2019-09-17:
sry
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 11
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: PREINIT 4
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 12
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: PREINIT 3
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERIAL_TX 11
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: SERIAL_RX 12
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

hObBiT-iM-bLuTrAuScH, 2019-09-17:
Also: Flashed the ESC´s to the newest beta firmware 32.6.6 for HGLRC_DinoShot_60A_ST_Multi_32_66.Hex and deavtivated music on the ESC´s

etracer65, 2019-09-18:
Try:
```
set gyro_sync_denom = 2
set motor_pwm_protocol = DSHOT300
save
```

hObBiT-iM-bLuTrAuScH, 2019-09-18:
no change

jflyper, 2019-09-18:
@hObBiT-iM-bLuTrAuScH 

I can successfully spin motors without a problem with dshot bidir, by
1. Flash an original Omnibus F4 AIO with 4.1-RC3 STM32F405 target.
2. Config the board with `OMNIBUSF4.config`.
3. Add snippet from `https://raw.githubusercontent.com/betaflight/bidircfg/master/DEFAULT-upgrade.cf`.
4. Spin motors with motor tab sliders.

I would suggest you do the same and check if motors spin, and then start customizing.

hObBiT-iM-bLuTrAuScH, 2019-09-18:
How could it work with bidir disabled if it was the wrong firmware?

I love what you guys do and really want to support you, but i can only report back to you devs.

I´ll check if motors spin in betaflight soon.

etracer65, 2019-09-18:
> How could it work with bidir disabled if it was the wrong firmware?

Because when `dshot_bidir` is enabled the signal is inverted. The 32.6.6 beta BLHeli32 firmware is needed to understand that. When not enabled the signal is normal and will work with older BLHeli32 firmwares.

etracer65, 2019-09-18:
Please provide a screenshot of the BLHeli32 Suite showing your ESC settings.

hObBiT-iM-bLuTrAuScH, 2019-09-18:
Of course no spin in betaflight. Should have been clear, because i only get 3 beeps out of the esc/motors with bidir enabled. And again everything is FINE with bidir disabled.

hObBiT-iM-bLuTrAuScH, 2019-09-18:
FYI: I have to disable bidir to be able to connect in blheli.

[image]

hObBiT-iM-bLuTrAuScH, 2019-09-18:
So, as soon as bidir=on there is no more communication between fc and esc. there are only 3 beeps on powerup, which means there is no throttle signal recognized by the esc´s. I have gone through the entire process many times now. The fact that i can´t even connect in blheli with bidir=on tells me that the esc with the test firmware for this esc may not recognize the inverted signal, yet.

I can keep everything as it is, except i have to set bidir=off and i can connect, update firmware, fly and of course get my 5 beeps.

hObBiT-iM-bLuTrAuScH, 2019-09-18:
Where to report esc firmware issues or could i´ve been missed a setting?

jflyper, 2019-09-19:
@hObBiT-iM-bLuTrAuScH 

Did you try the procedure described above? (https://github.com/betaflight/betaflight/issues/8902#issuecomment-532704303)

hObBiT-iM-bLuTrAuScH, 2019-09-19:
The procedere described above is already part of the original instructions. So please take it for granted that I have read and considered this manual. I'm not looking for support. I'm just reporting back.

Report: No communication/ passthrough with bidir = on

Mateyhv, 2019-09-21:
I am having almost the same issue, but interestingly only on ESC #4. I flashed a SPRACING H7 Extreme with Betaflight 4.1.0 RC3. The Mamba 506 4in1 ESC was loaded with MAMBA_506_Multi_32_66.Hex, Dshot300. When Bidirectional Dshot is NOT activated all 4 motors spin correctly and follow throttle commands. As soon as Bidirectional Dshot is activated ESCs #1-2-3 work fine and show telemetry but ESC #4 shows red I:100%, and motor does not spin. Blheli32 recongizes all four ESCs but motor 4 still refuses to spin. As soon as I set Bidirectional Dshot to OFF, ESC #4 spins the motor correctly. Here is the config information sans VTX table and beepers.

# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep 15 2019 / 09:12:03 (644ab5d90) MSP API: 1.42

# reset configuration to default settings
defaults nosave

board_name SPRACINGH7EXTREME
mcu_id 0042001a3437511931353431
signature 

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER
feature ESC_SENSOR

# beacon
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 26 0 1800 1900 0 0
aux 2 35 0 1700 1800 0 0
aux 3 39 0 1200 1300 0 0

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set fpv_mix_degrees = 25
set serialrx_provider = SBUS
set rx_min_usec = 925
set rx_max_usec = 2075
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set current_meter = ESC
set battery_meter = ESC
set vbat_scale = 255
set vbat_divider = 22
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set debug_mode = GYRO_SCALED
set name = Mark 4

profile 0

# profile 0
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set throttle_boost = 0

# resource
resource BEEPER 1 D07
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 C06
resource MOTOR 8 C07
resource PPM 1 B15
resource PWM 1 B15
resource LED_STRIP 1 A08
resource SERIAL_TX 1 B14
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 B13
resource SERIAL_TX 6 C06
resource SERIAL_TX 8 E01
resource SERIAL_RX 1 B15
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 5 B05
resource SERIAL_RX 6 C07
resource SERIAL_RX 8 E00
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 E03
resource TRANSPONDER 1 B11
resource SPI_SCK 2 D03
resource SPI_SCK 3 B03
resource SPI_SCK 4 E12
resource SPI_MISO 2 C02
resource SPI_MISO 3 B04
resource SPI_MISO 4 E13
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 D06
resource SPI_MOSI 4 E14
resource CAMERA_CONTROL 1 E05
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C04
resource ADC_CURR 1 C00
resource ADC_EXT 1 C05
resource SDCARD_DETECT 1 D10
resource SDIO_CK 1 C12
resource SDIO_CMD 1 D02
resource SDIO_D0 1 C08
resource SDIO_D1 1 C09
resource SDIO_D2 1 C10
resource SDIO_D3 1 C11
resource OSD_CS 1 E11
resource GYRO_EXTI 1 D04
resource GYRO_EXTI 2 E15
resource GYRO_CS 1 A15
resource GYRO_CS 2 B12
resource VTX_POWER 1 B01
resource VTX_CS 1 B00
resource VTX_DATA 1 A06
resource VTX_CLK 1 A07

[image]

maGfpv, 2019-09-21:
I have exactly the same behavior. Its listed here:
"Bidirectional DSHOT not working on SPRACINGH7EXTREME #8836"

Mateyhv, 2019-09-21:
@maGfpv thanks! I will move my post to the #8836 thread.

joelucid, 2019-09-21:
This looks weird: dshot_bitbang isn't set in the diff so it should be set to default. So bidir should use bitbang - but the timer and dma allocation shows timer dshot being used.

joelucid, 2019-09-21:
@hObBiT-iM-bLuTrAuScH could you verify what dshot_bitbang is set to?

hObBiT-iM-bLuTrAuScH, 2019-09-21:
# get dshot_bitbang
dshot_bitbang = AUTO
Allowed values: OFF, ON, AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8

Mateyhv, 2019-09-21:
What the luck bitbang is?  😂

mikeller, 2019-09-22:
@joelucid:

> This looks weird: dshot_bitbang isn't set in the diff so it should be set to default. So bidir should use bitbang - but the timer and dma allocation shows timer dshot being used.

I think this is working as designed - timers 1 and 8 are both already in use, so bitbanged Dshot isn't enabled:

> TIM1:
 CH1: SERIAL_TX 11
 TIM8:
 CH2: SERIAL_RX 12

hObBiT-iM-bLuTrAuScH, 2019-09-22:
> I think this is working as designed - timers 1 and 8 are both already in use, so bitbanged Dshot isn't enabled:

You mean i should set "dshot_bitbang = on" instead of auto ?

mikeller, 2019-09-22:
@hObBiT-iM-bLuTrAuScH:

> You mean i should set "dshot_bitbang = on" instead of auto ?

No, that won't help - bitbanged Dshot needs exclusive use of either timer 1 and timer 8, and in your configuration, both of these are used by one softserial each - we'll have to shift something else around to make this work.

mikeller, 2019-09-22:
@hObBiT-iM-bLuTrAuScH: Try `timer c7 AF3`. This _might_ solve the resource conflict.

joelucid, 2019-09-22:
@mikeller I’m surprised though that dma streams are then allocated like they would for timer dshot. Is this something you added?

hObBiT-iM-bLuTrAuScH, 2019-09-22:
I copied the motor ressource remapping etc and did the instructions. But i did not set:

"timer c7 AF2" does the trick. Got 5 beeps, can connect in blheli, and motors spin in betaflight.

 amazing work, but where did it come from. Im pretty sure i did not put this in with my settings.

hObBiT-iM-bLuTrAuScH, 2019-09-22:
> @hObBiT-iM-bLuTrAuScH: Try `timer c7 AF2`. This _might_ solve the resource conflict.

did the trick

hObBiT-iM-bLuTrAuScH, 2019-09-22:
also works with dshot600

hObBiT-iM-bLuTrAuScH, 2019-09-22:
5 beeps with RC4 after flashing without making any setting. dshot600, bidir = on as default. thread close?

mikeller, 2019-09-22:
@joelucid: I thought that was part of one of the changes that you added (adding an `AUTO` option for `dshot_bitbang`) - if `dshot_bitbang` is `AUTO`, and not enough timers / DMA for bitbanged Dshot can be found it will fall back to timer based Dshot.

joelucid, 2019-09-23:
@mikeller: not really. The ``AUTO`` option uses ``dshot_bitbang`` when ``dshot_bidir`` is on and timer dshot otherwise. There's no automatic fallback from ``dshot_bitbang`` to timer dshot if no timers are found.

mikeller, 2019-09-23:
@joelucid: Yep - you are correct. It should fail and disable all motor output.

joelucid, 2019-09-28:
Ok - disregarding the ``resource all`` output above which doesn't seem to have been taken with dshot bitbang active it seems this is now fixed? If so could the submitter please close it?

closed 2019-09-28 by hObBiT-iM-bLuTrAuScH