betaflight/betaflight #8904: Viva STM32 F4 05 - No uarts in the ports tab

Garthvan, 2019-09-17:
Problem with my Viva STM32 F4 05 Flight controller.

I'm in the ports tab and there are no uarts - does anyone know how to fix this?

IvoFPV, 2019-09-17:
@Garthvan Please provide a `diff`. Try reseting to defaults by typing `defaults` in CLI if you've flashed the unified target.

Garthvan, 2019-09-17:
@IvoFPV sorry I'm pretty new to this and don't know what a diff is? I've typed defaults in CLI and that rebooted it but the issue is still the same.

IvoFPV, 2019-09-17:
@Garthvan No worries. Type `diff` in the CLI as well, then copy everything that shows up and paste it here.

Garthvan, 2019-09-17:
Got it!

# diff

# version
# Betaflight / STM32F405 (S405) 4.0.6 Sep  1 2019 / 01:08:29 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name 
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

IvoFPV, 2019-09-17:
@Garthvan Okay, so it looks like you've flashed the STM32F405 unified target but the config for your board isn't applied. You have to do that manually after flashing.
Copy the settings below into CLI and hit enter, after all of it gets loaded type `save` and hit enter again. It should work after that.
```
# Betaflight / STM32F405 (S405) 4.0.0 Mar  7 2019 / 06:40:55 (fee4ee5e0) MSP API: 1.41

board_name VIVAF4AIO
manufacturer_id VIVA

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 C06
resource MOTOR 3 A10
resource MOTOR 4 A08
resource MOTOR 5 C08
resource MOTOR 6 B01
resource PPM 1 B15
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 A03
resource CAMERA_CONTROL 1 A05
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

# timer
timer B15 AF9
# pin B15: TIM12 CH2 (AF9)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer A05 AF1
# pin A05: TIM2 CH1 (AF1)

# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0
dma pin A08 1
# pin A08: DMA2 Stream 1 Channel 6
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin A05 0
# pin A05: DMA1 Stream 5 Channel 3

# master
set mag_bustype = I2C
set mag_i2c_device = 1
set baro_bustype = I2C
set baro_i2c_device = 1
set adc_device = 1
set current_meter = ADC
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set system_hse_mhz = 8
set max7456_clock = DEFAULT
set max7456_spi_bus = 3
set dashboard_i2c_bus = 1
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW0
```

Garthvan, 2019-09-17:
I acually can't thankyou enough! where do I find the board configuration if this happens to me again? and how I stopo this happening to me?

IvoFPV, 2019-09-17:
@Garthvan You can find all unified configs [here](https://github.com/betaflight/unified-targets/tree/master/configs/default). You just got to remember to paste the config after flashing. 
For version 4.1 which is releasing really soon config will be applied as a part of the flashing process so you won't need to worry about it then.

mikeller, 2019-09-18:
@Garthvan: Just use configurator and select the correct target for your board - STM32F405 is not the correct target for your board.

Garthvan, 2019-09-18:
@mikeller thankyou but now I'm not sure what the correct target is? Sorry I'm new to this what should have I flashed?

mikeller, 2019-09-18:
@Garthvan: You'll have to consult the documentation that came with your flight controller to find out what target you are supposed to use.

closed 2019-09-19 by Garthvan