betaflight/betaflight #8906: 4.1.0-rc3

S-T-A, 2019-09-18:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. No Gyro can't, calibrate acc no rx ports can't setup
2. tried re flashing
3. Matek411rx with xm reciver to tx1 sbus pads vtx wired to tx2
4.  configuration  10.6.0 and 4.1.0-rc3
5. """
# diff

# version
# Betaflight / STM32F411 (S411) 4.1.0 Sep 15 2019 / 09:13:50 (644ab5d90) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF411RX

# resources
resource BEEPER 1 NONE
resource MOTOR 1 NONE
resource MOTOR 2 NONE
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 NONE
resource LED 1 NONE
resource SPI_SCK 1 NONE
resource SPI_SCK 2 NONE
resource SPI_SCK 3 NONE
resource SPI_MISO 1 NONE
resource SPI_MISO 2 NONE
resource SPI_MISO 3 NONE
resource SPI_MOSI 1 NONE
resource SPI_MOSI 2 NONE
resource SPI_MOSI 3 NONE
resource ADC_BATT 1 NONE
resource ADC_CURR 1 NONE
resource OSD_CS 1 NONE
resource RX_SPI_CS 1 NONE
resource RX_SPI_EXTI 1 NONE
resource RX_SPI_BIND 1 NONE
resource RX_SPI_LED 1 NONE
resource RX_SPI_CC2500_TX_EN 1 NONE
resource RX_SPI_CC2500_LNA_EN 1 NONE
resource RX_SPI_CC2500_ANT_SEL 1 NONE
resource GYRO_EXTI 1 NONE
resource GYRO_CS 1 NONE

# timer
timer A03 NONE
timer B10 NONE
timer B06 NONE
timer B07 NONE
timer B08 NONE
timer A00 NONE
timer A02 NONE
timer A09 NONE
timer A10 NONE

# dma
dma ADC 1 NONE
dma pin B10 NONE
dma pin B06 NONE
dma pin B07 NONE
dma pin B08 NONE
dma pin A00 NONE
dma pin A09 NONE
dma pin A10 NONE

# feature
feature -RX_SPI
feature RX_PARALLEL_PWM

# master
set rx_spi_protocol = V202_250K
set rx_spi_bus = 0
set rx_spi_led_inversion = OFF
set current_meter = VIRTUAL
set battery_meter = NONE
set ibata_scale = 400
set beeper_inversion = OFF
set beeper_od = ON
set system_hse_mhz = 0
set max7456_spi_bus = 0
set cc2500_spi_chip_detect = ON
set gyro_1_bustype = GYROAUTO
set gyro_1_spibus = 0
set gyro_1_sensor_align = CW0
set gyro_2_spibus = 0

profile 0

rateprofile 0

# end the command batch
batch end"""

IvoFPV, 2019-09-18:
@S-T-A Type `defaults` in the CLI.

mikeller, 2019-09-18:
@S-T-A: There is no 10.6.0 release of configurator - use a released version.

closed 2019-09-18 by mikeller

S-T-A, 2019-09-18:
i just down loaded 10.6.0 rc1 thanks mike