betaflight/betaflight #8908: KAKUTEF7 No serial after flashing 4.1

parazyd, 2019-09-18:
**Describe the bug**
Serial port doesn't appear after flashing 4.1. Only DFU works when I connect with the bootloader button pressed.

This does not happen on 4.0.6 and there it seems to work fine.

It seems like this person has had the same issue on an AnyFC: https://github.com/betaflight/betaflight/issues/8056

**To Reproduce**
Steps to reproduce the behavior:
*  Flash BF 4.1 to a KAKUTE F7.
* Disconnect USB
* Connect USB and dropdown the serial devices

**Expected behavior**
The COM port should appear.

**Flight controller configuration**
None

IvoFPV, 2019-09-18:
@parazyd Are you using the latest configurator 10.6.0-RC1? Which board are you selecting when flashing? Full chip erase enabled?

parazyd, 2019-09-18:
I'm selecting KAKUTEF7 and the latest 4.1 RC.

Yes, I am doing the full chip erase, and using the latest configurator.

IvoFPV, 2019-09-18:
@parazyd Could you try to flash `KAKUTEF7 (Legacy)` and see if that works?

parazyd, 2019-09-18:
I don't see it. There is only `KAKUTEF7`, `KAKUTEF7MINI`, and `KAKUTEF7V2`.

IvoFPV, 2019-09-18:
It should be there, you have to first set the build type to `Release And Release Candidate` for it to show up.
[image]

parazyd, 2019-09-18:
Oh, you're right. I'm sorry.

I just flashed it and it seems to work now!

What is the difference between the Legacy and non-Legacy targets?

IvoFPV, 2019-09-18:
Legacy uses a board specific target just like it was before in 4.0, and non-legacy uses a unified target STM32F745 for your board and a separate config that gets applied on top of that. It should be the same at the end.
If you have some time to troubleshoot this issue and potentially find what is causing it to stop working, could you select STM32F745 from the dropdown and see if it still connects? Could you also confirm it stops working just after you flash, you don't get to connect to the board after flashing at all?

parazyd, 2019-09-18:
Flashed `STM32F745`. The same thing is happening.

It stops working immediately after flashing. And when I reconnect USB, I can't get the COM port to appear. Bootloader mode (DFU) works fine and as you can see, I can reflash the FC normally.

IvoFPV, 2019-09-18:
Are you possibly running KAKUTE F7 V2?

parazyd, 2019-09-18:
No. Kakute F7 AIO. On the PCB I can read v1.3.

IvoFPV, 2019-09-18:
Not sure then, that board looks like it is using a F745 MCU and that unified target should work. Someone else will have to take a look at this. For now just use the legacy target as that seems to work okay.

parazyd, 2019-09-18:
Alright. Please ping me if you need any more help with troubleshooting. I'd be glad to help out!

mikeller, 2019-09-19:
This is odd.

I verified this on KAKUTEF7 and on another F745 based board (FPVM_BETAFLIGHTF7), and 4.1.0-RC3 of STM32F745 wedges after flashing on both boards, the non-unified targets of the same version work fine.

Bisected with STM32F745, and the changeset after which this breaks is c33cc10 (#8759), which does not seem to contain anything that would explain this.

jflyper, 2019-09-19:
Verified to flash okay on Nucleo-F746ZG, and reboots fine when loaded with `KAKUTEF7.config`.

[betaflight_4.1.0_STM32F745-bootwedgefix_0.hex.zip](https://github.com/betaflight/betaflight/files/3631945/betaflight_4.1.0_STM32F745-bootwedgefix_0.hex.zip)

PR #8914 (2019-09-21): STM32F745 Do not define USE_SPI_DEVICE_5 and USE_SPI_DEVICE_6
Fixes #8908

STM32F745 target defines USE_SPI_DEVICE_5 and USE_SPI_DEVICE_6, but SPI5 and SPI6 are only available in packages >= LQFP-144, and these are not defined in bus_spi_pinconfig.c either.
Accessing spiInitDevice() with SPIDEV_5 or SPIDEV_6 would cause hard fault.
closed 2019-09-21 by mikeller