betaflight/betaflight #8909: Additional MSP enables arming even when connected to Configurator

spatzengr, 2019-09-18:
When turning on MSP on an additional Uart (like you do with the DJI Air Unit) it allows the quad to arm on the bench when plugged into USB and a battery is connected.  This is especially dangerous as you need LiPO power to power the Air unit, so if a user adds LiPO power to see the AUX switches move while setting up the Modes in the Modes tab, they could inadvertently arm the quad.

etracer65, 2019-09-18:
I tested and don't think this is just "turning on MSP on multiple UARTS". I could not duplicate the problem like that. I think it's something specifically to do with the DJI air unit communicating over MSP. Either it's interfering with the arming disable message sent by the Configurator when connecting, or it's actively sending the message to enable arming.

spatzengr, 2019-09-18:
I tested on my iFlight 120RS.  I was able to enable MSP on Uart#2 and arm it.

etracer65, 2019-09-18:
Then you have something strange going on with your config or procedure. If you enable the safety switch on the motors tab, then yes you can arm. Verify the that log at the top of the configurator shows an "Arming disabled" message when connecting. Also a settings `diff` would be useful.

mikeller, 2019-09-19:
@spatzengr: Not sure if this is a Betaflight problem, or the problem of a broken peripheral. Can you please disconnect all peripherals that are connected to UARTs that have MSP enabled (supplying the information you were asked to provide when opening this issue would help with figuring out which ones), and then check if you can replicate the problem?

spatzengr, 2019-10-14:
This issue no longer occurs on RC6 with latest 10.6 configurator.

jflyper, 2019-10-14:
@spatzengr Really? I believe it is not a configurator problem.

I think #9040 is required to fix this issue.

jflyper, 2019-10-14:
@spatzengr 

Can you execute this procedure to see if your "safe" combination is still safe?

1. Connect to the FC from configurator.

2. Connect the DJI FPV goggles and make sure it's talking to the FC.

3. Enter AVIN mode on goggles.

4. Exit from AVIN mode on goggles.

spatzengr, 2019-10-14:
@jflyper , OK.  So here is the trick.  IF I'm plugged into the Configurator FIRST, then I power the quad with a LiPO and turn on the goggles and TX, I CAN arm the quad while plugged in.  
HOWEVER, if I power up the quad first from the LiPO and then plug in to the Configurator and connect, I CANNOT arm the quad.

I did try entering and existing AVIN mode in the latter to see if I could get it to arm and nada.

jflyper, 2019-10-14:
@spatzengr 
Da**

```

PR #9040 (2019-10-16): Support per MSP client arming disable
Fixes #8909

~PoC code for supporting~ Implements "Per MSP client arming disable control".

`MSP_SET_ARMING_DISABLE` has been used exclusively by configurator to prevent accidental arming by sending `MSP_SET_ARMING_DISABLE(disable)` upon connection to an FC, and releasing the disable state by sending `MSP_SET_ARMING_ENABLE(enable)` upon disconnection from the FC, thus providing "The craft does not arm while connected from the configurator (except when explicitly instructed)" property.

However, recent introduction of non-configurator MSP client such as an  [... cut]
closed 2019-10-16 by mikeller