betaflight/betaflight #8911: Turtle mode activating and deactivating during flight

psim, 2019-09-18:
**Describe the bug**
When flying, turtle mode randomly turns on and off

**To Reproduce**
Steps to reproduce the behavior: look at bbl file

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
Mamba F722
Frsky R-XSR

**Additional context**
Add any other context about the problem here.

[btfl_001.zip](https://github.com/betaflight/betaflight/files/3628105/btfl_001.zip)

IvoFPV, 2019-09-18:
>Create a diff and post it here in a code block. Put ``` (three backticks) at the start and end of the diff block (instructions on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

etracer65, 2019-09-18:
Turtle mode can't activate in flight. If it did you'd immediately crash. Your radio channels may be fluctuating causing them to get into the area where the mode is set to activate and that would get recorded.

Please provide the information requested.

etracer65, 2019-09-19:
You are using Aux 5 for your crash-flip mode switch. This is receiver channel 9. Your receiver appears to be outputting RSSI on channel 9. So whenever your RSSI gets in the selected range it's triggering the crash-flip mode (which fortunately does nothing while already armed).

From your blackbox. Note how whenever the RSSI climbs to high values it enables and when drops it disables. The solution is to use a different aux channel.

[image]

jflyper, 2019-09-19:
lol

Mateyhv, 2019-09-21:
LoL

closed 2019-10-31 by stale[bot]