betaflight/betaflight #8912: Battery low indicator in the configurator

Etruscian, 2019-09-18:
**Is your feature request related to a problem? Please describe.**
When working on the bench, I usually have a battery connected. I have set up all the settings in betaflight so that it watches my battery voltage. Since the configurator does not show these warnings, but does show a green battery bar if a battery is connected, I've almost killed a lipo because I watched the green bar instead of the voltage.

**Describe the solution you'd like**
I would like the green battery bar to turn red when the connected battery is almost empty when attached to the configurator. Maybe also a warning popup or something like that

**Describe alternatives you've considered**
I've thought about battery voltage meters but since I don't have one, I'd like to have this feature in the configurator/betaflight. Since I also didn't have a receiver connected, I had no other way of giving me a warning.

PS: I didn't know if this was the right place for this kind of feature request. If it isn't, please tell me where I should put it and I'll happily close this request

etracer65, 2019-09-18:
It is not recommended to have the battery connected while connected to the Configurator. Settings changes can result in some states that cause the motors to spin unexpectedly or uncontrollably. Connecting the battery while using the Motors tab to test motors or to check receiver function on the Receiver tab is okay if kept to a minimum.

etracer65, 2019-09-18:
[image]

I'm not really sure what more is needed.

Etruscian, 2019-09-18:
My proposition would be for the battery icon to flash red or something. If I'm testing my motors (checking rotation and things like that) or escs, my attention is not with my battery voltage. A flashing icon would draw my attention and prevent a potential dangerous situation.

IvoFPV, 2019-09-18:
I guess since we already have the voltage and cell detection in the configurator we could display a warning dialog or flash the icon when battery goes below set minimum battery cell voltage.

etracer65, 2019-09-18:
Moving this to the proper repository.