betaflight/betaflight #8919: Pasting diff doesn't update settings in CLI Configurator build #318 chrome OS

Gozzz, 2019-09-21:
Configurator build number 318 chrome OS on a mac.

When attempting to paste a diff in CLI it seems to quit after "defaults nosave" line. Pasting partial diff starting at #resources completes fine.

See attached screenshot.
[image]

mikeller, 2019-09-21:
Interesting - I was able to reproduce this with a CLRACINGF7, but on a MATEKF722 `defaults nosave` works just fine with the same Unified Target - this seems to be data dependent on the contents of the Unified Target configuration.

mikeller, 2019-09-21:
Update: I found the culprit: :drum: 
```
feature RX_SERIAL
```

No idea why, but this causes `defaults nosave` to hang.

mikeller, 2019-09-22:
Update 2:
- this seems to be only affecting some targets (STM32F7?) - I have not been able to reproduce this on SPEEDYBEEF4;
- this seems to affect all RX_ features - I've reproduced it whit RX_MSP as well, but the TELEMETRY feature is fine;

PR #8946 (2019-09-25): Fixed hanging on 'defaults nosave' when RX_SERIAL is enabled.
Fixes #8919.

The issue that this fixes is that the RX stack is initialised only on startup, based on the configuration at the time. But the RX stack and some other systems (telemetry) use bits of configuration (`RX_...` features, `serialrx_provider`) at runtime as well. When this configuration is changed at runtime, the resulting discrepancy can cause issues, including the lockup observed in #8919.

This fixes it by copying these bits of configuration into `rxRuntimeConfig` when the RX stack is initialised, and then using it from t [... cut]
closed 2019-09-25 by mikeller