betaflight/betaflight #8922: ESCs wont initialise when using 'feature ESC_SENSOR' if 'battery_meter = ADC'

Juzzle1, 2019-09-21:
If I enable '_feature ESC_SENSOR_', and set both _battery_meter = ESC_ and _current_meter = ESC_, everything works OK and I get current and voltage readings from the ESCs via the dedicated telemetry wire.

If I then set _battery_meter=ADC_, so that current sensing is via ESC and voltage now via ADC, the ESCs wont initialise and keep repeating the three BLHELI startup tones.
While in this state,  typing '_set battery_meter = ESC_' into the CLI, the ESCs immediately finish initialisation and start working properly.

To reproduce:
1) Enable ESC telemetry:
_feature ESC_SENSOR
set battery_meter = ESC
set current_meter = ESC,_

2) Notice that Voltage and Current readings are working via ESC telemetry,

3) Switch voltage monitoring to ADC
_set battery_meter = ADC_

4) Reboot FC and notice ESCs no longer initialise correctly

5) Switch back to ESC current sensing
_set battery_meter = ESC_

6) Notice ESCs immediately initialise

**Expected behavior**
It should be possible to use onboard voltage ADC when ESC_SENSOR is enabled

**Flight controller configuration**
```
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / MATEKF722 (MKF7) 4.1.0 Sep 17 2019 / 17:17:13 (0e79257) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722
mcu_id 003f00483237510431313935
signature 

# name: JUZZLE

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# feature
feature LED_STRIP
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon
beacon RX_SET

# serial
serial 0 1024 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led
led 0 3,4:W:AT:1
led 1 3,5:W:AT:6
led 2 3,6:W:AT:6
led 3 3,7:W:AT:6
led 4 7,4:E:AT:6
led 5 7,5:E:AT:6
led 6 7,6:E:AT:6
led 7 7,7:E:AT:6

# mode_color
mode_color 2 0 1
mode_color 2 1 1
mode_color 2 2 1
mode_color 2 3 1
mode_color 2 4 1
mode_color 2 5 1
mode_color 4 0 4
mode_color 4 1 4
mode_color 4 2 4
mode_color 4 3 4
mode_color 4 4 4
mode_color 4 5 0
mode_color 6 1 6
mode_color 7 0 7

# aux
aux 0 0 0 1650 2100 0 0
aux 1 2 0 1950 2100 0 0
aux 2 2 0 1575 1675 0 0
aux 3 2 0 1175 1300 0 0
aux 4 13 1 1450 1550 0 0
aux 5 13 1 1150 1225 0 0
aux 6 13 1 1950 2050 0 0
aux 7 15 1 950 1025 0 0
aux 8 15 1 1250 1325 0 0
aux 9 19 1 950 1250 0 0
aux 10 35 1 1925 2075 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 500

# rxfail
rxfail 6 s 1000

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = 15,6,29
set mag_hardware = NONE
set rssi_channel = 7
set dshot_idle_value = 0
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set battery_meter = ESC
set pid_process_denom = 1
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_distance = ON
set osd_warn_rssi = ON
set osd_rssi_alarm = 42
set osd_cap_alarm = 1100
set osd_alt_alarm = 400
set osd_vbat_pos = 2
set osd_rssi_pos = 14360
set osd_link_quality_pos = 57
set osd_remaining_time_estimate_pos = 14765
set osd_flymode_pos = 14754
set osd_g_force_pos = 14786
set osd_vtx_channel_pos = 20
set osd_current_pos = 14806
set osd_mah_drawn_pos = 14776
set osd_craft_name_pos = 2
set osd_altitude_pos = 448
set osd_warnings_pos = 14730
set osd_avg_cell_voltage_pos = 14743
set osd_battery_usage_pos = 14793
set osd_disarmed_pos = 14411
set osd_esc_tmp_pos = 14722
set osd_esc_rpm_pos = 257
set osd_rtc_date_time_pos = 4
set osd_core_temp_pos = 354
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set displayport_max7456_wht = 3
set name = JUZZLE

profile 0

# profile 0
set idle_min_rpm = 18

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 25
set thr_expo = 50
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: LED 2
A15: LED_STRIP
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: USB_DETECT
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2:
    CH1: LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: LED_STRIP
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 1
DMA2 Stream 3: MOTOR 2
DMA2 Stream 4: MOTOR 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 4

# 
```

**Setup / Versions**
 - Matek F722-STD
 - SPEEDIX GS35A running 32.6.6

Running test version 4.1.0 Sep 17 2019 / 17:17:13 (0e79257) provided by @etracer65 to fix VTX channel switch issue.

jflyper, 2019-09-21:
It doesn't reproduce on Matek F722 with 4.1.0-RC3.

etracer65, 2019-09-21:
A possibly important point that you failed to mention is that you're also using `dshot_bidir = ON` and `dshot_bitbang = OFF`. So you're using bidirectional dshot with the timer based approach.

etracer65, 2019-09-21:
So try turning off `dshot_bidir` as a test to see if you still have the same behavior. Also try `dshot_bidir = ON` and `dshot_bitbang = ON`.

Juzzle1, 2019-09-21:
OK, so I tired with dshot_bidir = off, and everything started working.

However, switching back to dshot_bidir = on (without changing any other settings) I can no longer reproduce the problem!!!

I'll do more tests when I'm back home later.
I'm not crazy, I have two identical quads doing the same thing!

Juzzle1, 2019-09-21:
OK so ADC setting is a red herring here, and dshot_bitbang has no effect either way/

Seems to be to do with 'beacon RX_SET' and starting up with no SBUS signal

If I power up my quad with the TX turned off, the beeper starts going and I get the problem.
If I switch off 'beacon RX_SET' everything is OK.

It's as if the beacon signal is getting to the ESC before DSHOT is initialised somehow.... does that sound possible?

Juzzle1, 2019-09-21:
To be clear. it's not the beacon sound I'm hearing when this happens, it's the three initial tones you hear when you power on a blheli esc, in a fairly random loop, as if the ESCs are repeatedly restarting.

jflyper, 2019-09-21:
Sounds like beacon command triggering ESC protocol recognition. Thoughts? @etracer65

etracer65, 2019-09-21:
I would guess that it might be sending DSHOT commands immediately on initialization and that might be interfering with the ESC protocol detection.

joelucid, 2019-09-21:
Yeah possible. I know that it works with JEsc so it might be a blheli32 issue.

etracer65, 2019-09-21:
With BLHeli32 32.6.6 with `dshot_bidir = on` (without bitbang even) I've been able to confirm some level or interruption with DSHOT commands being sent early in the initialization (like of the `beacon RX_SET` was active immediately). Haven't got the complete failure but had the 3 tones, pause then a complete initialization. Would have to do with the timing of when the DSHOT command happens to come through.

Let me play around with it a bit. Probably just need a quiet time after initialization to give the ESCs enough signal for protocol detection. I would guess that the inverted DSHOT detection in BLHeli32 is a little more sensitive.

etracer65, 2019-09-21:
@Juzzle1 Can you please try this version? It adds logic to prevent DSHOT commands during the ESC's initialization and protocol detection phase.

[betaflight_4.1.0_MATEKF722.hex.zip](https://github.com/betaflight/betaflight/files/3639107/betaflight_4.1.0_MATEKF722.hex.zip)

The test scenario is having the `RX_SET` beacon active at power connect. But also test the case when it's not.

Juzzle1, 2019-09-21:
@etracer65 looks like you nailed it...

I have tested as follows:

1) Using existing (problematic) firmware, configured betaflight so that the symptoms are produced every power on (beacon RX_SET, FrSky transmitter off)

Result: https://youtu.be/1jHZIEtLFuE

2) Flashed the new firmware (full chip erase disabled) and powered on, keeping everything the same as before except firmware.

Result: ESCs initialise correctly straight away.

3) Disable beacon RX_SET

Result: ESCs again initialise correctly.

I haven't done a flight test as it's late here, however with the new init logic, the motors respond in motors tab, and arming the quad props off appears to behave as expected.

I've repeated this test on another quad (exact same hardware / config) with the same result.

etracer65, 2019-09-21:
@Juzzle1 Great! Thanks for the testing. We're just waiting for some feedback from the BLHeli32 dev to get some official information on timing, but it looks like we're on the right track. Turns out this was just a generalized bug that's been around for a while (not a common use-case though). It's just that the new BLHeli32 32.6.6 that handles inverted DSHOT seems to require a longer protocol detection period so it's more likely to occur.

Juzzle1, 2019-09-21:
@etracer65 Thanks for the quick fix (again) - happy to test further if needed.

PR #8927 (2019-09-23): Prevent DSHOT commands during protocol initialization/detection
Fixes #8922 

Adds a fixed delay after the motors are enabled before DSHOT commands can be sent. Fixes a problem caused by sending DSHOT commands too soon and prevent the ESC from properly detecting the protocol.

The scenario is having DSHOT commands sent very early in the startup phase. Like for example having `beacon RX_SET` and the beeper mode active on power.

Currently the delay is hardcoded to 3 seconds which experimentally seems appropriate. But waiting for some feedback from the BLHeli dev for confirmation.

This is reall [... cut]
closed 2019-09-23 by mikeller