betaflight/betaflight #8923: Hardfault caused by hardfault handler.

hydra, 2019-09-21:
If a hardfault occurs during early startup, before the `leds` structure is initialized, via `ledInit` the code which attempts to enable LEDs in the hardfault handler will fail and cause writes to uninitialized memory locations.  As you can imagine, this *really* doesn't help when debugging hard faults.

See:

https://github.com/betaflight/betaflight/blob/master/src/main/fc/hardfaults.c#L128

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/light_led.h#L55

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/light_led.c#L92

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/light_led.c#L66-L82

Should we:
a) initialise LEDs earlier, then reinitialise after config is loaded?
b) not attempt to change LEDs if leds are not initialised?
c) something else?

In my case, the hard fault was occurring in code called by the Reset_Handler in persistentObjectInit which is *really* early in the startup, earlier even than than the code which zero-initializes data sections...

hydra, 2019-09-21:
It might also be worth moving the call to 'persistentObjectInit' to at least after the zero-initialised sections have been completed, so that at least `null` checking can work on the `led` structure content, which won't right now.

mikeller, 2019-09-22:
This is sort of a chicken and egg situation that we have with Unified Targets - we need a fair chunk of initialisation to happen to get to the point where we can access the configuration to configure the LEDs, and any hard fault during this initialisation will not have a good way to signal the problem.

I think the approach to have the hardfault handler check if the LEDs have been initialised, and not try to use them if they haven't is the right one - there is nothing we can do to make the LEDs work before we've read the configuration.

hydra, 2019-10-05:
Ok, noted.  When I get time I'll do a PR as per your preference.

hydra, 2020-07-15:
I won't be making a PR for this, someone else will need to implement a fix.

closed 2023-04-11 by hydra