betaflight/betaflight #8934: vtx_low_power_disarm does not set power to its lowest value

ElieLiabeuf, 2019-09-22:
**Describe the bug**
According to the documentation: https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings `vtx_low_power_disarm` should set vtx power to its lowest value. 

**To Reproduce**
* set your vtx-tables with at leat 2 power level
* set `vtx_low_power_disarm` to ON
* when powering your quad it will be on the second power level not the first one

**Expected behavior**
When setting `vtx_low_power_disarm` to ON, the first value of vtxtable powervalues should be used.

**Setup / Versions**
BF 4.1 RC4. Board and VTX not relevant.

**Possible solution
It seems that vtx-tables power level index start at 0 but the constant https://github.com/betaflight/betaflight/blob/master/src/main/drivers/vtx_table.h#L55 is set to 1.

ping @functionpointer

functionpointer, 2019-09-22:
Thanks for the report!
The power levels do actually start at 1 since vtxtables have been introduced. 0 means unknown now. So the constant is correct.

That said, i have managed to reproduce the problem. Not sure what the problem is yet. It worked fine last time i touched it in https://github.com/betaflight/betaflight/commit/ebf2dcdb0eabc0994794137e731284fc227511b1. Also pitmode via AUX switch seems to be broken. Ill look into it!

etracer65, 2019-09-23:
@ElieLiabeuf Can you please provide your `diff` as requested when you opened the issue. It provides important information useful when troubleshooting problems.

etracer65, 2019-09-23:
And I think the problem is here:

https://github.com/betaflight/betaflight/blob/a54c33b952e1d1af4c1aa313daee096c7ba71d0c/src/main/drivers/vtx_table.h#L52-L56

If a unified target was used (why the `diff` is important) then `USE_VTX_RTC6705` will be defined and the `VTX_TABLE_DEFAULT_POWER` will be set to 2 instead of 1.

We'll have to think about the best way to fix this. Wouldn't be a problem for the legacy targets because they wouldn't define `USE_VTX_RTC6705` unless they actually had the VTX onboard.

ElieLiabeuf, 2019-09-23:
Yes I used the unified target version and not the legacy one. Issue is updated with the diff. I can't post the ressources at the moment.

etracer65, 2019-09-23:
I just think the concept of setting the default power index to 2 for cases where there's and onboard rtc6705 is a legacy from pre-vtxtable and no longer valid. rtc6705 is using vtxtable now so the power index should default to the first entry just line any other VTX type.

The logic looks flawed anyway with setting `VTX_RTC6705_DEFAULT_POWER_INDEX` to 2. From what I can see this would set it to the maximum power - not the minimum. So the flaw in the low-power logic was considering that `VTX_TABLE_DEFAULT_POWER` was the minimum when it was not. It was only the default setting.

mikeller, 2019-09-23:
@etracer65: Not sure if these are 0 based or 1 based. Setting this to not the minimum value on some RTC6705 stems from the fact that some models have a built in power switch, and therefore the minimum power level is 'off'.

functionpointer, 2019-09-23:
`VTX_RTC6705_DEFAULT_POWER_INDEX` was set to 2 for consistency. On normal vtxes (SmartAudio, tramp, etc) the first entry (index 1) in the table sets the vtx to a low power state. However, in the case of rtc6705, it turns the vtx off (if supported by the hardware). This is a unique feature to the rtc6705, as normal vtxes don't support complete power-down.

By contrast, normal vtxes support pitmode (or at least their protocols do).
So here is my suggestion: Rework the rtc6705 driver to only feature two power modes: LOW and HIGH. Additionally, add pitmode support and turn the vtx off when pitmode is active.
With this solution, `VTX_RTC6705_DEFAULT_POWER_INDEX` can be set to 1 and thus the entire special case from vtx_table.h may be removed.

etracer65, 2019-09-23:
The `vtxtable` example for RTC6705 disputes this a little:
```
        "powerlevels_list": [
            {
                "value": 0,
                "label": "OFF"
            },
            {
                "value": 1,
                "label": "MIN"
            },
            {
                "value": 2,
                "label": "MAX"
            }
        ]
```
So setting to 0 implies "OFF", 1 should be low power, and the current default of 2 sets to high power.

etracer65, 2019-09-23:
Another option is to fix the pit mode processing for RTC6705 and make it actually work (by setting the power value to 0). The the `vtxtable` for it can be corrected to just have 1 = low, 2 = high and not include the "pit mode" value just like every other vtx protocol.

etracer65, 2019-09-23:
@mikeller 

> Not sure if these are 0 based or 1 based. Setting this to not the minimum value on some RTC6705 stems from the fact that some models have a built in power switch, and therefore the minimum power level is 'off'.

The "index" to the `vtxtable` power levels is 1-based. So 1 is the first entry in the table, 2 the second, etc. The power values themselves for RTC6705 (the value set in the register) is 0-based. So 0 is off, 1 = low, 2 = high.

The fundamental flaw is that we're treating the `vtxtable` and pit mode functionality differently between SA/tramp and RTC6705. For SA/tramp pit mode is handled outside the `vtxtable` power levels. For RTC6705 it's implicitly in the power levels and setting the entry to the one that causes a 0 power value results in pit mode (off).

So I think I have an idea on how to workaround this problem in the short term as a bug fix. But in the bigger picture we probably need to rework the RTC6705 driver to actually emulate a true pit mode and change the `vtxtable` concept for it to only include the low (1) and high (2) values.

functionpointer, 2019-09-23:
@etracer65 Ill see if i can rework the driver this evening.

etracer65, 2019-09-23:
@functionpointer That's really something beyond a bug fix that would normally be included after a feature freeze. So I think we'll need to go with the short term bug fix and then the driver rework can be added after release and given adequate time for testing.

etracer65, 2019-09-23:
@ElieLiabeuf Can you please try this version?

[betaflight_4.1.0_STM32F405.hex.zip](https://github.com/betaflight/betaflight/files/3642217/betaflight_4.1.0_STM32F405.hex.zip)

**IMPORTANT: make sure you flash WITHOUT doing a full chip erase.** Since this is a unified target we want to preserve your target level settings and a full chip erase would reset them. This is a special case where I'm only giving you the base unified target .hex file and not a full target config.

@functionpointer Could you try with a board that has a RTC6705? I don't have any. The branch is here if you need to compile another target:

https://github.com/etracer65/betaflight/tree/fix_vtx_low_power_disarm

ElieLiabeuf, 2019-09-23:
@etracer65 Yep it works.

Thank you to all of you !

functionpointer, 2019-09-23:
So i just changed the rtc6705 driver:
https://github.com/functionpointer/betaflight/tree/fix_rtc6705_side_effects
Unfortunately, the OSD display of the power level still doesnt work and i suspect that vtx_low_power_disarm isnt working either. Still not sure. Ill investigate and fix. Once i am done i will post a PR.

Jetrell, 2019-09-24:
I was helping a friend over the phone upgrade his FuryF4OSD. And he is having the same `vtx_low_power_disarm` issue with a TBS Unify Pro 5.8 V3. 
This VTX has -25 200 500 800mw.
 200mW now becomes the default `vtx_low_power_disarm` value.

I'm not having this issue with mine and we have the same hardware. So I sent him my VTX tables to paste into his CLi and nothing changed with his...problem remained.
But I'm using Sept, 6th development firmware #1859

So we then proceeded to flash his with the same firmware as I am using #1859. But his quad still had the same issues with the VTX.

His `diff` files are the same as mine in the most part, because the quads are the same type.
So the only difference that we didn't try was the Configurator. I used a Jenkins Configurator build from around the same time as the firmware. 
While we used 10.6.0 RC1 when loading his.

PR #8936 (2019-09-25): Fix VTX low power disarm when USE_VTX_RTC6705 is defined
Fixes #8934 

Affected all unified targets as USE_VTX_RTC6705 must be defined to allow supporting that VTX option. Would cause the low power index settings for RTC6705 to override the settings for other VTX type and set the `vtxtable` power level index to 2 instead of 1 during low power disarm.

Ideally the RTC6705 driver should be updated to handle pit mode properly and not require an implicit power off state by adding a 0 power value to the `vtxtable` power levels.
closed 2019-09-25 by mikeller