betaflight/betaflight #8948: After flashing BF4.1 on my MatekF411RX FC any RC (tested 1-4) the motors spin like crazy if controlling via transmitter (Frsky x9d)

markusbrand, 2019-09-24:
**Describe the bug**
My motors get crazy after connecting my transmitter (Taranis x9d+) via SPI frsky_X and turn on arm switch. If I raise the throttle, they spin faster and faster - if I try to lower throttle, they don't spin slower anymore, if I do roll, pitch ... they go crazy with full speed and the only solution is to disarm.

**To Reproduce**
After flashing BF4.1 RC4 and setting it up properly with RPM filters enabled (my ESCs are DYS Aria 32 bit flashed with 32.6.6, default configuration, only two motors reversed). I test the motors in the motors tab - raise and lower the bar, the motors are spinning as they should, also lowering the bar, slows down the motors. No error packets shown - so everything as it should be. I go to CLI and bind my transmitter with the command "bind_rx = true" -> Transmitter binds - everything as it should be. I set up the arm switch in the modes tab and after disconnecting I try to use the quad with my transmitter. Activate the arm switch -> motors are spinning. I raise the throttle, motors are spinning faster. I lower the throttle -> nothing happens. motors are still spinning with full speed. I use pitch, roll -> the motors spin like crazy. lowering the throttle doesn't help -> I can only disarm to switch the motors off. 

**Expected behavior**
lowering the throttle should reduce the spinning of the motors. Also pitch and roll act like crazy.

**Flight controller configuration**
```
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F411 (S411) 4.1.0 Sep 22 2019 / 13:04:57 (a54c33b95) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF411RX
manufacturer_id MTKS

# name: ChameleonF4

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 13 0 1700 2100 0 0
aux 2 30 0 1300 1700 0 0
aux 3 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 7 16 25 40
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 300
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_hardware = NONE
set acc_calibration = -74,-64,-358
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set fpv_mix_degrees = 22
set blackbox_device = NONE
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set bat_capacity = 1500
set pid_process_denom = 1
set osd_vbat_pos = 14744
set osd_rssi_pos = 14362
set osd_tim_2_pos = 14732
set osd_mah_drawn_pos = 14721
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5732
set vcd_video_system = NTSC
set frsky_spi_tx_id = 65,122
set frsky_spi_offset = -93
set frsky_spi_bind_hop_data = 2,92,182,37,127,217,72,162,17,107,197,52,142,232,87,179,32,122,212,67,157,12,102,192,49,137,227,82,172,27,117,207,62,152,7,97,187,42,132,222,77,167,22,112,202,57,147,0,0,0
set frsky_x_rx_num = 3
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 1
set name = ChameleonF4

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10

rateprofile 0

# end the command batch
batch end
```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_CC2500_TX_EN
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_CC2500_LNA_EN
A14: RX_SPI_CC2500_ANT_SEL
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - FC: Matek F411RX with frsky receiver included (SPI)
 - VTX: TBS unify 5v 800mw
 - ESC: DYS Aria 32 bit (seperate ESCs)
- Motors: T-Motor F40 ProII 2400KV
- battery CNHL 4S 1500 mah
- also using a capacitor on the main battery lead

**Additional context**
never had this problem before - I used BF for a very long time already. Only appeared in 4.1. 4.0.6 was fine (also using RPM, but of course different blheli version: 32.64). Same quad since approx. 2 years (no hardware change).

IvoFPV, 2019-09-24:
@markusbrand Props on or off?

markusbrand, 2019-09-24:
Props off of course otherwise it would damaged a lot 😀

IvoFPV, 2019-09-24:
@markusbrand Take a look at this video...
https://www.youtube.com/watch?v=xBaefeQtA1c

markusbrand, 2019-09-24:
Thanks a lot for sharing! Got your point now. I'll immediately try tomorrow
outside with props, currently dark outside. I'll keep you posted!

joelucid, 2019-09-24:
Just sounds like wind-up, yes. One thing I notice is that the default of motor_poles hasn’t been changed. The f411rx is often used for micro builds around 3-4” where most motors have 12 poles. Not sure what your setup is.

markusbrand, 2019-09-24:
Hi, thanks for your help! It's just motor wind up - I did a stupid beginner mistake. The quad is 5" and motor pole of 14 is correct. I just did a hover test and everything works as expected! Keep on going with your great work on Betaflight!

closed 2019-09-24 by markusbrand