betaflight/betaflight #8950: Kakute F7 Mini -- crazy barometer

olegklimov, 2019-09-25:
I flashed 4.1.0-RC4 on my Kakute F7 mini board, using 10.6.0-rc1 configurator. It asked "load custom defaults" and I pressed yes.

Everything seems to work (great job with unified targets, I know this kind of changes can be painful).

But, the barometer is crazy. It reports 80 meters and then suddenly zero. I was previously using betaflight 3 series, barometer worked ok.

[image]

To reproduce: just go to sensors.

Please tell how I can help, I can try different hardware configs / hardware ports / IRQs / whatever and produce a patch. I need some kind of starting point, what do I play with to try to fix it?

olegklimov, 2019-09-25:
I tried Legacy version of the firmware for my board.

Target: HBRO/KAKUTEF7MINI
Firmware: BTFL 4.1.0

The barometer is still crazy:

[image]

So this doesn't seem to be the problem introduced by unified targets.

olegklimov, 2019-09-25:
Hey @mikeller can you please direct me to "betaflight hardware troubleshooting guide" of some kind so I can fix it and come back with a patch, not a bug report?

mikeller, 2019-09-25:
@olegklimov:

It's probably easier if you post the information that you were requested to post when opening this:

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

olegklimov, 2019-09-25:
4.1.0-RC4 (baro doesn't work, the rest works)
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: LED 1
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: USB_DETECT
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: SERIAL_RX 2
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: SERIAL_RX 7
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM2: FREE
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: MOTOR 3
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

3.5.7 (baro works, I reflashed and checked again)
```
# resource
resource BEEPER 1 D15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 E09
resource MOTOR 4 E11
resource MOTOR 5 C09
resource MOTOR 6 A03
resource PPM 1 E13
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 A02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 4 E06
resource ESCSERIAL 1 E13
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource FLASH_CS 1 A04
resource OSD_CS 1 B12
resource SPI_PREINIT_IPU 1 E04
resource SPI_PREINIT_IPU 2 A04
resource SPI_PREINIT_IPU 3 B12
```

"how are the different components wired up" -- nothing unusual, SBUS from DJI FPV, a GPS unit. Both work, I can control the drone and I can see 3d fix from GPS. Seems unlikely something interferes with the barometer, because 3.5.7 works fine.

jflyper, 2019-09-25:
Tested with Nucleo-F746ZG, with Adafruit BMP280 breakout module.
`baro_i2c_address` was set to `119` as Adaruit's module pulls up SDO to VDDIO.

BMP280 seems to work in both legacy and unified firmware.

4.1.0-RC4 (legacy)
```
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=46degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3613, Max available config: 32768
, BARO=BMP280
System Uptime: 93 seconds, Current Time: 2019-09-25T07:47:41.474+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 33, System rate: 9
Voltage: 913 * 0.01V (3S battery - CRITICAL)
I2C Errors: 4
Arming disable flags: NOGYRO RXLOSS CLI MSP

# diff

# version
# Betaflight / KAKUTEF7MINI (KF7M) 4.1.0 Sep 25 2019 / 16:20:33 (064ca9b75e) MSP API: 1.42

# start the command batch
batch start

board_name KAKUTEF7MINI

# master
set baro_i2c_address = 119

profile 0

rateprofile 0

# end the command batch
batch end
```

4.1.0-RC1 (unified)
```
# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=45degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3799, Max available config: 32768
, BARO=BMP280
System Uptime: 84 seconds, Current Time: 2019-09-25T08:03:42.235+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 33, System rate: 9
Voltage: 911 * 0.01V (3S battery - CRITICAL)
I2C Errors: 6
SD card: Not configured
Arming disable flags: NOGYRO RXLOSS CLI MSP

# diff

# version
# Betaflight / STM32F745 (S745) 4.1.0 Sep 25 2019 / 16:56:35 (064ca9b75e) MSP API: 1.42

# start the command batch
batch start

board_name KAKUTEF7MINI
manufacturer_id HBRO

# master
set baro_i2c_address = 119
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end
```

Sensor graph with 4.1.0-RC4 (unified)
[image]

jflyper, 2019-09-25:
Above test shows that I2C, barometer sensor code and BMP280 driver works.

I came to notice that the sensor itself is detected, but values read out by configurator is not good.

@olegklimov Can you supply `diff` output from 3.5.7 and 4.1.0-RC4, as well as `status` from both versions?

mikeller, 2019-09-25:
@olegklimov: I can tell you what the 'problem' is - this also goes a long way to show why it is important to supply the information you are asked for to help the people helping you.

Also, I have psychic powers today, and can tell you that you are living close to the sea (or in a very flat country). :stuck_out_tongue: 

> I can control the drone and I can see 3d fix from GPS. 

One thing that is a problem is that the graph in configurator has the wrong title, and says 'Barometer' where it should say 'Altitude'.

The second thing that changed between 3.5 and 4.1 is that the altitude calculation was changed to now also use information from the GPS, and more specifically, the altitude will be tracked as absolute altitude when disarmed if a GPS fix is available, but as relative altitude otherwise:

https://github.com/betaflight/betaflight/blob/064ca9b75e6fc20335ad13537df2bb26f40e6512/src/main/flight/position.c#L148-L163

So 80m - ish is your absolute altitude that you get when the GPS has got a fix, but then (and I assume you tested indoors) it loses the fix, and the altitude jumps back to the relative altitude from the barometer, and back again to absolute when the fix is obtained again.

olegklimov, 2019-09-26:
Yes, I can confirm -- without GPS lock it shows normal barometer readings!!! 4.1.0-RC5, not legacy version.

Probably you are very much used to Sensors being what it is in the code. From outsider's perspective (that I still have despite reading some `gps_rescue.c` code) sensors mean hardware. I visit that tab in the configurator to check if my sensors are working.

To test the barometer, I lift my quad up and down. If GPS is also working -- well I lose that ability to test.

It's great @mikeller you changed "Barometer" to "Altitude", now it's less confusing. From user's point of view GPS altitude is best viewed in GPS tab, and sensors tab should show raw sensors and have a quick way to test if it's working, in my opinion.

P.S. hah, I found that minimum distance for GPS rescue is 100m. That's why I crashed couple of times, right into the ground, quad was away maybe 70m. 😄

closed 2019-09-26 by olegklimov

mikeller, 2019-09-26:
@olegklimov:

> From user's point of view GPS altitude is best viewed in GPS tab, and sensors tab should show raw sensors and have a quick way to test if it's working, in my opinion.

Agreed. All that betaflight/betaflight-configurator#1690 is fix the discrepancy between name and contents - getting back to providing barometer data on the 'Sensors' tab is a change that will have to wait until after the upcoming release - at the moment there isn't even an MSP command to poll barometer readings - the existing command is MSP_ALTITUDE, and shouldn't be changed as this is also used by external OSD devices to poll the craft's altitude.

GrinFactor, 2022-07-19:
I'm having really strange gps issues with my quads using betaflight 4.3 and 4.4 nightly. Altitude is all over the place. Seems like its not updating fast enough