betaflight/betaflight #8951: Beeper Vifly V2 does not beep when actioning switch

frankitunes, 2019-09-25:
Hi and thanks for all the work done on bf!

ISSUE : The beeper does not beep when actioning the radio dedicated switch **(but the esc beacon are beeping correctly)**

I have **the same issue** on 2 different setup... 
1 - 5" frame based on Mamba F722 with a Vifly 2 under 4.0.6 (MAMBAF722 (MBF7) 4.0.6)
2 - 3" frame based on Mamba mini stack F411 with a Vifly 2 under 4.0.5 (FURYF4OSD (FY4O) 4.0.5)

I have connected the Vifly 2 according to the FC recommandation and setup betaflight as requested (diff all below for F722)

Wiring:
**FC  ->  Vifly**
Buzz+ -> Red (5V)
Buzz- -> Yellow (BZ)
Gnd -> Black (Gnd)

BUT the Vifly...
- is charging correctly
- beeps when I disconnect the LIPO
- beeps (1x, 2x or 3x) when I change the rate profile with the dedicated switch (as you can see in the diff all, I have adjrange for this rateprofil change)

Thanks a lot for your work.
Best,
Frank

McGiverGim, 2019-09-25:
Maybe I'm wrong but...
> beeper -RX_SET

This disables the beeper in a switch.

frankitunes, 2019-09-25:
> Maybe I'm wrong but...
> 
> > beeper -RX_SET
> 
> This disables the beeper in a switch.

Thanks a lot, thats it... I search ecerywhere except this and I asked on several forums...
Sorry for the disturbance.
Thanks again.$
Best!!!

closed 2019-09-25 by frankitunes