betaflight/betaflight #8954: CL Racing F7 V1.0 PitMode Switch Is Broken

azixaka, 2019-09-26:
I am on 4.1 non-legacy dev build 1892 and the PitMode that supposed to switch off VTX via the VTX Pit Mode switch doesn't work.

It works on all stable 4.0 versions including the 4.0.6

Tried this:

resource PINIO 1 A14
set pinio_box = 39,0,0,0
save

Also tried assigning it to 41 in case it's got something to do with the mode itself. No luck.

Not sure if it is fixed by this - https://github.com/betaflight/betaflight/issues/8943. I am going to try the latest dev build tonight.

azixaka, 2019-09-26:
Broken still. :(

Another thing I noticed while taking a diff is why is it based on the STM32F7X2 version? I thought X2 is for dual gyro boards only. The CLR F7 V1.0 has just one gyro.

diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Sep 22 2019 / 13:06:12 (a54c33b95) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF7
manufacturer_id CLRA
mcu_id 0043002a3237510e31313935
signature 

# name: babomish

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beacon
beacon RX_LOST
beacon RX_SET

# map
map ATER1234

# serial
serial 1 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 0 115200 57600 0 115200

# aux
aux 0 0 0 1400 1600 0 0
aux 1 13 3 1300 2100 0 0
aux 2 35 1 1700 2100 0 0
aux 3 39 2 1300 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 13 20 26
vtxtable powerlabels 25 100 400

# master
set gyro_to_use = FIRST
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_bidir = ON
set vbat_scale = 161
set small_angle = 180
set deadband = 2
set osd_vbat_pos = 2403
set osd_rssi_pos = 2421
set osd_tim_2_pos = 373
set osd_flymode_pos = 1
set osd_g_force_pos = 218
set osd_throttle_pos = 2412
set osd_current_pos = 355
set osd_mah_drawn_pos = 373
set osd_gps_speed_pos = 374
set osd_gps_lon_pos = 400
set osd_gps_lat_pos = 386
set osd_gps_sats_pos = 18
set osd_home_dir_pos = 27
set osd_home_dist_pos = 23
set osd_flight_dist_pos = 409
set osd_altitude_pos = 2
set osd_power_pos = 0
set osd_pit_ang_pos = 44
set osd_rol_ang_pos = 214
set osd_esc_tmp_pos = 288
set osd_esc_rpm_pos = 217
set osd_esc_rpm_freq_pos = 257
set osd_log_status_pos = 352
set osd_stat_rtc_date_time = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_flight_dist = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set vcd_video_system = NTSC
set pinio_box = 39,0,0,0
set gyro_2_align_yaw = 900
set name = babomish

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 181
set pitch_rc_rate = 181
set yaw_rc_rate = 181
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

Gozzz, 2019-09-30:
Mine is set to-
set pinio_box = 39,255,255,255
aux 2 39 3 900 1550 0 0

For me VTX is on by default, using a 2pos switch to disable it when needed. Works on build 1897 last time I checked. I can retest it today at lunch though to be sure.

Gozzz, 2019-10-03:
Have you tried this in cli @azixaka ?

set pinio_config = 1,1,1,1
set pinio_box = 39,255,255,255

azixaka, 2019-10-03:
I was so annoyed with many issues I had with this FC that I took it out of my racing drone and replaced with Matek F722 mini. Everything just works now and I am happy with it.

I do however still have this FC and it still has VTX connected to it. I will try your setting tonight and report the result.

hObBiT-iM-bLuTrAuScH, 2019-10-10:
just to keep this alive: i have the same issue on my CLRacing F7. Plus i can not save a pit frequency. also the OSD does not show the correct VTX Power. Does this have something to do with smartaudio 2.1? FC is outputting 2.1 signals. last checked on build 1908

ChenRiTianS, 2019-10-14:
DALRC F405      RUSH Tank
Pitmode is always on, ground station and osd modulation cannot exit,
only exit pitmode through the button on the image

mikeller, 2019-10-15:
@azixaka: betaflight/unified-targets#53 should fix it. Try flashing it again now.

closed 2019-11-21 by stale[bot]