betaflight/betaflight #8957: Smartaudio 1 on AKK vtx is broken (works on 4.0)

pmarkiewicz, 2019-09-26:
**Describe the bug**
No reaction from VTX after installation 4.1 RC5.
Works after installation 4.0
Issue with 2 VTxs one AKK and one Eachine (2 different machines, 2 different FC)

**To Reproduce**
No reaction of VTx after any changes to VTx settings on console after save.
vtx_low_power_disarm set to OFF

**Expected behavior**
VTx should change params 

**Flight controller configuration**
```
# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.1.0 Sep 22 2019 / 13:04:34 (a54c33b95) MSP API: 1.42

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id HBRO

# name: SOURCE ONE

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01

# feature
feature RX_SERIAL
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 4 1024 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# aux
aux 0 0 1 1300 1700 0 0
aux 1 13 0 1750 1850 0 0
aux 2 26 1 1450 1550 0 0
aux 3 32 0 1350 1450 0 0
aux 4 35 0 1550 1650 0 0

# adjrange
adjrange 0 0 2 900 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 7 16 25 40
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set mag_hardware = NONE
set fpv_mix_degrees = 30
set blackbox_p_ratio = 48
set blackbox_device = SPIFLASH
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set current_meter = ESC
set battery_meter = ESC
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_cap_alarm = 1500
set osd_vbat_pos = 14774
set osd_rssi_pos = 6602
set osd_link_quality_pos = 230
set osd_tim_2_pos = 14806
set osd_flymode_pos = 97
set osd_g_force_pos = 9
set osd_crosshairs_pos = 14541
set osd_current_pos = 6529
set osd_mah_drawn_pos = 2465
set osd_craft_name_pos = 6643
set osd_gps_speed_pos = 192
set osd_gps_lon_pos = 17
set osd_gps_lat_pos = 1
set osd_gps_sats_pos = 32
set osd_home_dir_pos = 47
set osd_home_dist_pos = 55
set osd_altitude_pos = 215
set osd_avg_cell_voltage_pos = 14786
set osd_disarmed_pos = 14635
set osd_esc_rpm_pos = 161
set osd_rtc_date_time_pos = 480
set osd_flip_arrow_pos = 2415
set osd_stat_max_spd = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 4
set vtx_channel = 3
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5780
set vtx_pit_mode_freq = 5780
set vcd_video_system = PAL
set gyro_1_align_yaw = 2700
set name = SOURCE ONE

profile 0

# profile 0
set p_pitch = 32
set d_pitch = 27
set p_roll = 30
set d_roll = 25
set d_min_roll = 14
set d_min_pitch = 15

rateprofile 0

# end the command batch
batch end

# 
```

```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 3
A03: MOTOR 1
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 2
B01: MOTOR 4
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: LED 3
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: OSD_CS
B15: INVERTER 3
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_CS 1
C05: GYRO_EXTI
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: BEEPER
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

**Setup / Versions**
Kakute F4 V2 with VTx AKK FX2 (bought year ago)
Mamba F722  with Eachine (other copter)

**Additional context**
No special firmware version needed for BF 4 (RDQ)

etracer65, 2019-09-26:
That VTX is SA 2.0

ivkostic, 2019-09-27:
Is there a chance the VTX AKK & RDQ patch be added to BF4.1?

pmarkiewicz, 2019-09-27:
Can be closed, user error, looks like modern VTx are using smart audio 2.
This have to be documented somewhere to avoid confusion.

etracer65, 2019-09-27:
> This have to be documented somewhere to avoid confusion.

@pmarkiewicz I would suggest you contact the manufacturer and ask them to update the documentation for their product.

closed 2019-09-27 by etracer65

mikeller, 2019-09-28:
@ivkostic: The AKK / RDQ patch will be available for 4.1 releases (not release candidates) in configurator.