betaflight/betaflight #8958: D_min debug mode is not D_min, but DSHOT RPM

Cc2Gre, 2019-09-26:
D_min debug mode is not D_min, but DSHOT RPM

IvoFPV, 2019-09-26:
Please provide the requested information when opening this issue, paste a `diff`.
Are you saying it gets set wrong in the configurator?

Cc2Gre, 2019-09-26:
ok sorry !

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / OMNINXT7 (ONX7) 4.1.0 Sep 25 2019 / 14:01:12 (08e8afa09) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNINXT7
mcu_id 002600423136511136323831
signature 

# feature
feature GPS

# beeper
beeper -ON_USB

# beacon
beacon RX_SET

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 4 0 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 1 900 1250 0 0
aux 1 1 2 900 1750 0 0
aux 2 13 3 1750 2100 0 0
aux 3 26 3 900 1250 0 0
aux 4 28 0 900 1250 0 0
aux 5 35 4 900 1250 0 0
aux 6 36 0 1750 2100 0 0
aux 7 49 4 1750 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 0
set gyro_notch1_hz = 225
set gyro_notch1_cutoff = 204
set gyro_notch2_hz = 118
set gyro_notch2_cutoff = 114
set dyn_notch_width_percent = 2
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 150
set acc_calibration = 18,-4,224
set align_mag = CW270
set mag_align_yaw = 2700
set max_check = 2000
set rssi_scale = 200
set rc_smoothing_auto_smoothness = 49
set serialrx_provider = FPORT
set serialrx_inverted = ON
set airmode_start_throttle_percent = 15
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set min_throttle = 1050
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = STAGE2
set align_board_roll = 180
set bat_capacity = 3000
set ibata_scale = 179
set small_angle = 180
set gps_provider = UBLOX
set gps_ublox_use_galileo = ON
set gps_rescue_descent_dist = 50
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_rssi_alarm = 60
set osd_cap_alarm = 2500
set osd_tim1 = 1280
set osd_vbat_pos = 2349
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2423
set osd_flymode_pos = 2382
set osd_g_force_pos = 2137
set osd_throttle_pos = 2232
set osd_vtx_channel_pos = 2389
set osd_current_pos = 2369
set osd_mah_drawn_pos = 2401
set osd_gps_speed_pos = 2209
set osd_gps_lon_pos = 2433
set osd_gps_lat_pos = 2449
set osd_gps_sats_pos = 2081
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2092
set osd_flight_dist_pos = 2113
set osd_altitude_pos = 2273
set osd_warnings_pos = 2250
set osd_avg_cell_voltage_pos = 330
set osd_pit_ang_pos = 2295
set osd_battery_usage_pos = 2409
set osd_esc_rpm_pos = 87
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set debug_mode = D_MIN
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5732
set gyro_rpm_notch_harmonics = 1

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 150
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set pid_at_min_throttle = OFF
set anti_gravity_gain = 15000
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set throttle_boost = 0
set p_pitch = 72
set i_pitch = 105
set d_pitch = 60
set f_pitch = 85
set p_roll = 70
set i_roll = 105
set d_roll = 60
set f_roll = 60
set p_yaw = 60
set i_yaw = 100
set d_min_roll = 46
set d_min_pitch = 48
set d_min_boost_gain = 45
set d_min_advance = 15

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 215
set pitch_rc_rate = 215
set yaw_rc_rate = 215
set roll_expo = 90
set pitch_expo = 90
set yaw_expo = 90
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
set tpa_rate = 30
set tpa_breakpoint = 1300

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

Cc2Gre, 2019-09-26:
[image]

IvoFPV, 2019-09-26:
@Cc2Gre Interesting, could you post that log?

Cc2Gre, 2019-09-26:
Apparently yes because when I chose d_min, it is confirmed in the configurator but it does not correspond to the curves and the title...

[BTFL_BLACKBOX_LOG_20190926_183356.zip](https://github.com/betaflight/betaflight/files/3660126/BTFL_BLACKBOX_LOG_20190926_183356.zip)

Cc2Gre, 2019-09-26:
[image]

IvoFPV, 2019-09-26:
You should build a newest version of blackbox explorer and test again.
[image]

Cc2Gre, 2019-09-26:
Ok ! thanks for your time.

Cc2Gre, 2019-09-26:
It works fine !

closed 2019-09-26 by Cc2Gre