betaflight/betaflight #8959: Matek F722-SE Unstable Take-off

goooodday, 2019-09-27:
I build with Matek F722-SE.
Up the throttle and the drone jump and spin.
Betaflight firmware : 4.0.6

Below is the dump file
------------------------------------
# version
# Betaflight / MATEKF722SE (MF7S) 4.0.6 Sep  1 2019 / 00:51:20 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name
name Goooodday

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A01
resource PWM 3 A00
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
[... 897 lines cut ...]
set thr_mid = 50
set thr_expo = 100
set rates_type = BETAFLIGHT
set roll_rc_rate = 60
set pitch_rc_rate = 60
set yaw_rc_rate = 60
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

Mateyhv, 2019-09-27:
Are you sure the board is correctly configured?

etracer65, 2019-09-27:
First, don't provide a `dump`, the instructions ask for a `diff`.

You're problem is caused by one or more of the following configuration errors:
1. Board/gyro alignment is incorrect. Make sure the 3D model movement on the Setup tab matched the real motions of the quad. Fix the alignment on the configuration tab if needed.
2. Motor order is wrong. Test each motor on the Motors tab (PROPS REMOVED!) and make sure the proper motor spins based on the graphic at the top.
3. Motor direction is wrong. One of more motors is spinning the wrong way. With the same test as in 2 make sure the each motor spins the way indicated in the graphic.
4. Props are on wrong. Upside down or normal/reversed props are on the wrong motors.

Lastly:
> **Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**
>
>Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

closed 2019-09-27 by etracer65

goooodday, 2019-09-27:
You're Answer is incorrect.

This problem is firmware filter error!

> First, don't provide a `dump`, the instructions ask for a `diff`.
>
> You're problem is caused by one or more of the following configuration errors:
>
> 1. Board/gyro alignment is incorrect. Make sure the [...]

mikeller, 2019-09-28:
@goooodday:

Please read this, and stop asking in here for help with your configuration:

> **Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**
>
> Please get help from other user support options such as asking [...]