betaflight/betaflight #8960: integrating blackbox logger feature in RunCam Device Protocol

ozzzp, 2019-09-27:
Hi, I have a Mobula7 HD cinewroop with the Caddx Turtle V2 1080p sd-recordable camera, and   is works fine with RunCam Device Protocol via two extern wire as the tutorial in the Betaflight wiki.

However, the FC, CrazyBeeF4FR, does not has on board memory. Hence I cannot get blackbox log on board and the extern openlog logger is too large and heavy for Mobula7 HD(There is a sd card in Mobula7 HD! why should we need another one?).

So, Could it possible to integratie blackbox log feature in RunCam Device Protocol, or It is already be and just need to open another soft-serial  with the same pad?

etracer65, 2019-09-27:
No, this is not possible. Your camera does not expose the SD card as a mass storage device first of all so the camera is not capable of this. And secondly Betaflight requires a direct SPI connection to the SD card and doesn't support writing to mass storage devices.

jflyper, 2019-09-27:
Two thoughts here.

1. If the craft can carry Caddx Turtle V2, external logger is at a fraction of the weight and size of that device. I would just tack the logger somewhere accessible, provided there is no way for me to do bb logging.

2. The serial protocol that Caddx Turtle V2 use (whatever that is), can be extended by Caddx so that the camera can record incoming serial bb log stream from at relatively high speed. The way the serial connection is mux'ed between camera control protocol and bb logging stream should be negotiated among stake holders including Runcam, Caddx and Betaflight devs, of course, to avoid unnecessary confusion.

@ozzzp I think you should talk to Caddx first if you want to utilize the SD card on Caddx Turtle V2. Like @etracer65 wrote, there's not much we can do at this point with the device.

ozzzp, 2019-09-29:
@etracer65 @jflyper Thanks for reply!

closed 2019-11-05 by stale[bot]