betaflight/betaflight #8964: CLRacing 4.1.0-RC5 OSD Broken

TheZeroBeast, 2019-09-29:
**Describe the bug**
OSD Chip not detected and isn't functional, good test is to try upload your font to the OSD which hangs indefinitely.

**To Reproduce**
Load 4.1-RC5 and attempt to use OSD. ie. try load fonts or see if it's overlaying its data.

**Expected behavior**
OSD works and allows configuration changes through the configurator.

**Flight controller configuration**
full chip erase no changes to initial flashing of 4.1-RC5

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - CLRacingF7

**Additional context**
OSD was working fine on 4.1-RC3 and when flashing back to stable 4.0 or 4.1-RC3 OSD is functional again, allowing font upload and actually overlaying on video signal.

jflyper, 2019-09-29:
@TheZeroBeast Please follow the instructions given.

TheZeroBeast, 2019-09-29:
which instructions would they be @jflyper?

TheZeroBeast, 2019-09-29:
Looks like A15 OSD Chip Select is no longer mapped maybe.

mikeller, 2019-09-29:
@TheZeroBeast:

> Looks like A15 OSD Chip Select is no longer mapped maybe.

Well spotted. Can you test if `resource OSD_CS 1 A15` fixes this?

TheZeroBeast, 2019-09-29:
Yes, doing now 😉

TheZeroBeast, 2019-09-29:
It doesn't fix the issue, I do not that it doesn't label A15 correct after assigning it.

```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: PINIO 1
A15: PREINIT 4
B00: LED 1
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: PREINIT 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

mikeller, 2019-09-29:
@TheZeroBeast: Sorry, I wasn't very forthcoming with information in my last comment. Resources are only allocated on reboot, so you need to do `resource OSD_CS 1 A15`, then `save`. The flight controller will reboot after this, and after the reboot it should work / show up in `resource show all`.

TheZeroBeast, 2019-09-29:
Yep, figured as much and still had no joy

mikeller, 2019-09-29:
@TheZeroBeast: Looks like there was a bunch of stuff missing in this Unified Target configuration. Try this:

[CLRACINGF7.config.txt](https://github.com/betaflight/betaflight/files/3667341/CLRACINGF7.config.txt)

TheZeroBeast, 2019-09-29:
Ah, that'd do it. Thanks @mikeller, I'll give this a go tonight and report back.

This would just need to be paste into CLI, then saved and rebooted, is that correct?

closed 2019-09-29 by mikeller

mikeller, 2019-09-29:
@TheZeroBeast:

> This would just need to be paste into CLI, then saved and rebooted, is that correct?

For testing yes.

But since betaflight/unified-targets#29 was merged you should be able to re-install the KAKUTEF7 target from the drop-down and it should work.

TheZeroBeast, 2019-09-29:
Ah ok, is the KAKUTEF7 equivelant to the CLRACINGF7? hardware connectivity wise of course

Docteh, 2019-09-29:
the linked Pull Request changed the config file for CLRACINGF7, give that one a whirl first.

mikeller, 2019-09-30:
@TheZeroBeast: Sorry, my bad. I had been testing KAKUTEF7 and got confused, I meant to say you should be able to install the CLRACINGF7 target from the drop-down.

TheZeroBeast, 2019-09-30:
Hey @mikeller, all good, just tested and it's working great now, thanks for the speeding fix. This unified target is proving damn handy.