betaflight/betaflight #8965: BF 4.1 RC5 Kakutef7 fatal SD card error

derekkeithchan, 2019-09-29:
I cant seem to get SD working for BBlogging.. says fatal card error. any idea why? using kakutef7 target

jflyper, 2019-09-29:
@derekkeithchan 

Please closely follow the instruction provided when you opened this issue.

And if you are using unified target and haven't read https://github.com/betaflight/betaflight/issues/8964#issuecomment-536297648, you might want to give it a try. If you are on legacy target, then forget about the potential fix mentioned in the comment in this url.

derekkeithchan, 2019-09-29:
Thanks, will get the rest of the info asap. What did you mean by give #8964 a try? I can't use CLracing F7 stuff for the kakutef7 can I?

derekkeithchan, 2019-09-29:
@jflyper included information needed

mikeller, 2019-09-29:
@derekkeithchan: I tried to replicate your problem with a KAKUTEF7 and exactly your configuration, but the SD card was recognised without a problem. Have you tried re-formatting your SD cards with the official tool (https://www.sdcard.org/downloads/formatter/)?

derekkeithchan, 2019-09-30:
yeah i cant believe i had 2 SD cards that had messed up formatting.. sorry for the trouble!

closed 2019-09-30 by derekkeithchan

mikeller, 2019-09-30:
@derekkeithchan: It's easy to do - unfortunately pretty all common operating systems allow you to format SD cards with on-board tools, and when doing so create a format that is not practical for writing streaming data to the cards (which is what blackbox and video / still cameras do). :frowning_face: