betaflight/betaflight #8966: MSP Arming flag override

fremebondo, 2019-09-29:
I use a separate microcontroller on board, reading attitude and rx data from FC via MSP serial link on UART2. In BF4.0.5 this kind of communication prevents the craft from arming. 

Is possible to add a flag to override MSP arming prevention?

etracer65, 2019-09-29:
The arming disabled is not triggered merely by having a MSP connection active. It's triggered explicitly by the Configurator when connecting to the firmware. So if you want to have MSP active for some other reason this won't cause the arming disabled MSP flag (unless you specifically make the MSP call to disable arming - which you won't).

fremebondo, 2019-09-29:
ok, thanks, I'll try that

mikeller, 2019-09-29:
@fremebondo: If you want to see some example code how MSP can be used to read telemetry in-flight, have a look at https://github.com/ShikOfTheRa/scarab-osd/tree/master/MW_OSD - this is exactly what external OSD devices do.

fremebondo, 2019-09-30:
@mikeller: Thank you!

ricardodeazambuja, 2019-10-03:
@fremebondo I've been controlling my drone using a RPI Zero W and a PS4 joystick, all based on MSP... therefore you should be able to do what you want without any problem. Here you can find an example of a very simplified user interface: https://github.com/ricardodeazambuja/BetaflightMSPy

closed 2019-11-09 by stale[bot]