betaflight/betaflight #8967: Dshot 1200

localidiot, 2019-09-29:
DSHOT 1200 missing from configurator 10.6.0 RC-2
BF 4.1 RC5

IvoFPV, 2019-09-29:
@localidiot It has been removed and is no longer supported. 

> Motivation for this change:
> - no benefits in using Dshot1200;
> - not or only poorly supported in common ESC firmware (BLHeli_S, BLHeli_32);
> - performance issues when using bitbanged Dshot with Dshot1200.

closed 2019-09-29 by etracer65

jcutcher01, 2019-09-29:
@IvoFPV thank you for posting the Motivation. I just ran into the same question and I agree with your summary.

yoodog, 2019-10-04:
Hi, any update when. stable will. be released Betaflight Configurator 10.6 and BF 4.1? I'm asking because both have Milestones completely Done and 4.1 was set for 01.10, but no announcement has followed. 

please advice as I'm about to set up new quad and may just wait few extra days

IvoFPV, 2019-10-04:
@yoodog 4.1 RC5 has been stable and you could use it right now if you are in a rush and update to stable when it releases. Nothing will probably change for final release other than a small fix for H7 boards that has been merged. I would say 4-5 more days.

hdrodel, 2020-08-30:
i was running a vivafpv 4in1 esc & flight controller with dshot 1200 on betaflight 4.0.* i upgraded to the newes Betaflight and tried using Dshot 600 which causes one motor to not spin. I also tried all the other dshot options. I can't get it to work. Multishot works normal. I had to flash emu flight now which works with dshot 1200 to be able to fly with dshot. how can i get dshot to work with betaflight again?

mikeller, 2020-08-31:
@hdrodel: This is not an issue of Dshot600 or Dshot1200 - Dshot1200 always worked worse and with a much higher error rate than Dshot600 on all common ESC firmware.
it looks like the target you are using for your flight controller is the wrong one, or some settings like bidrectional Dshot are not properly supported by your hardware.