betaflight/betaflight #8968: Jeti exbus issue 4.x.x on XROTOR FLIGHT CONTROLLER F4 G3.

Skym4x, 2019-09-29:
HI, Since 4.0.0 jeti receiver dosent work on any rx UART. I have try many solution but nothing work ( i have try pinio_config = 1,1,1,1  and that doesnt work too ).
With old Firmware, work fine , I have try on omnibus F4v3 and last firmware, works fine.
Sometime that can work but it's totaly randomly and very rare.
For  information, jeti exbus it's not reversed

**Flight controller configuration**
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.1.0 Sep 28 2019 / 23:17:44 (fcaa16cd7) MSP API: 1.42

# start the command batch
batch start

board_name OMNIBUSF4SD

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# serial
serial 0 64 115200 57600 0 115200

# master
set serialrx_provider = JETIEXBUS
set motor_pwm_protocol = MULTISHOT

profile 0

rateprofile 0

# end the command batch
batch end

Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

**Setup / Versions**
 - Flight controller HOBBYWING - XROTOR COMBO FC F4 G3 + ESC 60A
- Other components VTX : dji air system
 - sbus wiring : GND, 5V, UART RX1
- dji airsytem : RX6,TX6,GND,12V

**Additional context**
I have try all RX and same issue

Thank

mikeller, 2019-09-29:
@Skym4x: Looks like the flight controller you have is another badly made clone of the OMNIBUSF4SD. As these clones have various design deficiencies that require design specific workarounds it is really a matter of contacting your manufacturer / distributor and asking them to provide a workaround, or a refund for the defective part they sold you.

As a last ditch, can you try if `resource pullup 1 b8` fixes this for you?

Skym4x, 2019-09-29:
@mikeller you mean try this  ?: 
set pinio_config = 1,1,1,1
If its that, i have try yes but still issue :(.
I have buy a hobbywing offical product, it's possible that was a clone ??? and why it work fine in 3.5 with the same board ?
I hope we can find a solution without unmount everything :D.
Thanks :)

mikeller, 2019-09-30:
@Skym4x: No, I mean what I said: `resource pullup 1 b8`. I do not know where you got the `set pinio_config = 1,1,1,1` from, but this in itself will definitely never fix your problem.

Skym4x, 2019-09-30:
@mikeller could you explain how ressource pullup works ? what i have to do, maybe i can try :).
Thank

mikeller, 2019-09-30:
@Skym4x: What Hobbywing is doing is that they are cloning the OMNIBUSF4SD design, and as evidenced by the problems that you are having their clone contains some deviations from the original design - original OMNIBUSF4SD boards work just fine with 4.1.0.
Since Hobbywing does not provide us with schematics or engineering samples of their boards it is impossible for us to determine how these can be made to work, and you should really contact your manufacturer / distributor to get this fixed with any degree of certainty.

Resource pullup works by pulling a given pin (in your case B8) to logical high - it is nothing but a guess that this might fix your board as it seems to fix other faulty clones of the OMNIBUSF4SD board. To try it, enter `resource pullup 1 b8` in CLI, then `save`, and after the reboot check if it is working now.

Skym4x, 2019-09-30:
> What Hobbywing is doing is that they are cloning the OMNIBUSF4SD design, and as evidenced by the problems that you are having their clone contains some deviations from the original design - original OMNIBUSF4SD boards work just fine with 4.1.0.

Ok now i understand more, they cloning but dont show how they have do, nice play :D.
@mikeller have you some exemple of board which are no clone ? for me hobbywing was "official"...

I confirm the `resource pullup 1 b8` doesent fix the problem :(, the board still in : RX_FAILSAFEPARALYZE, and sometime on many reboot that work one time ...

It's very fustrating i think it's a little bootstrap will be fix this, but we dont have it :D.

Thank for your help 👍

mikeller, 2019-09-30:
@Skym4x: The OMNIBUS boards were originally developed and manufactured by Airbot. They don't make the original ones any more, but they have a good range of improved successors:

https://store.myairbot.com/flight-controller/omnibus-f3-f4.html

Skym4x, 2019-09-30:
Thank for link :)

Skym4x, 2019-10-02:
@mikeller I have test on an another board and I have the issue, have you someone who arrived to use Jeti with 4.X.X ? if yes on witch board plz ? 
Thank

Skym4x, 2019-10-02:
Ok a solution is found by my local fpv shop, with jeti we need to solder on TX pin af an UART and use télémétrie

mikeller, 2019-10-02:
@Skym4x: Sorry, this had slipped my mind, but there was a change to JetiEx in Betaflight before 4.0: #6767
@marv-t: This seems to be coming up regularly - do we have documentation for this change somewhere?

RoadyFPV, 2019-10-05:
@mikeller Yes here are the doc. 
https://github.com/betaflight/betaflight/blob/master/docs/Rx.md 
Under JetiEXBus

RoadyFPV, 2019-10-05:
@mikeller I have wrote a little pdf, maybe it will help?
[How to setup your FC-konvertiert.pdf](https://github.com/betaflight/betaflight/files/3692988/How.to.setup.your.FC-konvertiert.pdf)

mikeller, 2019-10-05:
@RoadyFPV: I am sure this will be helpful for users of the Rex3 struggling with the configuration. However, I think @Skym4x' problem has got to do with wiring up JETIExBus in Betaflight, and how this is done was changed in #6767, and I do not think we have documentation for this.

Skym4x, 2019-10-06:
Hi, Thank for documentation, @RoadyFPV , for information, that work with RSAT2 on EXT pin too.
@mikeller I still have problem with télémetry, on one board that work perfectly, on other board télémetry doesn't work ;( ( sometimes a values appear but flickering ). I have check the duplex mode and it's same for the two board. Have you some Idea for this ?
Thank for all :)

RoadyFPV, 2019-10-06:
@Skym4x thats right. I will revise the instructions and maintain a list with the receivers support the EX bus to which output. But this will take a while..

marv-t, 2019-10-07:
Please excuse the belated answer. I've just finished building our house and moving, so I did not have much time right now. 

The configuration of the receiver has already been described in the documentation. I will probably have to revise it again. 
The following Jeti receivers were tested: 
- R9 
- RSat 2 
- REX 3 
- REX 10 

Since the changeover, only the **_Tx_** pin of the uart is used. 
The compatibility with the resistance between Rx & Tx is still given. There may be a problem since the changeover. If the receiver is only connected to a Rx pin and no telemetry is used (no resistor between Rx & Tx), no receiveing is possible. I have already started writing a test to initialize that intercepts this possibility. 

@RoadyFPV: Can I incorporate your guide into the Betaflight documentation? 

@Skym4x: Unfortunately I do not have your FC to test it, although we have already tested some F4 and it worked.

@mikeller: The documentation is already available, but I will make it more detailed.

mikeller, 2019-10-07:
@marv-t: Thanks for your reply. Do you have a pointer to the documentation? I thought that you'd added it when you made the changes, but I could not find it.

marv-t, 2019-10-07:
@mikeller 
Here is the link: https://github.com/betaflight/betaflight/blob/master/docs/Rx.md#jeti-ex-bus

I think I have to be more specific about that.

mikeller, 2019-10-07:
@marv-t: Yes, I think for users with existing wiring wanting to upgrade to the new way of doing things it will be helpful if there is some more documentation. Thanks.

RoadyFPV, 2019-10-07:
@marv-t no problem you can use it. Here is a new file
[How to setup your FC and RX for Jeti.pdf](https://github.com/betaflight/betaflight/files/3698416/How.to.setup.your.FC.and.RX.for.Jeti.pdf)

RoadyFPV, 2019-10-20:
@marv-t should I change something on the document?

simored5, 2019-12-24:
Can you post a picture of how you have wired the jeti receiver to the hobbywing fc? Please

Skym4x, 2019-12-24:
No sorry, its closed in a cinewhoop drone and we dont see wiring ;(, but I need to change FC and receiver, at this time i will take a pict. Now I have a crossfire in the JETI radio, because I have some range issue with it.
By the way the wiring is more simple, 5V, GND and one of TX uart, active serial on this UART, cfg betaflight on EXBUS et your Jeti too, dont miss to use a compatible receiver, Rsat are perfect :).

RoadyFPV, 2019-12-26:
@marv-t I have revised the document in the area of ​​the R-Type Receiver

here is the new file
[How to setup your RX and FC for Jeti Ex-Bus.pdf](https://github.com/betaflight/betaflight/files/4002169/How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf)

simored5, 2019-12-28:
Hi. I read the file you provided. Thanks for that. One question. I am setting up my ds16 on an R3ex and when I go to the pin setup per your pdf I only have channel options as servo, digital input or digital output. I did go into the recieved general setup and change it to be on exbus though. 

Is there something I'm missing to be able to see the exbus option for the channel I'm using?

RoadyFPV, 2019-12-28:
@simored5 thanks for your feedback. Not all R Type recievers support Ex-Bus out.
What you can try is go to the JetiStudio and have a look for a new Firmware.
But i think that the R3ex dont support this. I have not many R-Type to test it.
I have contacted jeti about this and hope to get feedback soon. Then I can definitely revise the list again.

My advice is to use a REX3 for that. Its small and have full range. I use it on all my builds without any problems. When you have no space in your build you can pull the reciver out of the housing, desolder the pins and solder the wires direct to the pads and put a heatshrink over it. I have do this in my KX5 frame.

simored5, 2019-12-28:
@RoadyFPV 

Thanks for your reply. I updated my transmitter and receivers to the latest firmware.

I have a few different rex recovers. Unfortunately I dont have the R3 Rx you are using. Don't think it's available in Canada

Ive tried the Jeti Duplex EX R3/RSW 2.4GHz Receiver & RC Switch which is supposed to support jeti ex bus

http://www.espritmodel.com/jeti-duplex-ex-r3-rsw-2-4ghz-receiver-and-rc-switch-w-telemetry.aspx

I have the reciever plugged into the port 4 that the instructions note and the transmitter indicates also
EX Bus/Servo: 1-3 Servo, 4/Ext EX Bus/USB Programming

I have beta flight setup and UART 1 turned on

Sim with an rex r7 recieved. I get no option in the transmitter to change the port I'm using to exbus. With the r3/rsw reciever port 4 or the ext port does not appear in the ports under device explorer to change that specific port to ex bus.

I have an rex9 to I am going to try later. Any assistance in getting this figured out would be greatly appreciated.
Thanks again

Simon

mikeller, 2019-12-28:
￼@RoadyFPV:

> I have revised the document in the area of ​​the R-Type Receiver

I think this documentation will be helpful for Jeti users. Do you mind if we add it to the official Betaflight documentation?

simored5, 2019-12-29:
@RoadyFPV & all
After doing a lot of searching and reading about jeti exbus issues (not working) on BF 4. Something I found that others were finding heir setup worked on BF 3.5.7

I reverted back to that with the same setup and reciever, the r3/rsw and it works!

I'm now seeing that I have a connection and under the BF reciever tab I see things moving when the sticks are moved.

This is my first build, an armattan marmotte so I'm happy things are now working and am goin6 to carry on with this version for now.

If you are able to help resolve the issue in the latest release of BF I'll try again.

Thanks for your help so far. Much appreciated
Simon

RoadyFPV, 2019-12-29:
@simored5  
> I have a few different rex recovers. Unfortunately I dont have the R3 Rx you are using. Don't         think it's available in Canada

I use a REX3 not a R3ex! The names for the receivers at jeti are somewhat confusing.

> I reverted back to that version 3.5.7 with the same FC and reciever, the jeti r3/rsw/hobbywing    XRotorMicroF4G3 and it works!

Teke care in 4.0 you must Solder your signal wire on the TX Port of your UART like i discribed in the PDF.

> Sim with an rex r7 recieved. I get no option in the transmitter to change the port I'm using to exbus. With the r3/rsw reciever port 4 or the ext port does not appear in the ports under device explorer to change that specific port to ex bus.

You need the newest .bin files for the Transmitter to see the new options in the device explorer.
Normally the latest devices should be included when the transmitter is updated.

@mikeller 
> I think this documentation will be helpful for Jeti users. Do you mind if we add it to the official Betaflight documentation?

Sure thats no problem, i will revise the document as soon as I have received feedback from jeti about the EX-Bus option for the R-Type Recievers.

Should i take a close-up of the FC UART area for soldering?

mikeller, 2019-12-29:
@RoadyFPV:

> Sure thats no problem, i will revise the document as soon as I have received feedback from jeti about the EX-Bus option for the R-Type Recievers.

Sounds good, let me know when you've got the final version, and we can open a pull request to have it added into `/docs`.

> Should i take a close-up of the FC UART area for soldering?

If you think that this will be helpful for other users then that's a good idea.

simored5, 2020-01-05:
Hi. Hope this is ok but I have a slightly different issue with my jeti/hobbywing setup. Hope it's ok to add here rather than create a new issue.

So I have my quad complete and am going through the setup but I'm having an issue getting the quad  to arm. I'm trying to use the jeti throttle cut and assign a switch but in betaflight the arming setup does not seem to be recognized. Also I have freshly charged 6s batteries and the odd in BF keeps indidicating low voltage. 

Be been searching online for solutions but nothing yet. Hoping you can help.

Thanks
Simo

RoadyFPV, 2020-01-06:
@simored5 its not necessary to set up the jeti throttle cut in the transmitter.
Pls use only the Betaflight arming i use the prearm with a stick button. it works great.

When u have no movements u must setup your radio correctly.
See on page 3 in my .pdf

create a function on a free channel, define a switch u want and use it in BF.
Up to 16 Channels are supported

simored5, 2020-01-07:
@RoadyFPV  thanks for the reminder on the assignment of functions. 

I am getting some movement / indication that the radio is connected but the constant issues with battery voltage are really impacting progress.
I'm using fully charged 6s batteries that betaflight thinks are 8 cell and tried a fully charged 4s pack that BF thinks is 5 cell.

Looking online for various options to fix and gave tried calibration from a battery checker as well as using the options to change scale and divider voltage and I have the max v per cell at 4.3 not 4.2. 

I was able to get the 4s pack to finally show as a 4 cell pack in the power and battery tab but the odd still shows low voltage.

Hoping you might once again be able to help

Simodred5

RoadyFPV, 2020-01-07:
@mikeller i have a feedback from Jeti about the R-Type Receivers.

Here is the final version of my doc. You can check it and when its fine you can use it.

[How to setup your RX and FC for Jeti Ex-Bus.pdf](https://github.com/betaflight/betaflight/files/4031745/How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf)

simored5, 2020-01-10:
@RoadyFPV 

Wanted to let you know the radio is now working in the latest release. I've read the document you posted multiple times and it didn't hit me but I read it again today and having looked at this thread again it finally hit me that I had the receiver signal wire soldered to rx not tx. 

Installed the latest BF after modifying the soldered connect and jeti exbus is now working for me in the latest release.

I am having an issue with failsafe and signal loss but at least some progress has been made

Thanks

RoadyFPV, 2020-01-21:
@mikeller did you check out my new pdf? Should i change something?

mikeller, 2020-01-21:
@RoadyFPV: Sorry, I haven't had time yet - I will get around to it eventually.

mikeller, 2020-02-13:
@RoadyFPV: I finally got around to it: #9478

mikeller, 2020-03-15:
Documentation for this added in #9478.

Apart from this this seems to be a hardware issue.

closed 2020-04-23 by stale[bot]