betaflight/betaflight #8969: Unreadable Blackbox with SP Racing F3

fremebondo, 2019-09-29:
I am not able to get a fully readable BlackBox log on SP Racing F3

I tried logging on internal flash as on OpenLog and external SD as well, i always get unreadable files.
Usually BlackBox Explorer says that no flight can be parsed, only a few times it opens a log but only a fewpoints are shown, and they have no sense at all. 

I'm running on BTFL3.5.7 as I do not need 4.x.x features, and I do not want to lose led strip features.
I also tried running with all features disabled (UART2 too) and no luck with this issue.
However I also tried running 4.0.6 with all features disabled (UART2 too) and still no luck.

Attached my diff and a few logs (here FC is shaked on the workbench, but got identical result with real flights).
[3logs.zip](https://github.com/betaflight/betaflight/files/3667904/3logs.zip)

Please, do not scream for my "all zero" mixer... stabilized attitude is read from an Arduino Micro (via UART2) and some more mixing is performed to control a vertical take-off and landing plane. 

diff
# Betaflight / SPRACINGF3 (SRF3) 3.5.7 Mar 15 2019 / 22:21:31 (e9130527c) MSP API: 1.40

name FremeVTOL
resource MOTOR 7 NONE
resource LED_STRIP 1 A02
mixer CUSTOMAIRPLANE
smix 0 0 0 100 0 0 100 0
smix 1 1 1 100 0 0 100 0
smix 2 2 2 100 0 0 100 0
smix 3 3 0 100 0 0 100 0
smix 4 4 1 100 0 0 100 0
smix 5 5 2 100 0 0 100 0

feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature RX_SERIAL
feature MOTOR_STOP
feature LED_STRIP
serial 1 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
led 0 5,6::CO:2
led 1 5,7::CO:2
led 2 5,8::CO:2
led 3 5,9::CO:2
led 4 8,6::CO:5
led 5 8,7::CO:5
led 6 8,8::CO:5
led 7 8,9::CO:5
aux 0 0 6 1250 2100 0 0
aux 1 1 2 1300 2100 0 0
aux 2 13 4 1300 1700 0 0
aux 3 26 8 1300 1675 0 0
aux 4 31 8 1700 2100 0 0
set gyro_sync_denom = 2
set gyro_lowpass_hz = 50
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 10
set acc_trim_pitch = 4
set acc_trim_roll = -2
set acc_calibration = -139,625,-188
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set blackbox_p_ratio = 16
set motor_pwm_protocol = OFF
set align_board_yaw = 90
set current_meter = NONE
set vbat_scale = 10
set vbat_divider = 1
set deadband = 12
set yaw_deadband = 14
set pid_process_denom = 8
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 4900
set throttle_boost = 14
set acro_trainer_angle_limit = 50
set p_pitch = 51
set i_pitch = 3
set d_pitch = 6
set f_pitch = 71
set p_roll = 40
set i_roll = 3
set d_roll = 15
set f_roll = 105
set p_yaw = 102
set i_yaw = 5
set p_level = 25
rateprofile 0

set roll_rc_rate = 49
set pitch_rc_rate = 40
set yaw_rc_rate = 32
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 30
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set tpa_rate = 0
set throttle_limit_percent = 99

#

fremebondo, 2019-10-07:
Any help? Thanks

fremebondo, 2019-10-14:
The real problem is that I CANNOT DECREASE logging rate. I only can select 500Hz or Disable.

chupocro, 2019-10-14:
> 
> 
> The real problem is that I CANNOT DECREASE logging rate. I only can select 500Hz or Disable.

Yes, 500 Hz is the lowest value that could be selected in the configurator but I think there is some divider/denominator parameter for configuring the logging sample rate that can be set by CLI. There is **blackbox_p_ratio** parameter which might be the divider between something (maybe PID loop count) and the blackbox log count but I still didn't have time to test it. If that parameter is the divider then multiplying the parameter by **x** should lower the logging rate by **x times**. There is a use case for lowering the logging rate to only about 100 Hz or even lower - when you just want to record the stick positions to make video overlay. BTW, when you increase the PID loop frequency then you can select 1 kHz too.

mikeller, 2019-10-15:
@fremebondo:

Try `set blackbox_p_ratio = 4` in CLI.

closed 2019-11-21 by stale[bot]

fremebondo, 2019-12-05:
@mikeller , blackbox_p_ratio = 4 does not work. =256 or =1024 neither.
[bbl.zip](https://github.com/betaflight/betaflight/files/3928186/bbl.zip)

in the zip you will find p4 and p1024 bbls