betaflight/betaflight #8970: SP Racing H7EX Gyro Configuration Is Set To First

azixaka, 2019-09-30:
https://github.com/betaflight/betaflight/blob/4af1127d7ab59c0e3ad9552822e24de291b1c238/unified_targets/configs/SPRACINGH7EXTREME.config#L156

Should say BOTH

nerocide, 2019-10-01:
Looks to be corrected in :
```
# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep 28 2019 / 23:25:54 (fcaa16cd7) MSP API: 1.42
# get gyro_to_use
gyro_to_use = BOTH
Allowed values: FIRST, SECOND, BOTH
```

azixaka, 2019-10-01:
I am looking at this file in master - https://github.com/betaflight/betaflight/blob/master/unified_targets/configs/SPRACINGH7EXTREME.config

and it says FIRST still.

IvoFPV, 2019-10-01:
@azixaka That unified config is irrelevant in this case because there is no H7 unified target for now. If you flash your SPRACING H7 EXTREME with the latest version it will be set to use BOTH as @nerocide has noted. 
As you can see in the actual non-unified target which you are flashing it is set correctly. Please close this issue.
https://github.com/betaflight/betaflight/blob/fcaa16cd7d5e29fc9f9076293edcc264224cecd0/src/main/target/SPRACINGH7EXTREME/target.h#L161

azixaka, 2019-10-01:
We still want that issue to get fixed in the unified target, no?

IvoFPV, 2019-10-01:
Well not yet. Think about it... That config is **not used** as I've said therefore so what's the point in fixing that? It will get re-generated with new settings if unified H7 target becomes a thing. BTW configs currently in use are here and not in the directory you posted. https://github.com/betaflight/unified-targets/tree/master/configs/default

McGiverGim, 2019-10-02:
I'm not too sure about it... if the file is wrong, it must be fixed. Maybe someone uses it to make some tests.

azixaka, 2019-10-03:
If that config is not used and will get re-generated if it becomes a thing, lets remove this file set altogether then?

IvoFPV, 2019-10-03:
@azixaka Not sure why are you worrying so much about this file. It doesn't bother anyone. As mcgivergim says it could be that it is useful for some who are testing therefore it should stay.

azixaka, 2019-10-03:
No idea why I worry about it. Just love this FC so much that I try to keep it clean. :)

Closing the issue as no action needs to be taken.

closed 2019-10-03 by azixaka