betaflight/betaflight #8984: Set Spektrum_Sat_bind=9 - Invalid Name

fletch3d, 2019-10-02:
Eachine Lizard 95 with Omnibus controller
**To Reproduce**
Steps to reproduce the behavior:.  All worked fine, but switched to a new TX and needed to rebind.  Regardless of what firmware 4.0.0, 4.0.1 - .6 gives me the same result (Invalid Name)

**Flight controller configuration**
'''feature -RX_PPM
feature RX_SERIAL
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
map TAER1234
serial 0 1 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 1 900 1275 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1700 0 0
aux 3 4 1 900 1300 0 0
aux 4 13 2 1550 2100 0 0
aux 5 35 3 900 1700 0 0
set gyro_sync_denom = 2
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_cap_alarm = 550
set osd_tim1 = 768
set osd_vbat_pos = 2432
set osd_rssi_pos = 2426
set osd_tim_1_pos = 2368
set osd_tim_2_pos = 375
set osd_flymode_pos = 2458
set osd_throttle_pos = 322
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2400
set osd_craft_name_pos = 2440
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2408
set osd_disarmed_pos = 2252
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set pwr_on_arm_grace = 0
set vcd_video_system = NTSC
profile 0

rateprofile 0

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80'''

Resource:

'''resource BEEPER 1 C15
resource MOTOR 1 B08
resource MOTOR 2 B09
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B07
resource MOTOR 6 B06
resource PPM 1 B04
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource LED 1 B03
resource TRANSPONDER 1 A08
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 B04
resource ADC_BATT 1 A00
resource ADC_CURR 1 A01
resource BARO_CS 1 A13
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 C14
resource OSD_CS 1 B01
resource GYRO_EXTI 1 C13
resource GYRO_CS 1 A04'''

jflyper, 2019-10-03:
The Spektrum bind feature (available at `FEATURE_CUT_LEVEL < 3`) is not available for Omnibus (F3) (target OMNIBUS), as the target is at `FEATURE_CUT_LEVEL = 9`.

fletch3d, 2019-10-03:
Not sure what that means? Is there no binding available?

mikeller, 2019-10-03:
@fletch3d: Yes. Unfortunately the OMNIBUS target is very short on flash space (due to the on-board OSD), and a lot of features including Spektrum bind have had to be removed to make the firmware fit. To my understanding the binding is only required once - so you should be able to flash older firmware, bind to your TX, and then flash the latest available firmware back.

etracer65, 2019-10-03:
It means there is not enough space available on the OMNIBUS F3 to fit the Spektrum bind feature in BF4.0. So either bind externally or temporarily use an older BF version and bind and then upgrade back to BF4.0.

fletch3d, 2019-10-03:
Thank you.  Worked.

closed 2019-10-03 by fletch3d