betaflight/betaflight #8986: Blackbox Tab -> Erase Flash Dialog Box Stucks

azixaka, 2019-10-03:
RC2 erase flash button shows dialog that stays in the erasing... state forever. FC is Matek F722 Mini and is using the unified target which I believe is the only target available now. If I cancel the dialog and check flash memory status it's empty and that is reflected by the progress bar showing full 32MB available as well.

BF 4.1 RC5

The same software and feature works fine on my FC CL Racing F7 V1.0 unified target that has built-in flash as well but it's 128MB rather than 32.

mikeller, 2019-10-03:
@azixaka: I tried to reproduce, but erasing the flash worked fine for me with a MATEKF722MINI (32 MB flash) running 4.1.0-RC5 with configurator 10.6.0-RC2. It was slow, ~6 MB took about a minute, so if your flash chip is full it might take ~5 minutes. If it really does not work, can you please post a `diff` of your firmware, or reset the firmware to defaults, and then try to erase the flash?

azixaka, 2019-10-03:
I didn't expect to wait that long as my CL Racing F7 takes just a second to erase, so that dialog disappears immediately.

I'll try waiting for 5 minutes to test the theory. But it didn't do it on 4.0.6 and 10.51 so it's a regression I guess?

etracer65, 2019-10-03:
I tested and can't replicate any problems. Takes the expected ~64 seconds to erase the 32MB flash. Same in 4.0.6 or 4.1 with legacy *or* unified target. Same whether erased on the blackbox tab or if done from the CLI (`flash_erase`).

etracer65, 2019-10-03:
And CLRACING F7 uses a different type of flash (NAND) vs. the MATEKF7-mini (NOR). So they're not comparable.

closed 2019-11-09 by stale[bot]