betaflight/betaflight #8993: auto-land : why the 30 second delay?

laurentopia, 2019-10-04:
When flying in mountainous or woody area the cause of rxloss is occlusion so auto-land with a high throttle value to bring the quad up high within LOS seems like the perfect way to recover RX

but

"Re-arming is locked until RC link is available for at least 30 seconds and the arm switch (if used) is in the OFF position."

Why those constraints?

IvoFPV, 2019-10-04:
>seems like the perfect way to recover RX without having to install a GPS

Until one day it failsafes when you don't expect it to and it flies high throttle into someones face. Sorry but this is really unsafe in my opinion. Get a GPS module, they are cheap and work great.

laurentopia, 2019-10-04:
> > seems like the perfect way to recover RX without having to install a GPS
> 
> Until one day it failsafes when you don't expect it to and it flies high throttle into someones face. Sorry but this is really unsafe in my opinion. Get a GPS module, they are cheap and work great.

The same can be said if auto land happens above people's head so I edited my post to focus on my question.

etracer65, 2019-10-04:
> Why those constraints?

To prevent cases where you lose signal, have a failsafe and start walking to pick up your quad. But then you forget to turn off your arm switch and while walking to pick up the quad your radio link comes back. Without the constraints the quad would arm.

laurentopia, 2019-10-06:
> > Why those constraints?
> 
> To prevent cases where you lose signal, have a failsafe and start walking to pick up your quad. But then you forget to turn off your arm switch and while walking to pick up the quad your radio link comes back. Without the constraints the quad would arm.

Got it, thanks.

closed 2019-10-06 by laurentopia