betaflight/betaflight #8994: Sliders (or similar multiplier concept) in firmware

fujin, 2019-10-04:
The new sliders are awesome but make field tuning via BF CMS (OSD) tricky- you can kind of do it by copying the profile, making tweaks in CMS, then plugging back into Configurator and  tweaking sliders in your original profile until you get values that are close enough. The relationship between parameters has to be well understood in this case.

It would be neat if the slider multiplier/calculations were done in firmware and exposed to the various interfaces: CLI, CMS, MSP and eventually Lua.

This will reduce drift in the outer interfaces as well as the calculation will be standard/centralised.

**Describe the solution you'd like**
* Slider calculation/algorithm/multipliers copied into firmware
* CLI for slider multiplier values
* BFOSD CMS interacting with slider multiplier values
* Eventually: MSP (maybe the multiplier and range so the UI can be built dynamically?), update Configurator, Lua.

**Additional context**
I have had a number of users suggest extending the slider/multiplier concept to Rates to shift all rates at the same time, by the multiplier value. This could be cool as well.

IvoFPV, 2019-10-04:
I am looking to implement this as previously discussed. Current configurator only based sliders we have for 4.1 will give us insight to how users are using the feature. I think it will be a good idea to wait a month after release to get all the feedback and information we need to build on top of the current implementation, possibly adding different sliders, adjusting ranges, scaling and so on. 
After that input and confirmed usefulness of the slider concept I will try to make everything available in the firmware as you say. Adjusting in CMS, Lua is important and will improve usability very much. It will have a lot of time to get further tested by guys running nightly builds so it is ready for 4.2

fujin, 2019-10-04:
@IvoFPV I agree completely about waiting until after 4.1.0, just wanted to make sure we had an issue reflecting the plan. p.s. I love it.

etracer65, 2019-10-04:
The actual slider calculations need to happen in the firmware. The MSP would be multiple parts; initial ranges (get) and set new multiplier (set) (which causes the calculation to occur). The detailed values affected by the sliders are already available in MSP so those could be refreshed to show the results (numbers).

nerocide, 2019-10-05:
> The new sliders are awesome
I love it to

PR #9119 (2021-01-05): Implement tuning sliders, add to CMS, MSP
Fixes #9166 

Initial implementation of tuning sliders to make them available in firmware. This makes them easier to use because they are now available in CMS. Could be also made to work with Lua scripts.
Going to keep it as draft for now, because we should still wait for more feedback on how sliders are working on different setups people use and possibly change some things in the calculation.
Fixes #8994
Also, I'm sure some things could be improved in the code, firmware side is not my thing 😄

Things to improve:
- [x] Discuss and change calculation ba [... cut]
closed 2021-01-05 by mikeller