betaflight/betaflight #8995: BF 4.1RC5 + Typhoon32 v2.2 (32.7fw) - no volt or Amp, RPM data are ok!

yoodog, 2019-10-04:
I run Omnibus f7v2 on BF 4.1RC5 + Typhoon32 v2.2 (32.7fw). 

- RPM telematics in CLI are received with motors spinning, but Voltage and Amps data is not showing up.
- After a few test motor spin up Dshot lost packets are 0 in BlHeli
- ESC has 4 sensors all show 0v and 0A
- ESC was replaced recently together with update to new ESC FW and new BF, so may be HW issue, but seems unlikely, as there are no other indication of issues on any level so far.

I'm not sure where to place this - bug, config issue, or other place. I think there is some Configuration missing in f7v2 templet for telematics perhaps?

Diff ALL attached

# Diff ALL

# version
# Betaflight / STM32F745 (S745) 4.1.0 Sep 25 2019 / 14:09:35 (08e8afa09) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF7V2   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF7V2
manufacturer_id AIRB
mcu_id 0042003e3335511035303934
signature 

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 16384 115200 57600 0 115200
serial 2 4 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# aux
aux 0 0 2 900 1300 0 0
aux 1 1 0 1700 2100 0 0
aux 2 26 2 900 1300 0 0
aux 3 30 0 1300 1700 0 0
aux 4 33 2 900 1300 0 0
aux 5 36 1 1700 2100 0 0
aux 6 39 2 900 1300 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5808 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# rxfail
rxfail 4 s 1800

# master
set gyro_to_use = SECOND
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = 21,5,108
set mag_hardware = NONE
set rc_interp = MANUAL
set rc_smoothing_input_hz = 50
set rc_smoothing_derivative_hz = 100
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set fpv_mix_degrees = 25
set serialrx_provider = SBUS
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_delay = 6
set failsafe_throttle = 1130
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1300
set vbat_max_cell_voltage = 420
set current_meter = ESC
set battery_meter = ESC
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 45
set osd_cap_alarm = 1100
set osd_tim1 = 1537
set osd_rssi_pos = 2075
set osd_link_quality_pos = 2073
set osd_tim_1_pos = 2060
set osd_flymode_pos = 2093
set osd_crosshairs_pos = 2286
set osd_ah_sbar_pos = 2287
set osd_ah_pos = 2159
set osd_current_pos = 2528
set osd_mah_drawn_pos = 2509
set osd_compass_bar_pos = 2443
set osd_warnings_pos = 14698
set osd_avg_cell_voltage_pos = 2540
set osd_esc_tmp_pos = 2496
set osd_core_temp_pos = 2464
set osd_log_status_pos = 2048
set osd_stick_overlay_left_pos = 2208
set osd_stick_overlay_right_pos = 2231
set debug_mode = GYRO_SCALED
set vtx_band = 2
set vtx_channel = 2
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5752
set gyro_1_align_yaw = 900

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set throttle_limit_type = SCALE
set throttle_limit_percent = 70

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

etracer65, 2019-10-04:
To make use of BLHeli32 telemetry you must enable `feature ESC_SENSOR` and you MUST connect the additional telemetry signal wire from each ESC to a UART RX pin. Bidirectional DSHOT used for RPM filtering only includes RPM data.

yoodog, 2019-10-04:
I tried. I enable ESC Sensor in GUI - save - after reboot it is disabled. Not sure how or why it does that

yoodog, 2019-10-04:
I try CLI now, I took it as if it was not supposed to be enabled.

yoodog, 2019-10-04:
this is what I get 
get ESC_SENSOR
esc_sensor_halfduplex = OFF
Allowed values: OFF, ON

Set on?

etracer65, 2019-10-04:
https://oscarliang.com/esc-telemetry-betaflight/

closed 2019-10-04 by etracer65

yoodog, 2019-10-04:
> https://oscarliang.com/esc-telemetry-betaflight/

... Mate :-) not my first rodeo :)

I have had it working beautifully before on same FC + same ESC before it burned out in 400m crash.

- For some reason I can not Enable UART7 - ESC, which is required in order to be able to turn on ESC_Sensor in GUI. After saving UART7

yoodog, 2019-10-04:
> https://oscarliang.com/esc-telemetry-betaflight/

Not my. first rodeo, I had all working on same set up before with earlier BF and BH.  Now I could not set UART7 to ESC, which is required in order to activate ESC_Sensor tap in gUI. 

After esc_sensor_halfduplex setting ON, I was able to set UART7 to ESC. Not sure why. After that ESC_SENSOR tab saved and activated successfully. 

Thanks for input!