betaflight/betaflight #8997: Adjustments broken in 4.1RC5

haplm, 2019-10-05:
**Describe the bug**
I'm using two adjustment sequences to switch between rate profiles and OSD profiles:
[image]

Those worked just fine on 4.0, but on 4.1, it works only one way. That means that moving AUX5 into the middle position for the first time, Rateprofile 2 is selected. However, moving it back to low position doesn't switch back to Rateprofile 1. If I now move AUX5 to high position, Rateprofile 3 is selected, however, I can no longer go to either Rateprofile 2 or Rateprofile 1.

OSD profile has the same behaviour.

tarite, 2019-10-05:
Try using only one slider per switch with the entire range set.
That's how I have mine setup for osd.

haplm, 2019-10-05:
Thank you, @tarite! 

It works. Something must have changed, though, because my approach was working as well since 3.3 or so.

Do I get it correctly that BF has a hardcoded mapping of the channel range to first 3 rate profiles?

etracer65, 2019-10-06:
There's no hardcoded mapping exactly. The full range is just split evenly based on the number of options. So 3 equal parts for Rate Profile (or 6 if `rate_6pos_switch = ON`). The correct way to set it up is to use a single adjustment for the entire range - not 3 separate adjustments.

mikeller, 2019-10-06:
@haplm: What has changed is that the 'slot' facility was removed from adjustment ranges (#8369). This removes an unnecessary limitation from adjustment ranges, but it also means that your old configuration won't work any more, as it is not the correct way to get what you want.

haplm, 2019-10-06:
Understood, makes sense.

I have set it up (I believe) based on some ancient Blue Falcon's video a long time ago, and never really though much about it since... Thanks!

closed 2019-10-06 by haplm