betaflight/betaflight #8998: RPM filter amplifying harmonics on YAW axis on the ground

Mateyhv, 2019-10-06:
Idling on the ground when RPM filters are ON, quad harmonics go through YAW axis, strength being proportional to P term (P=100 in this test). Motors speed up immediately after start and began "fighting" on YAW axis creating huge vibrations at roughly 64Hz as if P term was too high. They dessapear if any arm is touched or if the quad is lifted but reappear on the ground.

Same configuration with RPM filters OFF and the motors spin smoothly on the ground with same high P term = 100 on YAW

The high P on yaw P is for ilustrating purposes, however my original setting was P=80 but since there were more differences in the setup the vibrations were much stronger then what is shown here.

```board_name FURYF4OSD
mcu_id 003d0045324d501820303256
signature 

# name: Mark II

# feature
feature -AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# serial
serial 0 0 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0

# vtxtable

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set fpv_mix_degrees = 25
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set gyro_rpm_notch_harmonics = 1
set name = Mark II

profile 0

# profile 0
set p_yaw = 100

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#```

```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
This happened on both quads with latest updates BF4.1, esc fw and so on: 1st quad showed here is MambaF4 on JESC firmware and Fport; 2nd quad SpH7Extreme, Mamba506 4in1 and Sbus.

I provide a video and a log of a factory default except a few settings to make it work like F.port. Fist RPM is on, then off , then on again. Log shows all three phases.

[Goolge Drive link](https://drive.google.com/open?id=18EBGBaKFDpnHsB-IKLPZSBjbmDAJjCTO)

IvoFPV, 2019-10-06:
It just looks like P oscillation caused by P being too high. When you touch it you attenuate the vibration so it stops. 
Now why does disabling RPM filter solve the issue is interesting. My guess would be it is just too much filtering causing latency creating a feedback loop therefore P grows and goes sine wave. 
I would try to disable gyro lowpasses and see if that helps, just as a test. Also does it happen if you are at higher RPM?
[image]

etracer65, 2019-10-06:
Why in the world would you set yaw P so high? Of course it's going to oscillate! And as stated, enabling RPM filtering without changing any other filtering will add a small amount of additional delay (just like adding any extra filtering). That extra delay along with the tendency to oscillate because of high P is likely just getting into a positive feedback loop.

So there's nothing about RPM Filtering itself that 's causing a problem. You would be able to produce the same effect without RPM filtering by just adding more filtering - like a notch or 2, or lowering the lowpass cutoffs. But the real answer is don't use high P gains for yaw.

joelucid, 2019-10-06:
One important piece of info missing in this issue is that the test was run without props. You should put props on, use a yaw p value close to what we recommend and see what happens when you try flying the quad.

Mateyhv, 2019-10-09:
Sorry for the late response and thanks everybody for the inputs. Gave me some good ideas.

@etracer65, P that is not randomly set, The quad was tuned so that gyro tracks the setpoint as close as possible. Lowering it is effectively detuning. Its nowhere high though, take a look at the recent YAW pid tunning youtube video from Mark Spatz where he is testing P as high as 200 on yaw with no ill effects.

@joelucid you are right, I forgot to mention the props, anyway its shown in the video. The issue was catched in the final preflight checks thus it never flew. Comming from rc helicopters anything that doesn't look right is not aproved for flight until solved. $$$ and safety in play with 1.5m diameter rotor.

@IvoFPV, thanks for the input, you nailed it! The problem appears to be too much filtering leading to excessive latency. For consistency the tests in this thread were done on default BF4.1 setup (filters, PID and rates) except yaw P and I raised to 100 and 90 respectively. **As soon as I disabled dynamic filters idle was smooth as silk** with those same settings. It might help to add a note in the wiki preventing people to fall in the same problem.

Since the issue is resolved I proceed to close it. Thanks everyone for the hard work!

closed 2019-10-09 by Mateyhv

IvoFPV, 2019-10-09:
@Mateyhv Even though P for you is making it track better it will introduce a lot of issues at high gain values. I would instead try increasing Feedforward on yaw to get it to track better. This will give you better tracking without any drawbacks of high P gains.
> It might help to add a note in the wiki preventing people to fall in the same problem of overfiltering with rpm filters.

It is clearly stated on the wiki that it is recommended to change gyro lowpass and dynamic notch filter settings when you enable RPM filter.

Mateyhv, 2019-10-09:
> It is clearly stated on the wiki that it is recommended to change gyro lowpass and dynamic notch filter settings when you enable RPM filter.

@IvoFPV yes, maybe I overlooked that detail but I thought thats to be done once RPMs are on, not expecting to find this behaviour with PT1 filters on, not even biquad. 
Regarding yaw, P is not what mainly pushes the quad, its I. P prevents overshooting. My original FF was around 170, for current tests it is on default.