betaflight/betaflight #9002: vtx not changing with smart audio

pa2pla, 2019-10-06:
the mach 2 doesnt change freq or power when instructed by smart audio or lua

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
vtx led to show change and vtx to change freq. 

**Flight controller configuration**
'''# diff

# version
# Betaflight / BETAFLIGHTF4 (BFF4) 4.1.0 Sep 25 2019 / 13:38:03 (08e8afa09) MSP API: 1.42

# start the command batch
batch start

board_name BETAFLIGHTF4

# name: PA2PLA XL5

# feature
feature -AIRMODE

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# aux
aux 0 0 0 1200 2050 0 0
aux 1 1 0 1500 1625 0 0
aux 2 13 0 1025 1100 0 0
aux 3 28 0 1625 1725 0 0
aux 4 35 0 1750 1850 0 0

# adjrange
adjrange 0 0 2 900 1200 12 2 0 0
adjrange 1 0 2 1350 1625 12 2 0 0
adjrange 2 0 2 1825 2100 12 2 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 23 27 29
vtxtable powerlabels 25 200 500 800

# master
set min_check = 1010
set max_check = 2000
set rssi_channel = 12
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set battery_meter = ADC
set ibatv_scale = 223
set small_angle = 180
set pid_process_denom = 1
set osd_vtx_channel_pos = 14402
set debug_mode = PIDLOOP
set vtx_band = 2
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5866
set name = PA2PLA XL5

profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# end the command batch
batch end'''

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` '''# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B03: FREE
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: FREE
C07: FREE
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: INVERTER 2
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: MOTOR 3'''

**Setup / Versions**
 - Flight controller betaflight f4
 - Other components rx crossfire nano, vtx rdq mach2 
 - how are the different components wired up.
all wired up the same as when on beta flight 3.5
**Additional context**
Add any other context about the problem here.

ivkostic, 2019-10-06:
Same behaviour with eachine5258, connected via led pad on softserial, luxf4osd board, no response from smartaudio with OSD as well on lua.

etracer65, 2019-10-07:
> - Other components rx crossfire nano, **vtx rdq mach2**

So then I presume you were using the "AKK - RDQ VTX Patch" firmware versions. This is not available for pre-release Betaflight versions. You'll have to wait until after BF4.1 is released.

pa2pla, 2019-10-07:
No I wasn't using the patch this is what is stumping me. But if that is the case and problem I will wait. Might go and re try other firmware and see if it's the same

pa2pla, 2019-10-11:
issue fixed my fault i set value to db like in bardwels video its ment to be value 0,1,2,3 lable 25,200,500,800

closed 2019-10-11 by pa2pla