betaflight/betaflight #9003: ANGLE + PARALYZE flags on RC6 furyF4OSD

packocrayons, 2019-10-06:
1. ANGLE and PARALYZE arming disable flags show on RC6. ANGLE is only present when battery is plugged in. MSP flag doesn't show.
Same flags do not exist and can successfully arm quad in 4.0.6.
Flight controller: mamba F405 mini mk2 (from a 20mm stack).
Spektrum satellite (spek1024) on UART3
No other peripherals.
Interestingly, THROTTLE flag (min_throttle was above 1050) doesn't show in 4.1.0 RC6 but required a min_check change in 4.0.6

Configurator: 10.5.1 (I believe this may be a bit old)

Diff:
```
# version
# Betaflight / FURYF4OSD (FY4O) 4.0.6 Sep  1 2019 / 00:40:20 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set min_check = 1064
set serialrx_provider = SPEK1024
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end
```

mikeller, 2019-10-07:
@packocrayons:

> Configurator: 10.5.1 (I believe this may be a bit old)

Looks like you've solved the problem all by yourself - Betaflight configurator 10.5.1 indeed predates Betaflight 4.1.0-RC6, and does not know about the new arming disabled code - you should switch to Betaflight configurator 10.6.0-RC2.

packocrayons, 2019-10-07:
Does this mean that the arming flag numbers have changed in 4.1.0? Or am I receiving some other code (I'll figure that out when I get home and upgrade again) that didn't exist when configurator 10.5.1 was released?

mikeller, 2019-10-08:
@packocrayons:

> Does this mean that the arming flag numbers have changed in 4.1.0?

Yes.

etracer65, 2019-10-08:
https://github.com/betaflight/betaflight/wiki/Arming-Sequence-&-Safety#description-of-arming-prevention-flags

packocrayons, 2019-10-08:
Good stuff, thank you!

closed 2019-10-08 by packocrayons