betaflight/betaflight #9005: MSP not working on CLRACINGF7 (BF 4.1RC6)

haplm, 2019-10-07:
When I click on "Activate Mass Storage Device Mode" in configurator, FC correctly reboots, but it is never mounted (MacOS). My other BF4.1 FCs don't have this problem, it seems to be isolated to CLRACINGF7 target.

I'm using Betaflight 4.1 RC6, on first version of CL-Racing F7 controller.

mikeller, 2019-10-07:
@haplm: Can you please add the information that you were asked to provide when opening this issue, including what version of MacOS you are using?

etracer65, 2019-10-07:
@haplm In addition to providing the requested info, can you try the `msc` CLI command instead of the butting in the Configurator?

Also it's not likely this is your problem, but I found a case where if the flash was completely full (128MB used) there was a potential problem. I don't think it will make a difference, but you could try https://github.com/betaflight/betaflight/pull/9007 .

It's more likely some issue with this new NAND flash used on the CLRACINGF7. I don't have one to test so I can't say for sure. No other targets use this new flash other than the SPRACINGH7's and H7's don't support mass storage mode yet.

haplm, 2019-10-07:
I'm sorry for not including the details earlier

diff:

```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct  6 2019 / 06:48:40 (c6452a55c) MSP API: 1.42
# manufacturer_id: CLRA   board_name: CLRACINGF7   custom defaults: YES

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id CLRA

# name: Hyperlow CG

# resources
resource MOTOR 1 B08
resource MOTOR 2 B06
resource MOTOR 3 B09
resource MOTOR 4 B07

# feature
feature -RX_PARALLEL_PWM
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -ON_USB

# serial
serial 1 2 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 3 1700 2100 0 0
aux 4 13 2 1700 2100 0 0
aux 5 26 3 1300 1625 0 0
aux 6 30 0 1500 2100 0 0

# adjrange
adjrange 0 0 4 900 2100 12 4 0 0
adjrange 3 0 6 900 2100 29 6 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 300
[... 227 lines cut ...]
Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
Build details: https://rotorbuilds.com/build/13472
MacOS: 10.14.6 (18G103)

It is definitely not a problem of having the flash full. I never go over 1/2 of the storage.

What is strange is that it works today (both ```msc``` command and the UI button). However, this never was the case and the only thing that changed is that the flash is empty (I have never tried to activate it with an empty flash). So I need to fly for couple minutes and try again to validate this assumption...

etracer65, 2019-10-07:
Go to the CLI and do `set blackbox_mode = ALWAYS” then save & reboot. Wait a couple minutes and set back to “NORMAL”. You’ll have a couple minutes of log without flying.

haplm, 2019-10-07:
Nice trick. 

While it takes more and more time to switch to mass storage with increasing size of the logs in the flash, it is now switching reliably every single time. I can't imagine how doing ```msc``` once could achieve this, but it apparently did.

I will close this for now. I have one more copter to upgrade to 4.1 (same FC), so I will report back if I find anything new around this. Thanks for the help with troubleshooting!

etracer65, 2019-10-07:
Taking longer for the volume to mount with more logs is normal and expected. The FAT32 filesystem is emulated and at startup we have to parse the flash to identify the different logs. That means reading through all the used flash sectors.

Doing the `msc` command didn't fix anything. The underlying function is the same whether invoked via the button or from the command line. I was just wanting to rule out a problem in the Configurator not actually starting mass storage mode.

haplm, 2019-10-19:
Hi @etracer65,  after flying for real, the issue is back. Neither the GUI button, nor `msc` turns on the mass storage (even if I wait excessive amount of time for it to come up). Is there anything I could try to isolate the problem?

jflyper, 2019-10-19:
How is your activity LED flashing when the FC reboots for MSC?

haplm, 2019-10-20:
Steady red and steady green first, then just steady red

etracer65, 2019-10-20:
@haplm The only thing I can suggest is for you to open the Console application and watch the log as the volume tries to mount in mass storage mode. Perhaps the OS will report problems that might be a clue.

If you haven't yet, upgrade to the 4.1 release to test if any fixes after the RC's had any effect.

Also look for a pattern, like it works if the flash is mostly empty but fails when full. Or something like that.

Unfortunately I don't have any engineering samples of this flight controller or ones with this flash chip that support mass storage mode so I don't have any way to test/debug.

haplm, 2019-10-20:
I have upgraded to 4.1, nothing changed.

Here is the console output after I press MSC button:
[image]

haplm, 2019-10-20:
When I check System report while in MSC mode (not working), I see this:
[image]

When I delete the flash (98MB was taken), I see this:
[image]

Of course that now it mounts perfectly fine. Sigh. I will have to generate really large log, I guess.

haplm, 2019-10-20:
BTW - I have just noticed the reported capacity - 269MB looks like around double of the real capacity, is it normal?

etracer65, 2019-10-21:
> BTW - I have just noticed the reported capacity - 269MB looks like around double of the real capacity, is it normal?

Yes, this is normal and correct. The emulated filesystem is always padded to be a minimum of 256MiB so that it's considered a FAT32 filesystem. Otherwise it'll be interpreted as a FAT16 filesystem. There's a hidden "PADDING.TXT" file that is added to bring the reported capacity up to 256MiB. Even if you only had a 16MB flash chip it would still report 256MB when mounted.

Also the filesystem will appear to use twice the actual recorded log space. This is because all the individual logs are there plus a "virtual" file that is all the logs combined together for a single download.

etracer65, 2019-10-21:
I'm not sure, but are you indicating that the filesystem will mount when there's not much in the flash but won't when it's more full? If this is repeatable, then it would be useful to iteratively test filling the flash gradually to see the point where it stops working. Would likely be some significant boundary like 32MB, 64MB, etc.

haplm, 2019-10-21:
Interestingly enough, it stops working after around 24 MB gets filled. I would almost swear that when I was trying it earlier, I was able to get past that amount, but today it was pretty consistent. It didn't matter how big was each log, but past around 24 MB total size, I could no longer mount it.

etracer65, 2019-10-27:
Well unfortunately I think we're at an impasse. I'm unable to replicate the problem with other boards and unfortunately I don't have any samples of the CLRacingF7 board with the 1Gb NAND flash to allow me to work on this further.

@bnn1044 Can you test or validate that mass storage mode (MSC) works correctly on MacOS with the CLRacingF7? In particular when the blackbox flash storage is significantly full?

bnn1044, 2019-10-27:
@etracer65 I will test it.

etracer65, 2019-11-26:
@bnn1044 Have you had a chance to test?

bnn1044, 2019-11-26:
I did test it. Forgot to report it back.  It work with not too much data on the chip. When it is about full. Mac system won’t recognize it. Window system with no problem.

etracer65, 2019-11-26:
@bnn1044 Thanks for testing. That confirms some problem either with the flash driver or the emulated FAT filesystem. Do you want to debug? I don't have one of these flight controllers and the only other boards that I know of using this flash chip is the SPRACING H7's and mass storage mode is not yet implemented for H7's so it's impossible to test with them.

etracer65, 2019-11-26:
@bnn1044 I messaged you on our slack.

etracer65, 2019-12-01:
@haplm I think I found and fixed the problem. Big thanks to CL Racing for sending me a board to debug with.

Can you please try this? (will be included in 4.1 as a fix, but easier to give you a 4.2 version now).
[betaflight_4.2.0_CLRACINGF7.hex.zip](https://github.com/betaflight/betaflight/files/3908332/betaflight_4.2.0_CLRACINGF7.hex.zip)

The problem turned out to be the large amount of flash taking too long to scan for logs and causing the operating system to timeout when mounting the mass storage device. That's why the problem manifested as the flash was more full. Once the flash usage exceeds ~50MB it was taking too long to scan. Since we can't really make the scanning go faster as it's a function of reading the pages from flash, I restructured the logic to do the scanning before the USB device is initialized. That way the filesystem is immediately ready when the operating system tries to mount the device.

Note that the time to mount the storage device will still depend on how much flash is used. To make it clear that something is happening the status LED on the board will rapidly flash while the flash is being scanned.

To fully test make sure your flash is significantly filled - at least > 50%. As a reminder, a way to fill up the flash without flying is to simply do:
```
set blackbox_mode = ALWAYS
save
```
Wait a while (stay out of CLI) and let the flash fill up. Then go to the CLI and
```
set blackbox_mode = NORMAL
save
```

haplm, 2019-12-01:
Thanks, @etracer65, that's a great news. I will test it ASAP, hopefully tomorrow evening.

PR #9252 (2019-12-03): Restructure flashfs MSC to prevent device timeout on MacOS
Fixes #9005 

For devices using the 1Gb NAND flash the scanning phase that discovers the individual logs was taking to long resulting in the operating system thinking the mass storage device had become unresponsive. It is simply a factor of the larger flash size and the amount of time needed to scan all 2K blocks.

Restructured to scan the flash **before** initializing and starting the USB device. That way the mass storage device can start quickly and be immediately ready for operating system requests.

Tested on MacOS and Windows 10.
closed 2019-12-03 by mikeller