betaflight/betaflight #9008: manufacturer id apply

eroticisim1978, 2019-10-07:
Hi All:
 We are a new FC manufacturer in shenzhen ,china.
Now we are prepare to get the firmware for our own FC board.
And there is a problem to apply the manufacturer id?
Would you please tell me how to get the manufacturer id for our own FC board.
.MCU STM32F722RET6
.Brand  CORDDAR_F722DUAL

Thanks for your help.

mikeller, 2019-10-08:
Hi @eroticisim1978.

To get a manufacturer id, can you please specify the name and website address for the company that will be selling these flight controllers and provide customer support for them?

eroticisim1978, 2019-10-09:
Mikelller:
 thanks for your reply.
 We want to use the name of the CODDAR-F7.
 The following is the information of our company and brand.

Head office name
 CODDAR LIMITED
Shenzhen office name
SHENZHEN GNBL LIMITED

Company website (under construction...)
 www. Coddar.net
 
CODDAR USA on line shop (under contruction..)
www.coddarrc.com

mikeller, 2019-10-09:
Neither of these pages seem to work - is there any other web presence that we can direct your users to for support?

eroticisim1978, 2019-10-11:
Dear Mikeller:
 We using the www.coddar.net as our werbsite. please check it .

mikeller, 2019-10-12:
@eroticisim1978:

This website is currently used by 'conjoint Digital Company':

[image]

Is this the company that should be listed as sales and support contact for this flight controller?

closed 2019-11-18 by stale[bot]