betaflight/betaflight #9014: Matek F405-CTR Led strip issue

DangerD256, 2019-10-08:
Only 7 Leds are working, 8th led is always green.

Tried to reflash to inav, everything works fine there (but on S5 pin)

DangerD256, 2019-10-09:
> etracer65 added the Missing Information label 16 hours ago
What exactly is missing?

mikeller, 2019-10-10:
@DangerD256: The information you were asked to provide when opening this issue:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

jflyper, 2019-10-10:
You have a conflict of pin function between I2C and LED_STRIP (PB6 is claimed by `I2C_SCL 1` in your `resource show` output) or, you haven't assigned S5 as LED_STRIP output (there is no `LED_STRIP` entry in your `resource show` output).

If you are using LED pad to connect your LED strip, then `I2C_SCL 1` must be freed by
```
resource I2C_SCL 1 none
```

If you are using S5 pad to connect your LED strip, then you have to assign S5 (PA15) as LED_STRIP output by
```
resource LED_STRIP A15
```

closed 2019-11-16 by stale[bot]