betaflight/betaflight #9015: Late starting motors on Betaflight 4.0.6 and 4.1 RC6

cheetah2k, 2019-10-09:
**Describe the bug**
Upon arm, motor 3 seems to always start late - which when you crash means the quad will flip every time regardless of whether any props are damaged, etc.

**To Reproduce**
Reproduce issue on 4 identical brand new quad builds using BF 4.0.6 or BF 4.1 RC6
Issue is not present on BF 3.5.7

**Expected behavior**
All motors should start at the same time on Arm

**Flight controller configuration**
FuriousFPV Racepit
Stock setup except for:- DSHOT600 and 8/8 loop times

**Setup / Versions**
Flight Controller - FuriousFPV RacePit
ESC - Aikon RD32 F3 4in1 FC

**Additional context**
1. Updated the 4in1's to latest FW - same issue presents.
2. I have to run Auto Timing, and 48khz - anything lower and both Motor 3 and 4 start late.
3. Changing to BF 3.5.7 the issue is gone.
4. All quads fly as expected in the the air (zero issues) however if landing, disarm, re-arm, the quads just flip over due to the seemingly out of sync / late starter Motor 3. To be able to repeat this issue on all 4 quads (and I'm an experienced builder) either means there's an issue with BF or BL32 - dropping back to BF 3.5.7 the issue is gone.

etracer65, 2019-10-09:
Please provide the information requested when opening the issue or it will be closed without further comment:

> **Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
>
>Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

cheetah2k, 2019-10-09:
```# diff all

# version
# Betaflight / FF_RACEPIT (RACE) 4.1.0 Oct  6 2019 / 06:24:44 (c6452a55c) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FF_RACEPIT
mcu_id 002a004b594b500820323457
signature 

# serial
serial 3 8192 115200 57600 0 115200

# aux
aux 0 0 1 1500 2100 0 0
aux 1 2 0 900 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 100 200
vtxtable powerlabels 25 50 100

# master
set acc_hardware = NONE
set min_check = 1020
set max_check = 1980
set rssi_channel = 16
set motor_pwm_protocol = PROSHOT1000
set bat_capacity = 1000
set small_angle = 180
set pid_process_denom = 1
set transient_throttle_limit = 10
set osd_cap_alarm = 1300
set osd_vbat_pos = 14785
set osd_rssi_pos = 14370
set osd_tim_2_pos = 14806
set osd_pidrate_profile_pos = 385
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_fft = ON
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 3
set vtx_freq = 5740
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 900

profile 0

# profile 0
set dterm_lowpass2_type = BIQUAD
set vbat_pid_gain = ON
set iterm_relax_cutoff = 35
set p_pitch = 54
set i_pitch = 80
set d_pitch = 30
set f_pitch = 90
set i_roll = 70
set d_roll = 28
set f_roll = 84
set f_yaw = 30
set d_min_advance = 0
set auto_profile_cell_count = 4

profile 1

# profile 1
set vbat_pid_gain = ON
set p_pitch = 48
set i_pitch = 80
set d_pitch = 28
set f_pitch = 90
set p_roll = 40
set i_roll = 70
set d_roll = 26
set f_roll = 84
set f_yaw = 30
set motor_output_limit = 70
set auto_profile_cell_count = 6

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_expo = 25
set pitch_expo = 25
set roll_srate = 68
set pitch_srate = 68
set tpa_breakpoint = 1650

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

cheetah2k, 2019-10-09:
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: I2C_SCL 3
A09: FREE
A10: CAMERA_CONTROL
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FLASH_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: LED 2
B09: LED 1
B10: MOTOR 4
B11: MOTOR 3
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: ADC_CURR
C02: ADC_BATT
C03: BEEPER
C04: FREE
C05: SYSTEM
C06: FREE
C07: FREE
C08: PINIO 2
C09: I2C_SDA 3
C10: FREE
C11: SERIAL_RX 3
C12: FREE
C13: FREE
C14: FREE
C15: INVERTER 3
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH3: CAMERA_CONTROL
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

etracer65, 2019-10-09:
Try using DSHOT instead of PROSHOT.

cheetah2k, 2019-10-09:
Now this is interesting because the Configurator says im using Dshot1200

etracer65, 2019-10-09:
Because you're using an older Configurator version that doesn't support BF4.1. You need to use the latest.

https://github.com/betaflight/betaflight-configurator/releases

cheetah2k, 2019-10-09:
Dshot600 still has the same issue.

cheetah2k, 2019-10-09:
Here is the issue with ESC timings Auto / PWM 48khz..   Motor 3 still has a slight stutter, which does not happen all the time (looks like 50%)

https://youtu.be/Z1nEyy1NbgU

However, if I leave the Aikon RD32 timings set on defaults, its a whole other story.  Check this.

https://youtu.be/4yABletktRo

I swapped out the FC for a new Aikon F7 FC, This is running BF 4.1 rc6, and BL32 32.7 fw.

cheetah2k, 2019-10-09:
Also worth noting, changing to Multishot, none of the start up chimes are ever in sync, even if I carry out a calibration in BF..   I just don't get it..  I've not seen this sort of issue with BF since BF 2.x

cheetah2k, 2019-10-22:
Latest BL32 and 4.1 still no dice with the RD32 4in1.  I've changed flight controllers to a few various ones (Aikon F7, FuriousFPV Fortini & HGLRC F7 Forward flight controllers, and no dice.  Still same issues.  Cannot run these 4in1's on Timing 16 and 24khz.  They have to be run in Auto Timing and 48khz or the motors just don't start in sync.  Frustrating - I think either Aikon have made a bad 4in1 esc or the software just doesnt work well with it.

Got no answers.  Returning them all back to Aikon...

mikeller, 2019-10-22:
@cheetah2k: Can you try running Dshot300? (You'll have to change the PID loop to 4k as well.) If this fixes it then there is something fishy going on with the wiring between your flight controller and your ESC.

cheetah2k, 2019-10-22:
Dshot300 and 4+4k makes it worse than 8+8 and dshot600..   

I noticed more on Dshot300 that its as if the motors cannot decide what direction they need to spin - so its more of a "stutter" than a desync..  

As noted before I've tried different flight controllers, different cables, checked for grounds, all 4 builds are clean.  I took the RD32 out of one of the quads and swapped for a Hobbywing 60A 4in1, and it has zero issues... All motors start up perfectly...

I give up....

cheetah2k, 2019-10-22:
> @cheetah2k: Can you try running Dshot300? (You'll have to change the PID loop to 4k as well.) If this fixes it then there is something fishy going on with the wiring between your flight controller and your ESC.

cheetah2k, 2019-10-22:
> @cheetah2k: Can you try running Dshot300? (You'll have to change the PID loop to 4k as well.) If this fixes it then there is something fishy going on with the wiring between your flight controller and your ESC.

I found dshot300 and 4+4k to be worse than Dshot600 and 8+8.  The motors seem to "stutter" on arm - its as if they dont know which direction they are supposed to be going.

I swapped a RD32 out of one of the 4 builds last night, and installed a Hobbywing 60A 4in1.  Works perfect on all settings..

I think i've tried everything with these builds - different FC's, checked for grounds, changed 4in1 to FC cables, etc, etc.  Can't work it out..  I give up

closed 2019-10-22 by cheetah2k

cheetah2k, 2019-10-30:
Why is this issue closed?  Clearly there is an issue with F3 based ESC's..

etracer65, 2019-10-31:
@cheetah2k You closed it.

And even so, you've confirmed there is no problem with another ESC. Just because you had a problem it's incorrect to jump to the conclusion that there's an issue with F3-based ESCs. We've had no other reports of this problem. So that would indicate a problem with your specific ESC, or possibly a BLHeli32 firmware issue affecting that particular ESC.

nightflasher, 2020-05-26:
Just wanted to add another point to this issue: built my ZMR250 with a "new" omnibus F3 board and old blheli xp12 esc and had the same issues AND on version 4.0.5 everything is fine (running multishot).