betaflight/betaflight #9019: Cannot enable bidirectinal dshot on kakute f4v2

ujjwalk450, 2019-10-09:
dump

# version
# Betaflight / KAKUTEF4V2 (KTV2) 4.1.0 Oct  6 2019 / 06:29:37 (c6452a55c) MSP API: 1.42

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id 

# name: -

# resources
resource BEEPER 1 C09
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 C07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 C08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 B15
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
[... 937 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 101
set pitch_rc_rate = 101
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 63
set yaw_srate = 75
set tpa_rate = 10
set tpa_breakpoint = 1650
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

etracer65, 2019-10-09:
Instructions ask for a `diff`, not a dump.

> **Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
>
>Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

Also describe how you think you can't enable bidirectional DSHOT. We can only guess what you mean or how you tried to enable it.

ujjwalk450, 2019-10-10:
I have tried to switch on bidirectional dshot by flipping a switch in configuration tab of betaflight. 
I have also tried to paste code snippet for BF4.1 as described in https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

jflyper, 2019-10-10:
@ujjwalk450 

Would you please try the following procedure?

(1) Re-flash with full chip erase.

(2) Immediately configure motors for DSHOT300 and Bidrectional DShot and nothing else.

(3) Test motors with motors tab.

(4) If this works, then slowly add your config diff output and see if any lines you've added will reset the bidir-dshot?

etracer65, 2019-10-10:
@ujjwalk450 Your settings show no sign of bidirectional DSHOT being enabled. Perhaps you forgot to click the "Save" button on the Configuration tab after enabling.

Also, this is a problem:
```
set scheduler_optimize_rate = OFF
```
This will prevent bidirectional DSHOT from working. Not sure why you have this as the default is "AUTO". I recommend you reset to `defaults` and configure manually (don't paste in a `diff`).

ujjwalk450, 2019-10-10:
@jflyper  thanks dev, this worked.

closed 2019-10-10 by ujjwalk450