betaflight/betaflight #9020: sudden ACC drifting / Matek F7 722SE / BF4.1.0-RC6?

hundemann, 2019-10-09:
Not sure if this is an issue, but I had on two same FC using that same FW a sudden ACC drift, after connecting to BF configurator (latest release).
What happened before: I was flying both already, but I almost had a nose crash with one of them during first launch. After connecting it to the configurator I noticed the ACC degrees were way off. 
The second one was flying great, but on that one I just could monitor now a continous drift, means while sitting on the table, the ACC was suddenly running off in the set-up screen.

On the first one I calibrated the ACC after I noticed the weird angles. And it is set to ~0 still now.
The second one just running off now I also calibrated the first time after FW flashing and it is also still on ~0°.

Both ones I did not calibrate after FW flashing, because I notices after connecting the ACC was calibrated perfectly (i mean really perfectly, usually after calibrating the ACC shows some 0,1 etc. values), so I did not bother further...

Thank you for any comment..
Malte

etracer65, 2019-10-10:
Always calibrate the ACC after flashing. The calibration offsets are stored in the firmware settings and are erased if you do a full-chip erase (or otherwise reset to defaults). Calibration is not just about setting the "level", it also determines the DC offsets for each axis needs to correctly interpret the sensor readings.

jflyper, 2019-10-10:
Shall we define and use a new arming prevention flag `ARMING_DISABLED_ACC_NOT_CALIBRATED`? I believe  iNav has an equivalent.

mikeller, 2019-10-10:
@jflyper: Not sure - I think that one difference between iNav and us is that in iNav pretty much everybody will be using the accelerometer, whereas in Betaflight a lot of users will probably never use it.
So if anything, ACC calibration should only be required if the user wants to actually use ACC - I am just not sure on how to best do this.

etracer65, 2019-10-10:
@mikeller I can look into it - shouldn't be that hard. Should just need to look at the potential consumers of the ACC - like self-level modes, GPS rescue, acro trainer, launch control, etc. There are a few other OSD non-flight critical like artificial horizon, etc.

The more difficult part is determining if the ACC has been calibrated. An uncalibrated case will have `acc_calibration = 0,0,0`, but it's technically possible (although unlikely) for the actual calibration to be all 0's. So we would probably need to store an additional iterm in the PG indicating whether calibration took place.

mikeller, 2019-10-10:
@etracer65: Yeah, it will just be herding cats to collect and keep on top of all the different uses of ACC. :wink: And yes, this will need an extra parameter to keep track of the ACC calibration state.

closed 2019-11-16 by stale[bot]