betaflight/betaflight #9025: RC dropped frame gets extremely high under increased CPU load

theely, 2019-10-10:
**Describe the bug**
Looks like there's a relationship between dropped RC frames on Crossfire (maybe other Serial Receiver Provider are affected too) and how loaded the CPU is.
I have tested BF 3.5.8 - BF 4.0.6 - BF 4.1RC6 and BF 4.1RC6+RPM on the same setup with the same testing conditions.
The FC I have been using is a CLRacing F7 V2, I was also able to reproduce the same problem on a Matek F722Mini.

As you can see from the following image, the more we load the FC CPU the more RC frames are dropped.
[image]

**To Reproduce**
Log RC_Smooting_rate on different BF versions with increasingly higher CPU load.

**Expected behavior**
Ideally RC frames should not be affected by CPU load unless it's extremely high (e.g. >75%).

**Flight controller configuration**
[BTFL_cli_20191010_175252-BF-3.5.8.txt](https://github.com/betaflight/betaflight/files/3713622/BTFL_cli_20191010_175252-BF-3.5.8.txt)
All test did run the same config, with exception of the RMP one
[BTFL_cli_20191010_183332-BF-4.1RC6-RPM-ON.txt](https://github.com/betaflight/betaflight/files/3713636/BTFL_cli_20191010_183332-BF-4.1RC6-RPM-ON.txt)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
[BTFL_cli_20191010_183433-resource.txt](https://github.com/betaflight/betaflight/files/3713679/BTFL_cli_20191010_183433-resource.txt)

**Setup / Versions**
 - CLRacing F7 V2

etracer65, 2019-10-10:
Not sure why you would think this is surprising or unusual.

> Ideally RC frames should not be affected by CPU load unless it's extremely high (e.g. >75%).

Why would you think that? We only have a single CPU core and no multitasking. And even so your blackbox logs aren't really convincing. Take at different times there's no way to say the frame delays weren't cause by OTA delays in the Crossfire protocol.

theely, 2019-10-10:
Crossfire frames com in every 150Hz max, right? So it's surprising that we can not fetch them under moderate CPU load.
But never mind, let's not focus on that! What I am concerned off is the high amount of frames that are dropped with RPM on. It's so high that it's start affecting setpoint and this is while the quad is on the work bench, once you start flying it gets even worst.

Could you elaborate what OTA delays in the Crossfire protocol are? I made several measurements over multiple days always with similar conditions, with the RX only few centimeters from the TX.
LQ was always 99. I doubt something else is affecting the RC frame drop in such a noticeable manner.

etracer65, 2019-10-10:
The receiver frames aren't "dropped", they're delayed. And it's a side effect of the scheduler optimization required to get the PID loop synchronized with an external timebase. As a result of scheduling the PID loop earlier than normal to counteract any delays from the previous iteration it impinges on the time available to do other tasks. That's also why it gets worse at 8K as there is only 125us vs 250us for 4K.

theely, 2019-10-10:
I see, and you are not worried about the consequences?
I find it a bit troubling that an F7 with 4K/4K and RPM on has so many delayed frames that it starts affecting flight performances.
On one side with RPM you improve setpoint tracking (not only) but then on the other side your setpoint gets worst because of delayed frames. Is this the right tradeoff?

etracer65, 2019-10-10:
I didn't say I like it. I'm just telling you the cause.

theely, 2019-10-10:
Is there something we can do about it?
Could be as simple as communicating the tradeoff in the BF 4.1 wiki page so people are aware of it and they don't waste huge amount of time (as I did) trying to find out why the quad is not flying well.

etracer65, 2019-10-10:
Without actually having access to any of these logs other than screenshots it's really hard to suggest anything. However the one screenshot showing the motors you appear to be highlighting a 60ms step. That would indicate a problem with your Crossfire setup. Likely in the middle of switching from 150hz to 50hz (20ms) mode and dropped some OTA (over the air) frames. Crossfire switches from 150hz to 50hz and back quite a lot - even when in close range. Other times when I've seen problems like this the antennas on the Crossfire receiver were bad. One other case the receiver itself needed to be replaced.

The jitter I was discussing is just a delay of the frames that looks like a ringing pattern if you look at the frame rate debugging.

theely, 2019-10-10:
Here are the logs I used for the screenshots: https://drive.google.com/open?id=1HYpHPtG7AFlrP33Ha0u0NHbamaKWx4-k

Here is the real bad one with motor the motor traces:
https://drive.google.com/open?id=19StII2i61gPHDL6o7ms87jpLrNct_CuU&fbclid=IwAR1TOItqGIIoBC-FiTSRBbezLd2eYm4Y7U_WZRu3AXtP0a9WQJhdLq5_0yU

To replicate the issue on my setup it's sufficient to lock Crossfire on mode 1 (50hz) and set 8K/8K the combination of slow frame rate and delayed frames is in my opinion what is causing the issue. The link quality was actually always good.

ctzsnooze, 2019-10-11:
Looking at the test log, the receiver is actually running at 150hz.  I've seen this in other CRSF logs; crossfire mode 1 is not actually locked to 50hz at all, it randomly switches 150 <-> 50 in that mode.  The only locked mode is locked 150hz.  

There are big steps in setpoint some 60ms or wider, but its not 100% clear, since the Rx refresh rate wasn't logged, if those were associated with a failure to receive a packet.  I guess we need a flight log with that debug active, and perhaps less RC smoothing (so that we can clearly see setpoint and FF steps in the normal log traces).

The RxRefreshRate debug in the test logs shows some packets are spaced at twice the normal interval.  In 4.1RC6 there are about some 4-6 packets / second; in 3.5.8 its 1-2, and a bit variable, but clearly less on the same hardware.

This kind of problem could, in simple terms, be attributed to three possible  causes:
- the transmitter never sent the packet to the rx
- the rx didn't receive a packet, or couldn't decode it, so reported the old value
-  the Tx and Rx were working perfectly, but Betaflight failed to detect the information being sent to it by the Rx because it was busy doing something else.

Given that the hardware is the same, the greater incidence of double Rx Rate intervals suggest and issue in betaflight that is worse with newer code.

I've seen similar issues with FrSky receivers, where for some reason set point doesn't change at the expected time, and the same value gets held over two normal steps.  My testing was done on a bench top with very high signal strength, and was done on multiple FrSky Rx's, and the same thing happens in close quarters test LOS flights.

Just one 'missed' packet causes issues with feed forward and boost, worse for FrSky, and is really bad if the link is in a 50hz mode.

I wrote anti-glitch code to ameliorate the magnitude of the glitches on feed forward, and found that interpolated_setpoint.c didn't always run at the expected regular intervals, which would only happen if the FC perceived that no new data packet had been detected.  

This issue seems worse than I recall, but it could just be that it's more obvious now we run more FF and have boost to exaggerate the effects.  I don't quite know how to go about identifying the root cause of the problem.  What I mean is I don't know how to determine whether the problem is an issue in the Tx-Rx link vs something intrinsic to betaflight.  I found the Rx code in betaflight to bee too difficult for me to follow.  

@etracer65 I'm not sure I understand your comments entirely right, but were you meaning that Betaflight could entirely miss incoming data packets despite them being sent to the FC properly by the Rx?  ie not that they would just be delayed a few PID loops, but lost completely?

etracer65, 2019-10-11:
No, I specifically said they weren't dropped.

> The receiver frames aren't "dropped", they're delayed.

adrianmiriuta, 2019-10-11:
@ctzsnooze
I have noticed this issue with FrSky receivers too,
but it turned out it is the normal way the FrSky protocol works.
Each time when a Telemetry frame is sent back from the receiver you get the old RC values.

azixaka, 2019-10-11:
Ah I get these weird motor traces and I simply can't get to the bottom of this. I wonder if it's this!

https://1drv.ms/u/s!AiGgeJ7isHv6kpQtNWLWXnO1Ojq3Zg?e=aF2siw

ctzsnooze, 2019-10-12:
@etracer65 - kind of how bad is that scheduler related delay, typically?  I mean, what kind of range of loop counts?  If it was one or two loops that wouldn't matter much, I think.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                

@adrianmiriuta - do you have a link that explains the telemetry frame issue?  Does seem remarkable that they would drop an entire data frame just to send a telemetry packet back.  If so would that mean that if I disable betaflight telemetry over SBus that the dropped packets should disappear?

adrianmiriuta, 2019-10-12:
@ctzsnooze 
they do not drop a data frame.
There is no data available because the RF timeslot is used to send  Telemetry back to the radio
not to receive RC data  from it.
So the SBUS or FPORT data frame sent to the FlightControler is identical to the previous one.

You can study some projects that reversed the FrSky protocols.
https://fishpepper.de/2016/09/07/opensky-on-d4r-ii-pinout-flashing-and-usage/
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module

mikeller, 2019-10-12:
@adrianmiriuta:

> There is no data available because the RF timeslot is used to send Telemetry back to the radio not to receive RC data from it.

Can you specify what protocols you are talking about? Could it be that this is specific only to the R9 protocol, potentially only when running in 50 Hz 'LoRa' mode?
This is definitely not the case for the FrSky 2.4 GHz 'D' and 'X' protocols. Have a look at 

https://github.com/betaflight/betaflight/blob/master/src/main/rx/cc2500_frsky_d.c

and

https://github.com/betaflight/betaflight/blob/master/src/main/rx/cc2500_frsky_x.c

to see how the over the air protocol is combining RC data and telemetry into the same frame.

ctzsnooze, 2019-10-13:
> Ah I get these weird motor traces and I simply can't get to the bottom of this. I wonder if it's this!
> 
> https://1drv.ms/u/s!AiGgeJ7isHv6kpQtNWLWXnO1Ojq3Zg?e=aF2siw

No, your log has zero feed forward in the configuration, so it has no relevance to this discussion.  The quad is not performing well because there is no feed forward and you are running a lot of P to compensate, the result is P wobbling and overshoot with quick setpoint changes.  Although your Rx is mostly in 20ms / 50hz mode and occasionally isn't smooth, it would still be better with some feed forward, just not a lot.

If you want help on tuning, post in a support group, not here, thanks.

adrianmiriuta, 2019-10-13:
@mikeller 
I am talking about all protocols.
It is about the RF transmission.

Let's take  FrSky D as example :

You have 2 states
    STATE_DATA,
    STATE_TELEMETRY,

every time when
     if ((packet[3] % 4) == 2) {
        *protocolState = STATE_TELEMETRY;

you will build a telemetry packet and use the next RF timeslot to transmit it  (receiver->RC)
in this timeslot there is no received data  (RC -> receiver) 

The firmware in the receiver sends a duplicate of the last received RF frame to the FC, 
what protocol you are using is not relevant.

You cannot combine RC data and Telemetry in one RF frame !
as you cannot transmit and receive RF at the same time.

mikeller, 2019-10-13:
@adrianmiriuta:

Ah I see. I assumed that this was happening within the 9ms RC data frame intervals, but I've never actually checked this - especially in the case of FrSky X where the telemetry packet is a lot shorter, and cannot be disabled from the TX side you'd expect that.

ctzsnooze, 2019-10-13:
@adrianmiriuta:

Does this mean that if I disable telemetry, when binding an X series Rx, I should see fewer situations where the same data is repeated on successive packets?

I have noticed that if I swing the sticks, there is often a duplicate data packet when the rate of change of the value is low, eg when the sticks are moving more slowly.  For example, if I swing the sticks back and forward, the duplicate data values appear to happen most often as the sticks reach the extreme of movement only.  Do you happen to know if telemetry packets (in FrSky X series) are sent at regular time intervals or sent 'adaptively' at times when there is least signal variability?

adrianmiriuta, 2019-10-13:
@ctzsnooze 
I do not know how it is handled on FrSky X protocol.

I studied only a reversed version  of FrSky D8 with FrskyHUB telemetry.
There every 4'th packet is a telemetry reply,
the 4'th SBUS frame is a copy of the 3'th frame.

theely, 2019-10-13:
I am trying to summarize the thread, please let me know if my current understanding is correct.
 
1. Beside link quality, there are many other reasons that can affect how precisely the stick input is translate into the raw RC commands signal:
   a. the receiver firmware and how RC frames are handled
   b. specifics of the protocol and optional features (e.g. telemetry)
   c. BF ability to fetch the RC frames in a useful timeframe

2. Current BF architecture is prioritizing pid loop and filtering (gyro + D) over RC signal. e.g. Fetching RC frames and computing setpoint only happens if there is enough time in between pid and filtering execution (this was counter intuitive to me, I would have expected that knowing where to go was of higher priority than knowing how to get there).

3. The more BF is improving in its ability to track setpoint the higher is the expected quality of such setpoin. In other words it's hard to track something that behaves unpredictably.

4. Current debug traces do not allow to clearly diagnose what are the issues affecting noisy RC commands.

Given the above points, would it make sense to start addressing #4? A specific debug mode to better understand what is happening? Is that even possible? If pointed to the right direction could someone like me, with limited understanding of BF code base implement such new debug mode?

ctzsnooze, 2019-10-13:
@adrianmiriuta 
Thanks for your reply.  I looked just now at even more logs (5 quads) and reliably the duplicate data happens only when the rate of change of signal is low.  I think this must be an innovation for the X series. The R8 module has a regular repeating pattern every 4 data packets. Wonder if one is telemetry ???
@Theely
Go `get debug_mode` in the CLI and you'll see many debug options.  Quite a lot are RF specific.  If you pull the GitHub repository you can search locally for the name of the debug and will see what each parameter is set to.  It's not hard to modify what they do.  
The above FrSky graphs use `set debug_mode = FF_INTERPOLATED`

adrianmiriuta, 2019-10-14:
@theely 

OpenTx and firmware of the TX module also play a major role  in timing and latency behavior.
There is Code in OpenTx for averaging stick input that also introduces latency. (up to 13ms).
see: [x-lite-latency-testing](https://oscarliang.com/x-lite-latency-testing/)

There are reports from users that reduced latency to Half using er9x in place of OpenTx.
see: [Reduce your Taranis Latancy](https://www.youtube.com/watch?v=hCX2Qi51ZtU)

I personally useGPL  firmware Openground + opensky
[openground](https://fishpepper.de/projects/openground/)
[opensky](https://fishpepper.de/projects/opensky/)

My setup has measured Stick to SBUS latency of 11.9ms
[Latency measurement](https://www.calculator.net/standard-deviation-calculator.html?numberinputs=9.1%2C14.0%2C11%2C10.2%2C8.2%2C12.4%2C13.2%2C9.3%2C19.5%2C13.8%2C18.3%2C11.7%2C10.4%2C10.4%2C11.1%2C8.1&ctype=s&x=61&y=20)

JamesKingdon, 2019-10-14:
Going back to the original problem description, the claim is that the number of dropped frames has become significantly larger with BF 4.1 than it was with F 3.5.8. Is that in fact true, and if so is it an unavoidable side effect of increased cpu load and/or the new scheduler, or is there anything that can be done to reduce the number of dropped frames?

bizmar, 2019-10-17:
I noticed this too, in my testing the logging rate also impacts the error rate. So it's very hard to say how much of an issue it is when not logging. With low PIDs and FF, the artifacts are very hard to notice in flight, only in logs. With high FF, I can hear the motors are choppy. 

Here is how I currently run my F4 quads, until the issue is resolved, 4K4K 216mhz OC, ACC off, RX telemetry off. 
```
set gyro_sync_denom = 2
set pid_process_denom = 1
set motor_pwm_protocol = DSHOT300
set cpu_overclock = 216MHZ
```
This is only one aspect of the RC poopshow that CTZ and gang have been unraveling throughout the year as they worked on improving FF. 😅

ctzsnooze, 2019-10-20:
I had authoritative confirmation from TBS that their telemetry packets are sent back every single frame, so that telemetry is not responsible for any 'missed' or 'duplicate' packet issues with CSRF.

When setpoint does not change on successive CRSF packets, it can be due to either:

- A duplicate packet being sent and received (same data in successive packets), or 
- A packet with new data being sent by the Rx but somehow 'missed' or 'ignored' by the FC
- No packet being sent at all.

I know for sure from ff_interpolated debug logs that received packets with identical data do happen.  CRSF will send a duplicate occasionally.  

The other two possibilities are tough to differentiate between, unless the Rx output is logged in parallel to a robust system that should always record every single packet output by the Rx, while, at the same time, the FC also logs its normal activity.  This kind of testing could be done by parallel wiring two FCs to the same Rx and having one with extremely low CPU load and the other normal load.  If the incidence of flat setpoint steps was greater under higher CPU load then this would mean that entire incoming packets were 'missed' or 'ignored'.  

Short delays in receiving or reacting to incoming Rx data due to high CPU load aren't a problem, but completely missing entire Rx packets is a significant issue.

ctzsnooze, 2019-10-20:
This is a graphic from CRSF where setpoint does not change for one Rx packet, and betaflight identified that a new packet actually arrived, triggering the FF2.0 calculations, and my custom branch then moved the counter from 1 to 2.  So this is a duplicate received packet.
/Users/chris/Desktop/TBS issues/SetpointStepZeroValuePacket.jpg

In contrast, from the same log, here are three similar steps in setpoint, but the ff code was not triggered because there was no incoming new Rx packet detected by betaflight.  It was as if the packet was not sent.  We need to make a test that confirms whether a packet was, or wasn't sent, in these situations:
/Users/chris/Desktop/TBS issues/SetpointStepsNoZeroValuePacket.jpg

ctzsnooze, 2019-10-20:
Here is a log file for the above.  Code is from https://github.com/ctzsnooze/betaflight/tree/ff_3and4_test
with debug ff_interpolated.
/Users/chris/Downloads/adc_off-ff_interpolated.bbl.zip

adrianmiriuta, 2019-10-21:
@ctzsnooze 
I think when you get old RC data in a new SBUS, FPORT,CRSF frame 
than the delay Stick to RC doubles.
Or do I have a logic error ?

ctzsnooze, 2019-10-21:
They have authoritatively answered this.  They always send telemetry data back after receiving data, all within the 6.66ms packet time.  So there are no dropped packets in CSRF due to telemetry.  
When there is a step like that and old data is sent in a new packet, of course, we have delay.  Much worse than that, we get huge steps in calculated feed forward since the feed forward of no change is suddenly zero, the acceleration is negative.  Then with the new packet, we get acceleration from zero upwards and a double height upwards FF spike.  This makes a huge glitch in PIDsum when feed forward and boost are used.  That's why this issue needs addressing at the source.  I have a branch with reasonable suppression of identified identical packets but it is imperfect, and cannot help when the FC doesn't see that a new packet arrived (or if it actually didn't arrive).

adrianmiriuta, 2019-10-22:
@ctzsnooze 
a 3.3ms RF frame, and new telemetry every CRSF frame ...
is really hard to believe if you look at some comparative latency tests.
[Tests](https://oscarliang.com/r9m-lite-crossfire-latency-testing/)

AlessandroAU, 2019-10-22:
@adrianmiriuta I don't believe it either. I'm developing my own RC link based on LoRa hardware and I get 10ms measured latency from a physical switch to RX serial output at 5ms frame interval. CRSF should be better than that if they can do 3.3ms...

ctzsnooze, 2019-10-25:
All I know is the FC shows new data every 6.66ms and their radio gets telemetry back from the receiver every 6.66ms also.  They made no statement about overall latency.  Latency is not the topic here, it is about dropped or missed packets.

adrianmiriuta, 2019-10-25:
@ctzsnooze 
that is exactly my point ... 
to receive RC data and transmit back telemetry in 6.6 ms 
you have to switch RX/TX every 3.3 ms 
and every second RF frame would be a telemetry frame.
But the latency tests speak more  for a 6.6 ms RF frame.

that means it is **Impossible** to have new RC data and telemetry each 6.6 ms

etracer65, 2019-10-25:
Crossfire uses full-duplex RX/TX. There's no need to "switch RX/TX" on a single half-duplex connection.

frozenskys, 2019-10-25:
This is all distracting from the fact that the FC seems to be having issues with some packets being either duplicated, delayed or lost - which is affecting FF. How do we go about determining the root cause and proper fix?

adrianmiriuta, 2019-10-25:
@etracer65  
I do not own a crossfire, 
but sending and receiving at the same time implies 
2 RF modules 2 antennas 2 different RF channels used.
I saw none of those in any product description.

@frozenskys 
the duplicate RC (sbus, fport, crsf) packets are because of telemetry, 
delayed and lost packets have other RF, protocol, betaflight ... causes 

How these packets should be handled is another question.

etracer65, 2019-10-25:
@adrianmiriuta Where's your evidence that the OTA transmission between the TX/RX takes the full 6.66ms? Your argument appears to be based on that.

AlessandroAU, 2019-10-25:
Crossfire isn't full duplex though? It's half duplex like all the other systems. You can go look at the FCC docs and see a picture of the internals, it's a single SX1272 module inside, same deal with the RX.

OpenTX latency is about 5-6ms, (I measured), if the frame time was 3.3ms we would expect the overall latency to be about 10ms, instead we see it's about 14 ms (https://docs.google.com/spreadsheets/d/1LPpkwxKVqz1JMhiC_-FVhWYFZbWIHVbVZDnu8YpqSvI/edit#gid=891788793) If the frame time was 6.66ms instead this would match up nicely. 

Most likely some of the on-air time is reserved for a telm response and the RX might simply send a duplicate packet to the FC in this case. If someone has a CRSF RX handy, (I do not) please probe the DIO0 pin of the RF chipset, typically a pulse is generated on this pin with each TX/RX event so it should be fair easily to confirm if the frame interval is really 3.33ms.

AlessandroAU, 2019-11-13:
I wrote some code to spit out CRSF packets (just raw RC data) at various speeds. I can't seem to get it (BF) to drop packets. (https://github.com/AlessandroAU/ESP32-CRSF-Packet-Gen)

The thing in I have seen this issue myself with my own RC link I've been developing.... (https://github.com/AlessandroAU/ExpressLRS) Issue might be somewhere else, but it seems when just processing RC commands BF seems to be ok? Could be an edge case with TLM or something else? needs more testing...

This image looks messy but the pkpk values are ~4.8-5.2 so everything working as it should it seems 
[image]

closed 2019-11-19 by mikeller

theely, 2019-12-01:
### Thanks @etracer65 for the fix!
As mentioned in the fix commit I can confirm the issue has been successfully fixed, looking forward for the official release of BF 4.1.2
Also, thanks everyone else who participated in the lengthy discussion of this thread and helped spotting a light on this issue.

 Here a before and after measurement:
[image]

JamesKingdon, 2019-12-02:
Great work! Could the same sort of problem be affecting other receivers?

bizmar, 2019-12-02:
Other receivers have a lower rate of transmission, so the issue wasn't as bad. Sbus, ibus and spectrum have been updated in the same way.

JamesKingdon, 2019-12-02:
That's great - many thanks.

PR #9190 (2019-11-19): Restructure CRSF frame processing and RC channel handling to prevent erroneous dropped frames
Fixes #9025 

If another packet (of any type) was transmitted by the receiver before the RX check function from the task scheduler had processed the previously received channel data packet, then the first byte from the new packet would corrupt the previous data and cause an invalid CRC check and the packet would be ignored (dropped). This is because there was only a single frame buffer and new data received by the ISR would write to the buffer that may be in processing by the RX task. This problem wou [... cut]