betaflight/betaflight #9028: RSSI reading low/wrong on 4.0.1 #1908

hObBiT-iM-bLuTrAuScH, 2019-10-11:
**Describe the bug**
The RSSI value is different in 4.0.6 RC6 and 4.1.0 #1908

**To Reproduce**
Flashing Firmware 4.0.6 RC6 where RSSI used to be correct (99-100%).
Flashing Firmware 4.1.0 #1908 and RSSI is only 50-70%

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Show RSSI value of 99-100% as RX and TX are close to each other.

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

\\\
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: PINIO 1
A15: OSD_CS
B00: LED 1
B01: FREE
B02: FREE
B03: CAMERA_CONTROL
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: PREINIT 2
C14: FREE
C15: FREE
D02: SERIAL_RX 5

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: CAMERA_CONTROL
TIM3: FREE
TIM4:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
\\\

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

CLRACING/ TALON F7 V1/ Banggood
Crossfire nano RX, TBS nano Pro32
Smartaudio from FC

**Additional context**
Add any other context about the problem here.

IvoFPV, 2019-10-11:
>Flashing Firmware 4.1.0 #1908

Please don't use #(number) for build numbers because that references another PR/Issue with that number.

Is this related to the previous issue where you were figuring out LQ and RSSI?
Could you join our Slack for help with configuration issues in general because this is not a right place for those? https://slack.betaflight.com/

hObBiT-iM-bLuTrAuScH, 2019-10-11:
This is not related to #9023. However i figured out the difference between the two firmware versions. i could set an rssi offset but i think that is not the correct way to report back an issue then. send an invitation request for slack. *@hotmail.com

hObBiT-iM-bLuTrAuScH, 2019-10-11:
```
# diff all

# version
# Betaflight / CLRACINGF7 (CLR7) 4.1.0 Oct 10 2019 / 09:08:51 (1b1647fc9) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF7
mcu_id 003000173237510131313935
signature 

# name

# feature
feature TELEMETRY
feature ESC_SENSOR

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 2048 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200

# aux
aux 0 0 0 900 1050 0 0
aux 1 13 3 2000 2100 0 0
aux 2 35 2 1800 2100 0 0
aux 3 39 3 900 1000 0 0

# adjrange
adjrange 0 0 2 900 2100 12 1 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set gyro_lowpass2_hz = 325
set gyro_to_use = FIRST
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 6,50,57
set mag_hardware = NONE
set serialrx_provider = CRSF
set blackbox_device = NONE
set motor_poles = 12
set bat_capacity = 850
set vbat_max_cell_voltage = 420
set yaw_motors_reversed = ON
set small_angle = 180
set transient_throttle_limit = 10
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_alarm = 50
set osd_cap_alarm = 850
set osd_vbat_pos = 2433
set osd_rssi_pos = 2074
set osd_link_quality_pos = 2069
set osd_rssi_dbm_pos = 12
set osd_vtx_channel_pos = 2048
set osd_current_pos = 407
set osd_mah_drawn_pos = 2456
set osd_motor_diag_pos = 396
set osd_craft_name_pos = 278
set osd_display_name_pos = 32
set osd_warnings_pos = 2217
set osd_battery_usage_pos = 5
set osd_disarmed_pos = 2250
set osd_esc_tmp_pos = 402
set osd_esc_rpm_pos = 163
set osd_esc_rpm_freq_pos = 192
set osd_flip_arrow_pos = 2286
set osd_core_temp_pos = 394
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_total_time = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5865
set vtx_pit_mode_freq = 5865
set vcd_video_system = NTSC

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set vbat_pid_gain = ON
set iterm_relax_cutoff = 35
set d_min_advance = 0
```

etracer65, 2019-10-11:
Where is the `diff` from previous versions that showed a different calculation?

hObBiT-iM-bLuTrAuScH, 2019-10-11:
Well, are they not the same? Well, i´ll do some more work, flash back to 4.0.6 and give u guys what u want :D

```
# diff all

# version
# Betaflight / CLRACINGF7 (CLR7) 4.0.6 Sep  1 2019 / 00:23:52 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF7
manufacturer_id 
mcu_id 003000173237510131313935
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1050 0 0
aux 1 13 3 2000 2100 0 0
aux 2 35 2 1800 2100 0 0
aux 3 39 3 900 1000 0 0

# adjrange
adjrange 0 0 2 900 2100 12 1 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass2_hz = 325
set gyro_to_use = FIRST
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 6,50,57
set mag_hardware = NONE
set serialrx_provider = CRSF
set blackbox_device = NONE
set motor_poles = 12
set bat_capacity = 850
set vbat_max_cell_voltage = 420
set yaw_motors_reversed = ON
set small_angle = 180
set transient_throttle_limit = 10
set osd_rssi_alarm = 50
set osd_cap_alarm = 850
set osd_vbat_pos = 2433
set osd_rssi_pos = 2074
set osd_link_quality_pos = 2069
set osd_vtx_channel_pos = 2048
set osd_current_pos = 407
set osd_mah_drawn_pos = 2456
set osd_motor_diag_pos = 396
set osd_craft_name_pos = 278
set osd_display_name_pos = 32
set osd_warnings_pos = 2217
set osd_battery_usage_pos = 5
set osd_disarmed_pos = 2250
set osd_esc_tmp_pos = 402
set osd_esc_rpm_pos = 163
set osd_esc_rpm_freq_pos = 192
set osd_flip_arrow_pos = 2286
set osd_core_temp_pos = 394
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set vtx_band = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5865
set vtx_pit_mode_freq = 5865
set vcd_video_system = NTSC

# profile
profile 0

set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set vbat_pid_gain = ON
set iterm_relax_cutoff = 35
set d_min_advance = 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 133
set pitch_rc_rate = 133
set roll_expo = 33
set pitch_expo = 33
set yaw_expo = 33
set roll_srate = 33
set pitch_srate = 33
set yaw_srate = 33

# rateprofile
rateprofile 1

set roll_rc_rate = 133
set pitch_rc_rate = 133
set roll_expo = 22
set pitch_expo = 22
set yaw_expo = 22
set roll_srate = 33
set pitch_srate = 33
set yaw_srate = 33

# rateprofile
rateprofile 2

set roll_rc_rate = 133
set pitch_rc_rate = 133
set roll_expo = 11
set pitch_expo = 11
set yaw_expo = 11
set roll_srate = 33
set pitch_srate = 33
set yaw_srate = 33
```[
![4 0 6 - correct RSSI](https://user-images.githubusercontent.com/32323389/66672483-a7ee3d00-ec5e-11e9-9341-18d70b670556.jpg)
](url)

hObBiT-iM-bLuTrAuScH, 2019-10-11:
4 1 0 RC6 - also wrong RSSI reading

[image]

hObBiT-iM-bLuTrAuScH, 2019-10-11:
Voltage reading from the USB also is different in all three firmware version

jflyper, 2019-10-13:
> Voltage reading from the USB also is different in all three firmware version.

Battery voltage read while battery voltage is absent does not mean anything. An ADC used (wherever that is) is reading some floating / phantom / leak voltage.

> So how many versions do u want me to go back to confirm that RSSI on a 2m Crossfire link should be 100%?

Preferably, you can bisect targets between 4.0.0 and 4.1.0-RC to find out where the value has started to change.

Reporting values of `debug[]` array when `debug_mode` is set to `CRSF_LINK_STATISTICS_UPLINK` may also help.

laz2wiringpi, 2019-10-15:
serialrx_provider =  CRSF  
was changed in 4.1 
it now uses native  CRSF_LINK_STATISTICS  if CRSF   is selected 
ie in 4.1 you have native LQ 0-299 (modes 0,1,2)    and RSSI DBM  
you can bypass to the old way  it with aux  channel if you like . 

  unknown..
  if the configurator  is valid as this is  DBm  (lower better)   and not % of rssi (higher better)   there two different ways of calculating rssi .

hObBiT-iM-bLuTrAuScH, 2019-10-16:
Thx, i already did set it up this way. Someone in Slack told me that this is a known issue, that RSSI reads different in other firmware versions. However, RSSI is different when i set an AUX channel. Someone should think about this and or refer to the known issue.

closed 2019-11-22 by stale[bot]