betaflight/betaflight #9030: TBS Unify Pro32 and CRSF passthrough

TheMrJahn, 2019-10-12:
Nowadays I can do two different configurations with TBS stuff:
1. Solder crsf receiver and pro32 to an fc, so bf can control vtx (I like this scheme most);
2. Solder additional crsf tx/rx pins directly to pro32. In this way bf cannot control vtx (smartaudio pin = crsf rx), but users are able to update unify fw using crsf.

It would be best to have an option, to connect pro32 to full uart port, use it as sa or crsf passthrough when needed.

[image]
[image]

mikeller, 2019-10-12:
Does TBS offer a way to update the VTX firmware by connecting the VTX to a PC, without having to use proprietary functionality built into the CRSF hardware? If not there is not much chance that this can be done, safe for one that would require a second UART to be connected to the VTX control pins on the CRSF RX, and then pass through the VTX control / update data.

TheMrJahn, 2019-10-13:
Pro32 HV can be updated through usb port, Pro32 Nano can be updated only via OTA when its connected to crsf rx directly.

mikeller, 2019-10-13:
@TheMrJahn: Doesn't look like Betaflight can do much to enable in-place updating for the Pro32 Nano then - it will likely only be possible to have it connected by Betaflight _or_ by the CRSF RX.

swissfreek, 2019-11-01:
But that's only because the only way to connect a VTx via CRSF is directly to the RX. It's my understanding that CRSF is a network, in that anything connected via CRSF can see anything else in the chain. So for example, if you connect a TBS Rx via CRSF to BetaFlight, and then also connect a Unify Evo to the same RX via CRSF, the Evo can see telemetry data from BetaFlight even though it is not directly connected.

Would it not theoretically be possible to enable CRSF communication to the VTx over a UART (which is likely currently being used for SmartAudio) and thus link everything so that the FC can still control the VTx like with SmartAudio would, but then the CRSF link is still active on the VTx to enable things like MyVTX or OTA updates, or whatever other neat features TBS is planning to add in the next "2 weeks"? It's possible that I am completely misunderstanding the nature of the system, but the fact that all the RXs and the Evo have the ability to maintain two CRSF ports makes me think it might be possible?

mikeller, 2019-11-01:
@swissfreek: Yes, the implementation of a 'CRSF hub' is possible, and has been planned for Betaflight for some time now: #6917 
But since there does not seem to be any interest from TBS to open up CRSF for use by third parties, this will not be possible in the forseeable future, as any future changes to the protocol from TBS' side will break whatever implementation is in Betaflight.
Also, taking into account the specification for CRSF, I do not think that the CRSF protocol will be used for things like the transfer of new firmware to devices, so this will likely require proprietary extensions to CRSF with specs that are not publicly available.

swissfreek, 2019-11-01:
I’m just grasping at straws here, but is CRSF not open already? If the FC can hear and speak CRSF to the receiver, aka sending telemetry, and if a Unify EVO connected to the receiver can also read that telemetry and place it on the OSD, should it not theoretically be possible to also send VTx commands back up through CRSF that the VTx could see and respond to? Or is that what TBS are restricting access to?

I’m not sure there would be a need for a hub. On a crossfire receiver that has two CRSF ports enabled, it doesn’t care which port is connected to the VTx and which is connected to the FC. They both get the full CRSF stream and parse it as needed, right? If I move the sticks to control my EVO through the OSD, it will also move the corresponding channels on the flight controller. So it stands to reason that whatever the crossfire module is using to change channels through LUA over CRSF should also be able to be sent from the flight controller. Or, again, is that what they aren’t sharing? Or am I oversimplifying due to lack of understanding?

mikeller, 2019-11-01:
@swissfreek: No, the CRSF specification is proprietary and owned by TBS. There is no point in Betaflight adopting more proprietary protocols that can be changed by the owner at any time, without any notification, effectively breaking all third party implementations.

swissfreek, 2019-11-01:
Fair enough. That's a bummer.

mikeller, 2019-11-01:
@swissfreek: Yeah. There were talks with TBS about opening up CRSF to the community, but they went nowhere unfortunately.

tgreer, 2019-11-03:
@mikeller, CRSF is already supported from within BF for RX. CRSF is like a bus connection as mentioned, so all that's missing is CRSF VTX config rather than SA. 

There's no past implementations to break as far as I can see, as it's just additional messages to be sent/received on the crsf port?

Is there really issues with TBS not giving you details?

swissfreek, 2019-11-03:
> @mikeller, CRSF is already supported from within BF for RX. CRSF is like a bus connection as mentioned, so all that's missing is CRSF VTX config rather than SA.
>
> There's no past implementations to [...]

Right, that’s how I perceived it, too, but quite possible I have no idea what I’m talking about.

We’re not asking TBS to allow folks to come up with new uses for CRSF, we’re asking for TBS to allow the use of CRSF in ways that it is already being used currently. I can’t imagine they didn’t expect that this request would come, and that the advent of CRSF and OTA would force all users to abandon controlling their VTX from the FC or otherwise solder a wire back and forth from the RX to the FC every time they wanted to switch from CRSF to SA. If that were the case they’d have a separate SA pin on every VTX like the Evo does so you could do both.

mikeller, 2019-11-04:
@tgreer: The CRSF support that is in Betaflight at the moment is for RX / telemetry support only, and it is for a proprietary version of the protocol without a publicly available specification.
After much negative experience with supporting the SmartAudio protocol that is in the same situation with respect to the protocol specification we do not have any interest in adding more support for proprietary protocols - in the end this will just be to the detriment of users.
The last time I brought this up with TBS, they were in agreement about the current situation, their statement was that at the moment they were not considering opening up CRSF for third party use, and that they would not support third party adoption by anybody.

swissfreek, 2019-11-04:
I understand your sentiment. The funny thing is when Trappy was asked about the possibility of adding a feature whereby you could change the power level via a switch like you can currently do with PitMode, his response was "Why are they [FC devs] not adding CRSF VTx control also?" Well, I guess now we know the answer to that question...

tbs-trappy, 2019-11-04:
each time this discussion comes up I have to post the same thing: the plans to open up the protocol have not changed. a public, open source release is still in the works. our current software is not yet stable enough for a public integration and therefore for a publication of the protocol. TBS SmartAudio is curently the only open VTx control, and each time we change the protocol we face issues with third party software having to integrate it. We are trying to avoid this situation for CRSF, because in comparison, the changes to SA are minimalistic when compared to the changes that happened to CRSF just in the last 4 weeks. if in the meantime there is demand to implement CRSF VTx control, please let us know and TBS will be happy to assist with the efforts.

tbs-trappy, 2019-11-04:
> @tgreer: The CRSF support that is in Betaflight at the moment is for RX / telemetry support only, and it is for a proprietary version of the protocol without a publicly available specification. [...]

all of the above is false. SA is open source, the protocol is publically available, AND TBS have shared the protocol with you 6 weeks ahead of release. I have also never publically (or privately) claimed that CRSF would not be opened, and a message of this kind was once released by one of our devs but it was a misunderstanding as to the timeframe, and it was discussed with you personally.

mikeller, 2019-11-04:
@tbs-trappy: 

> SA is open source, the protocol is publically available, it is not proprietary

Sorry, I was not aware of this change. Do you have a link to the protocol specification to point manufacturers wanting to implement SmartAudio with their VTX hardware to?

> I have never publically (or privately) claimed that CRSF would not be opened.

My apologies, it looks like I did not express myself clear enough - the last time I talked to Remo Masina and asked him for a timeline of when CRSF would be opened up in order to schedule the work to add support to Betaflight he told me that no date had been set for this.

swissfreek, 2019-11-04:
Sorry guys, did not intend to stir drama, good to hear it was simple miscommunication. Bummer that there’s (understandably) nothing to be done about it now, but it’s on the road map. Knowing that, I can wait patiently.

tbs-trappy, 2019-11-04:
> @tbs-trappy:
> 
> > SA is open source, the protocol is publically available, it is not proprietary
> 
> Sorry, I was not aware of this change. Do you have a link to the protocol specification to point manufacturers wanting to implement SmartAudio with their VTX hardware to?
Direct link: https://www.team-blacksheep.com/tbs_smartaudio_rev08.pdf
Linked to from all Video transmitter pages, e.g.: https://www.team-blacksheep.com/products/prod:unify_pro_hv_race2_m

> My apologies, it looks like I did not express myself clear enough - what I wanted to express is that the last time I talked to Remo Masina and asked him for a timeline of when CRSF would be opened up in order to schedule the work to add support to Betaflight he told me that no date had been set for this.
yes, this has also not changed. there is currently no date, once we feel it's stable it will be opened

mikeller, 2019-11-04:
@tbs-trappy:

> Direct link: team-blacksheep.com/tbs_smartaudio_rev08.pdf

Cool, thanks so much for opening this up - and sorry again for not having picked up on it.

> yes, this has also not changed. there is currently no date, once we feel it's stable it will be opened

Tanks. We'll be waiting for this to add support for CRSF configuration / pass through to Betaflight - as you say, there is little point in adding support for a protocol that is still undergoing frequent change and not yet fully finalised.

BobRob77, 2019-11-10:
It's really a bummer though - Just put together a new quad running a TBS Unify Pro32 Nano and a Crossfire Nano and with the included 20x20 pcb to mount them together, setting them up to use CRSF Serial communication is so convenient, and the functionality to control the VTX using lua is heavily improved and updating it is also extremely conventient.

Enter Betaflight 4.1 with it's own great improvements in managing the VTX and it falls apart due to having to use Smartaudio over CRSF to get any sort of communication with Betaflight. So, I can just add my whish for CRSF integration to be added, and I do realize this is on TBS, not Betaflight  to make possible.

closed 2019-12-17 by stale[bot]